MarlinFirmware/Marlin #10088: Wrong Z height on dual extension and nozzles goes down on tool change

FNeo31, 2018-03-12:
Hello,

I put Marlin 1.1.8 on my 3D printer ( senhai3D - one acrilic Chinese version of prusa i3)
And work well in single extruder. So I make some parts and reassemble all my printer like the attached image, and change the configuration for dual extruder and put my printer running again. I test the first extruder and everything was well, test the second and all was well to, so I tried to test the dual extrusion print to calibrate the relative position of my extruders and I check some situations:

1: When I print with the two extruders at same time the height of my printed object was wrong 8.20mm instead of 5mm
2: When I tried to change between extruders like : 
G0 T0; 
G0 T1;
G0 T0; 
G0 T1;
G0 T0; 
G0 T1;
G0 T0; 
G0 T1;
G0 T0; 
G0 T1;
The extruders goes down at each change.

Can you help?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

FNeo31, 2018-03-12:
Last image was the original version when I bought it

Studiodyne, 2018-03-13:
Send here your config.h

FNeo31, 2018-03-13:
Attached the config,

My board is a Arduino MEGA2560R3 with RAMPS 1.4 with some changed PINS to allows 2 Ceramics for each extruder heated Bed and control fans.

Induction Z endstop for Bed Leveling

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1808632/config.zip)

Studiodyne, 2018-03-13:
try to compile with 2.0
If you have problem to compile , search in issue tab , all is explain somewhere 
and give feedback

FNeo31, 2018-03-13:
I just need to replace the config files or I need to make something else?

thinkyhead, 2018-03-14:
I'll just post the boilerplate…

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

You can use your 1.1.8 configurations. Marlin will bark during compilation if anything needs to be updated.

FNeo31, 2018-03-14:
I tried to use the configuration.h and configuration_adv.h on bugfix-2.0.x and bugfix-1.1.x downloaded today before test.

In the bugfix-2.0.x I tried to compile and made the version error (image attached)

In the bugfix-1.1.x compiles successfully but the result was the same that I had before (in the dual printer test I didn't make the test yet)

I attached here all the files like I compile and the gcode that I used to test the tool change (tool change test.gcode)

Thank you for your time and availability

Tomorrow I will try to make the dual print test to make sure if the Z height trouble can be solved (DUAL TEST.gcode attached)

[image]

[Marlin-bugfix-1.1.x_10088_tool_change_dual.zip](https://github.com/MarlinFirmware/Marlin/files/1813343/Marlin-bugfix-1.1.x_10088_tool_change_dual.zip)

thinkyhead, 2018-03-15:
To get rid of that error, you need to have a line like this at the top of each configuration, matching the expected version. Either `010107` or `020000` depending on 1.1.x or 2.0.x…
```cpp
#define CONFIGURATION_H_VERSION 020000
```
```cpp
#define CONFIGURATION_ADV_H_VERSION 020000
```

Studiodyne, 2018-03-15:
Delete temp folder
And compilation will start a new clean build

FNeo31, 2018-03-15:
How do I clean the temporary?

FNeo31, 2018-03-15:
I'll change the version on configuration file and try to compile and test on bugfix-2.0.x

Studiodyne, 2018-03-15:
To have a good 2.0 , make is explained here 
[#7990](https://github.com/MarlinFirmware/Marlin/issues/7990)

If it's your first time , you create a temp folder ,no need to delete , but in the future , when you see something wrong , or if you compile 1.1 and after 2.0 , you have to clean before

Thks

thinkyhead, 2018-03-15:
Sorry, @Studiodyne is being misleading. The compile issue you posted about is simply because the configurations need updating. It has nothing to do with leftover bits from a previous build.

FNeo31, 2018-03-15:
at the moment I'm using the bugfix-1.1.x but didn't solve the problem, so I will try the bugfix-2.0.x to check if the problem continues or if the problem disappear.

I have 2 folders and don't open the 2 builds at same time

Bobsy5361, 2018-03-15:
OK.. i have tried it on bare 2560 .. 2560 with ramps 1.4 and 1.5 and 1.6 connected.. unpowered  powered. i would think it would compile as a bare board without any thing added.

thinkyhead, 2018-03-15:
@Bobsy5361 — Please file a separate issue.

thinkyhead, 2018-03-15:
@FNeo31 — I expect the problem will exist in both branches. The tool change code is always a little tricky, as it tries to manage a wide variety of configurations.

- Are you using any form of bed leveling?

To get started troubleshooting, please do the following:

- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do some tool-changes with `T0` and `T1` to expose the issue.
- Use `G0` and/or `G1` to move around if this is also required to expose the issue.
- Copy the log output into a **.TXT file** and attach it to your next reply.

thinkyhead, 2018-03-15:
@FNeo31 — I expect the problem will exist in both branches. The tool change code is always a little tricky, as it tries to manage a wide variety of configurations.

- Are you using any form of bed leveling?

To get started troubleshooting, please do the following:

- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Perform a `G29` to do bed leveling (if this is needed to expose the issue).
- Do some tool-changes with `T0` and `T1` to expose the issue.
- Use `G0` and/or `G1` to move around if this is also required to expose the issue.
- Copy the log output into a **.TXT file** and attach it to your next reply.

FNeo31, 2018-03-15:
I tried with the bugfix-2.0.x and make the same.

Attached a video of "tool change test.gcode" video start after make the G29.
position at Z2 and after you can see in the video. goes down each time that came from T0 to T1
[VID_20180315_205624 (1).zip](https://github.com/MarlinFirmware/Marlin/files/1817187/VID_20180315_205624.1.zip)

thinkyhead, 2018-03-15:
> goes down each time that came from T0 to T1

You've explained it well. So let's continue troubleshooting and gathering log output.

FNeo31, 2018-03-15:
I will make it in a few minutes.

FNeo31, 2018-03-15:
Now on position X70 Y70 make the opposite goes up when I change from T1 to T0 and continue going in each change from T1 to T0 and stay in position in Z from T0 to T1.

Just append after G29

with G28 stay at same position when change between T0 and T1 both ways.

[FULL LOG 1.1.x.txt](https://github.com/MarlinFirmware/Marlin/files/1817277/FULL.LOG.1.1.x.txt)

thinkyhead, 2018-03-15:
Thanks for the log and the feedback that bed leveling is involved. I'll study the log, examine the tool-change code, and see if anything obvious jumps out. If we need to gather more data I'll let you know.

thinkyhead, 2018-03-16:
Hi @FNeo31 — I'm unable to see the exact cause so far. But I understand the _general_ cause. Basically, when you switch nozzles with bed leveling enabled, the new nozzle will be at a different distance from the bed than the old one, just by virtue of the bed not being exactly level. So Marlin changes the current Z position to account for this. Apparently it's not getting the adjustment exactly right.

I'll make a branch with some extra logging for you to test tomorrow, and that should be more revealing.

While you're waiting, you might try Marlin 1.1.6, 1.1.5, or 1.1.4 to see if any of these have the same bug. If one of them turns out to work correctly, then we can compare and see what's changed. Of course, no worries if you want to skip all that. The extra logging should give us a good picture of the situation.

I'll post as soon as the test branch is ready for download.

Studiodyne, 2018-03-16:
@thinkyhead In life , there are some guys that knows all , that are perfect and everybody dream to be like them , to be like thinkyhead for example 😄 and there are guys that are not perfect and have to make their best as possible to solve all fact of life 😄 . And suggest to clean the folder is not stupid ... You can believe it ! So many times i have solved some compilations errors and help many users just with this !

thinkyhead, 2018-03-17:
@Studiodyne — That's the last straw. Get lost.

thinkyhead, 2018-03-17:
[image]

@Studiodyne — That's the last straw. Get lost and stop wasting our time! This is our workplace, not a playground, and we don't want you chiming in on every topic with impertinent suggestions. And we have [zero tolerance](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/code_of_conduct.md) for children posting personal attacks. You will shortly be blocked.

My apologies to you again, @FNeo31.

thinkyhead, 2018-03-17:
@FNeo31 — This issue is moved to #10130. This one is locked for review by the organization.

closed 2018-03-17 by thinkyhead