MarlinFirmware/Marlin #10089: 3D Printer not responding to G92, G28, G29, G0, or Pronterface homing commands

Power3DPrinting, 2018-03-12:
### Description
Marlin 1.1.8, SCARA config, will not move or home individual axis, dead set on moving two motors and has a heart attack whenever an endstop is triggered. G28 rotates both motors in the wrong direction but flipping homing direction does nothing, G29, G92, G0, non responsive, M84, M0 or subsequent motor commands like M360 have no effect or the machine crashes. M119, endstops, machine, all check out fine, purely a firmware problem.

### Steps to Reproduce

1. Download Marlin 1.1.8 and copy and paste SCARA config.h and config_adv.h, configure respectively
2. Get frusterated at github because it won't let you upload your files to share, sorry, I'd upload my files if I could to share my kinematics but for the most part they shouldn't interfere with homing since I neither removed or edited anything homing related. Also, if anyone knows where the homing script is, that would be a big help since Marlin_main.cpp is a cluttered mess.
3. Try gcode moves or even just pronterface and keep your finger over the power switch

**Expected behavior:** [What you expect to happen]
G28 orders the printer to rotate its X-Motor (homing direction inversable), stop at its X-min end stop, then the Y-Motor (homing direction inversable) to its Y-max endstop, finally the Z-Motor lowers the hot end to the table like any normal Cartesian 3D printer. Then the X and Y motor rotate a given number of degrees from their homed positions before I let my inverse kinematics kick in and marks that point as (0, 500, 0) and I can order a G29 to probe. 

**Actual behavior:*
Unfortunately, individual homing is not working, irreversible wrong homing direction and worse still is a single endstop causes the entire machine to stop, xy and z axis. I tried manual moves with G92 and G0 or just pronterface's interface and nothing takes effect. 
**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* Read the "Contributing" page (under "Helpful resources") for additional guidelines.

thinkyhead, 2018-03-13:
Sorry about that. SCARA is essentially broken in Marlin 1.1.8 and will also be broken in Marlin 1.1.9. You'll have to stick with the old Morgan SCARA fork until SCARA support is more mature.

I'm working for a client on a SCARA robot called the "MakerArm" and have just recently implemented homing and various other features, but unfortunately that work is not entirely complete, and I cannot contribute the code back to the main Marlin fork until they ship.

Power3DPrinting, 2018-03-13:
:c

Power3DPrinting, 2018-03-13:
Thanks for letting me know about the G29

Power3DPrinting, 2018-03-13:
Any thoughts on G92? That one is just stuck sending and would be a big help to figuring out what's wrong with the machine if I could set the current position to 0,0,0 with G92 X0 Y0 Z0, is my format or idea of how it works out of date?

Power3DPrinting, 2018-03-13:
Reading the G code commands, it looks like G90 and G91 set an absolute and relative coordinate for the machine, an absolute position sounds straightforward but that too is stuck sending, not entirely sure what G91 is meant to do by setting a relative position, the documentation there is pretty scarce

Power3DPrinting, 2018-03-13:
Also, I am familiar with the Makerarm project that was started in 2015 and failed to ship in 2016, I mean no offense, but I don't think there is much trust in makerarm. Credit to you for working out a solution, it might stand to reason Marlin has made auto leveling unreasonable then if it's not a simple mapping of several coordinate points.

thinkyhead, 2018-03-15:
> not entirely sure what G91 is meant to do by setting a relative position setting a relative position

It puts the machine in "relative mode" so sending `G1 X1` moves 1mm to the right, always. If you want to understand more about it, it is documented at linuxcnc.org, reprap.org, and marlinfw.org.

> I don't think there is much trust in makerarm

It's still in development, it's a very dedicated team of nice people, and they still pay me. We're working in earnest, but there have been numerous delays due to various unforeseen issues that I won't get into.

> it might stand to reason Marlin has made auto leveling unreasonable then if it's not a simple mapping of several coordinate points.

It is a simple mapping of several coordinate points.

boelle, 2019-02-20:
@thinkyhead has scara not been removed completely? and if its not going back then maybe we can close this one?

gnmrc-s, 2022-04-23:
I'm trying to home a left-handed SCARA arm using bugfix 2.0 with no success.
I have tried different methods, the one that has bringed me closer to an actual homing was explained here https://github.com/MarlinFirmware/Marlin/issues/22233.

It's the same that was used in "How to Mechatronics" video about a SCARA laser engraver.
 
video: https://www.youtube.com/watch?v=8qc044LY6hc

with this method x y motors move but they are not stopped by the endstops, z homing works fine.

I have also seen here https://github.com/MarlinFirmware/Marlin/issues/20764 that with an old marlin 2.0 version MP_SCARA homing was fuctioning and this correction was later implemented in marlin 2.0.8. but with the implementation of other features it gave bugs as reported.

This are my config files:
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/8547523/Configuration.files.zip)

gnmrc-s, 2022-05-07:
I have solved it, well not completely, it's a bug with the firmware that enable the endstops feature only if you home while settings the endstops as MIN endstops.

Loki44, 2022-11-03:
> I'm working for a client on a SCARA robot called the "MakerArm" and have just recently implemented homing and various other features, but unfortunately that work is not entirely complete, and I cannot contribute the code back to the main Marlin fork until they ship.

Did anything come of this?  It looks like MakerArm are listed as permanently closed on google.

closed 2024-12-20 by github-actions[bot]