MarlinFirmware/Marlin #10090: Issues compiling with tmc26xstepper library and ServoTimers.h

robertpdouglass, 2018-03-13:
Trying to setup two tmc2130's on my RADDS/DUE combo and am running into this error.  Not sure what to do.  I should also mention that the servotimers.h in src/HAL/HAL_DUE needs to be renamed to ServoTimers.h.

In file included from sketch/src/gcode/feature/trinamic/M906.cpp:28:0:
sketch/src/gcode/feature/trinamic/../../../feature/tmc_util.h: In instantiation of 'void tmc_set_current(TMC&, TMC_AxisEnum, int) [with TMC = TMC26XStepper]':
sketch/src/gcode/feature/trinamic/M906.cpp:44:5:   required from here
sketch/src/gcode/feature/trinamic/../../../feature/tmc_util.h:51:3: error: no matching function for call to 'TMC26XStepper::setCurrent(const int&, double, double)'
   st.setCurrent(mA, R_SENSE, HOLD_MULTIPLIER);
   ^
sketch/src/gcode/feature/trinamic/../../../feature/tmc_util.h:51:3: note: candidate is:
In file included from sketch/src/gcode/feature/trinamic/../../../module/stepper_indirection.h:55:0,
                 from sketch/src/gcode/feature/trinamic/M906.cpp:29:
/home/bobbyofna/Arduino/libraries/TMC26XStepper-master/TMC26XStepper.h:312:7: note: void TMC26XStepper::setCurrent(unsigned int)
  void setCurrent(unsigned int current);
       ^
/home/bobbyofna/Arduino/libraries/TMC26XStepper-master/TMC26XStepper.h:312:7: note:   candidate expects 1 argument, 3 provided
exit status 1
Error compiling for board Arduino Due (Programming Port).

thinkyhead, 2018-03-13:
> I should also mention that the servotimers.h in src/HAL/HAL_DUE needs to be renamed to ServoTimers.h.

Ah, wonderful. We love it when we get to fight with Git about case-insensitive filesystems. 😀

thinkyhead, 2018-03-13:
As for your main issue, don't enable `HAVE_TMCDRIVER` for the TMC2130. You should only enable `HAVE_TMC2130`.

closed 2018-03-13 by thinkyhead

thinkyhead, 2018-03-13:
Any comment on this @teemuatlut ? Have we generated an incompatibility when TMC26X stepper drivers are used with TMC2130 or TMC2208 in Marlin?

robertpdouglass, 2018-03-13:
> In the meantime, be sure not to enable HAVE_TMCDRIVER unless you're using TMC26X stepper drivers. For 2130 you should only use HAVE_TMC2130 and its axis options.

Both of those variables are enabled.  I posted the configs on drive to help if needed...

Configuration.h
https://drive.google.com/open?id=1jbNzwDYqdScTGcork1y6M4ffPiJ_vHuP

Configuration_adv.h
https://drive.google.com/open?id=1VBOQKDVCyM-9c6uuMk7LwC4yweJ0OLv_

robertpdouglass, 2018-03-13:
Oh wait I read that wrong, I was under the impression that I also needed to have the HAVE_TMCDRIVER enabled if I was using tmc2130's.  Now I am getting an error regarding undefined SPI pins...

`Arduino: 1.8.5 (Linux), Board: "Arduino Due (Programming Port)"

In file included from sketch/src/feature/../inc/../HAL/HAL.h:40:0,
                 from sketch/src/feature/../inc/MarlinConfig.h:28,
                 from sketch/src/feature/tmc_util.cpp:23:
sketch/src/feature/tmc_util.cpp: In function 'void _tmc_say_axis(TMC_AxisEnum)':
sketch/src/feature/../inc/../HAL/HAL_DUE/HAL_Due.h:72:50: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
 #define pgm_read_word(addr) (*((uint16_t*)(addr)))
                                                  ^
sketch/src/feature/tmc_util.cpp:227:25: note: in expansion of macro 'pgm_read_word'
   serialprintPGM((char*)pgm_read_word(&tmc_axes[axis]));
                         ^
sketch/src/module/stepper_indirection.cpp:141:119: error: 'TMC_SW_MOSI' was not declared in this scope
     #define _TMC2130_DEFINE(ST) TMC2130Stepper stepper##ST(ST##_ENABLE_PIN, ST##_DIR_PIN, ST##_STEP_PIN, ST##_CS_PIN, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK)
                                                                                                                       ^
sketch/src/module/stepper_indirection.cpp:148:5: note: in expansion of macro '_TMC2130_DEFINE'
     _TMC2130_DEFINE(X);
     ^
sketch/src/module/stepper_indirection.cpp:141:132: error: 'TMC_SW_MISO' was not declared in this scope
     #define _TMC2130_DEFINE(ST) TMC2130Stepper stepper##ST(ST##_ENABLE_PIN, ST##_DIR_PIN, ST##_STEP_PIN, ST##_CS_PIN, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK)
                                                                                                                                    ^
sketch/src/module/stepper_indirection.cpp:148:5: note: in expansion of macro '_TMC2130_DEFINE'
     _TMC2130_DEFINE(X);
     ^
sketch/src/module/stepper_indirection.cpp:141:145: error: 'TMC_SW_SCK' was not declared in this scope
     #define _TMC2130_DEFINE(ST) TMC2130Stepper stepper##ST(ST##_ENABLE_PIN, ST##_DIR_PIN, ST##_STEP_PIN, ST##_CS_PIN, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK)
                                                                                                                                                 ^
sketch/src/module/stepper_indirection.cpp:148:5: note: in expansion of macro '_TMC2130_DEFINE'
     _TMC2130_DEFINE(X);
     ^
sketch/src/module/stepper_indirection.cpp:141:119: error: 'TMC_SW_MOSI' was not declared in this scope
     #define _TMC2130_DEFINE(ST) TMC2130Stepper stepper##ST(ST##_ENABLE_PIN, ST##_DIR_PIN, ST##_STEP_PIN, ST##_CS_PIN, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK)
                                                                                                                       ^
sketch/src/module/stepper_indirection.cpp:154:5: note: in expansion of macro '_TMC2130_DEFINE'
     _TMC2130_DEFINE(Y);
     ^
sketch/src/module/stepper_indirection.cpp:141:132: error: 'TMC_SW_MISO' was not declared in this scope
     #define _TMC2130_DEFINE(ST) TMC2130Stepper stepper##ST(ST##_ENABLE_PIN, ST##_DIR_PIN, ST##_STEP_PIN, ST##_CS_PIN, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK)
                                                                                                                                    ^
sketch/src/module/stepper_indirection.cpp:154:5: note: in expansion of macro '_TMC2130_DEFINE'
     _TMC2130_DEFINE(Y);
     ^
sketch/src/module/stepper_indirection.cpp:141:145: error: 'TMC_SW_SCK' was not declared in this scope
     #define _TMC2130_DEFINE(ST) TMC2130Stepper stepper##ST(ST##_ENABLE_PIN, ST##_DIR_PIN, ST##_STEP_PIN, ST##_CS_PIN, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK)
                                                                                                                                                 ^
sketch/src/module/stepper_indirection.cpp:154:5: note: in expansion of macro '_TMC2130_DEFINE'
     _TMC2130_DEFINE(Y);
     ^
exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

teemuatlut, 2018-03-13:
You need to pick your SW SPI pins as there are no defaults for the RADDS board.

robertpdouglass, 2018-03-13:
okay.  I have my drivers wired up to the MOSI, MISO, and SCK pins being used by the SD card.  I can't find the pin numbers for those pins.  Any idea what they are or what their variable name would be?  I don't see them in pins_RADDS.h

robertpdouglass, 2018-03-13:
Nvm I spoke too soon, I found them.  Everything works now.  Thanks for all of the help!

ujfe71, 2019-09-04:
> Nvm I spoke too soon, I found them. Everything works now. Thanks for all of the help!

What was your solution ??? A have the same error...