MarlinFirmware/Marlin #10091: Autostart GCODE (Feature Request)

zafff, 2018-03-13:
### Description

Is there any plan in the future to hardcode gcodes when the printer get power? For example, In my (octoprint) script, I have lines like G28 and M140 S70 and having an option like that in the configuration files would be a good addition. I am aware of the autostart scripts but these require an SD card present at all times.

thinkyhead, 2018-03-13:
There is a caveat. With AVR processors Marlin can't tell the difference between a restart due to reconnecting to a host and a cold boot. So any startup G-code would be run in both of those cases.

zafff, 2018-03-13:
Based on what I generally encounter (octoprint losing connectivity at randon times), this should not be an issue since reconnecting to a host is not done automatically. For instance, octoprint connect automatically when getting power, but does not do so when losing connectivity. The possibilities that I can think of are:
1. If, In the middle of a print, connectivity is lost, then the print will fail if the host is sending the gcodes. Reconnecting will reset the microcontroller.
1. Printing is taking place from the SD card and being monitored from the host, then losing connection does not affect anything unless it is happening from a remote place. The print will continue and potentially succeed as well unless the end-user forces a connection which will reset the microcontroller.

If homing is part of the startup code, then the hotend can hit the printed part. This has the same effect as homing when there is still a printed part on the bed. Caution should be advised in that case.

dot-bob, 2018-03-15:
Perhaps specify specific gcode commands that call a firmware based start and end gcode script.  This way you could embed these codes generically in the slicer gcode along with firmware retraction and have gcode that will run on any printer like gcode can on many CNC machines.  If you have your printers setup correctly you should be able to slice for any printer as long as it has the same nozzle type.  Right now I have four printers tuned well enough that when I slice all that is different between them is the start and ending gcode.

InsanityAutomation, 2019-10-15:
@boelle @shitcreek this is done

closed 2019-10-15 by boelle