MarlinFirmware/Marlin #10095: Running out of Memory to program.

wbarber69, 2018-03-13:
I don’t know if I am doing something wrong, so delete this if I’m just being stupid. But I am using a bugfix branch download that is a few days old. I am using a MKS Gen-L V1.0 with TMC2130 drivers. When I add support for my RepRap discount full graphic screen with sd support and tmc2130 I run over by what arduino says is 192% of progmem. Is there something I’m missing or do I need to disable something.  If I comment out SDSUPPORT  I only end up being at 52%.  I’ve tried disabling menu items and other non critical features, but I can’t below 188%.  I don’t technically need an sd card but it would be nice.

Bob-the-Kuhn, 2018-03-13:
Please ZIP & post your configuration.h and configuration_adv.h files.  With those we can make suggestions on how to reduce the memory usage.

You have a 2560 based board so I'm surprised to see such a huge problem with memory.

Are you using bugfix-1.1.x or bugfix-2.0.x?

wbarber69, 2018-03-13:
I’ll pack it all up when I get home.  I’m on bugfix 1.1.x.

ManuelMcLure, 2018-03-13:
What board do you have selected in your Arduino settings? It almost feels like you incorrectly have selected a board that uses an ATMega 1280 instead of one that uses an ATMega 2560.

wbarber69, 2018-03-14:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1809476/Configuration.zip)

wbarber69, 2018-03-14:
ManuelMclure That is not the issue.

thinkyhead, 2018-03-14:
You're being overly ambitious in the `SDCARD_SORT_ALPHA` settings:
```cpp
#define SDCARD_SORT_ALPHA

// SD Card Sorting options
#if ENABLED(SDCARD_SORT_ALPHA)
  #define SDSORT_LIMIT       256    // Maximum number of sorted items (10-256). Costs 27 bytes each.
  #define FOLDER_SORTING     -1     // -1=above  0=none  1=below
  #define SDSORT_GCODE       false  // Allow turning sorting on/off with LCD and M34 g-code.
  #define SDSORT_USES_RAM    true   // Pre-allocate a static array for faster pre-sorting.
  #define SDSORT_USES_STACK  false  // Prefer the stack for pre-sorting to give back some SRAM. (Negated by next 2 options.)
  #define SDSORT_CACHE_NAMES true   // Keep sorted items in RAM longer for speedy performance. Most expensive option.
  #define SDSORT_DYNAMIC_RAM false  // Use dynamic allocation (within SD menus). Least expensive option. Set SDSORT_LIMIT before use!
  #define SDSORT_CACHE_VFATS 2      // Maximum number of 13-byte VFAT entries to use for sorting.
                                    // Note: Only affects SCROLL_LONG_FILENAMES with SDSORT_CACHE_NAMES but not SDSORT_DYNAMIC_RAM.
#endif
```
Change them to:
```cpp
#define SDCARD_SORT_ALPHA

// SD Card Sorting options
#if ENABLED(SDCARD_SORT_ALPHA)
  #define SDSORT_LIMIT       40     // Maximum number of sorted items (10-256). Costs 27 bytes each.
  #define FOLDER_SORTING     -1     // -1=above  0=none  1=below
  #define SDSORT_GCODE       false  // Allow turning sorting on/off with LCD and M34 g-code.
  #define SDSORT_USES_RAM    true   // Pre-allocate a static array for faster pre-sorting.
  #define SDSORT_USES_STACK  true   // Prefer the stack for pre-sorting to give back some SRAM. (Negated by next 2 options.)
  #define SDSORT_CACHE_NAMES false  // Keep sorted items in RAM longer for speedy performance. Most expensive option.
  #define SDSORT_DYNAMIC_RAM false  // Use dynamic allocation (within SD menus). Least expensive option. Set SDSORT_LIMIT before use!
  #define SDSORT_CACHE_VFATS 2      // Maximum number of 13-byte VFAT entries to use for sorting.
                                    // Note: Only affects SCROLL_LONG_FILENAMES with SDSORT_CACHE_NAMES but not SDSORT_DYNAMIC_RAM.
#endif
```

closed 2018-03-14 by thinkyhead

wbarber69, 2018-03-14:
the thing is i never changed any of these values from their default state. at least not on purpose...

Bob-the-Kuhn, 2018-03-14:
Confirmed

----

VR Memory Usage

Device: atmega2560

**256 entries**
 
Program:  147066 bytes (56.1% Full)
(.text + .data + .bootloader)
 
Data:      16065 bytes (196.1% Full)
(.data + .bss + .noinit)

\\\\\\\\\\\\\\\\\\\\\\

**40 entries**
Program:  147832 bytes (56.4% Full)
(.text + .data + .bootloader)
 
Data:       7184 bytes (87.7% Full)
(.data + .bss + .noinit)
 
\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\

**#define SDSORT_USES_STACK  true  
#define SDSORT_CACHE_NAMES false**  

Program:  147522 bytes (56.3% Full)
(.text + .data + .bootloader)
 
Data:       5577 bytes (68.1% Full)
(.data + .bss + .noinit)

Bob-the-Kuhn, 2018-03-14:
You have the same gremlins in your machine that I do!  Can't seem to get rid of them.  😒
FYI - here's the current default for that section
```
 //#define SDCARD_SORT_ALPHA

 // SD Card Sorting options
 #if ENABLED(SDCARD_SORT_ALPHA)
   #define SDSORT_LIMIT       40    
   #define FOLDER_SORTING     -1    
   #define SDSORT_GCODE       false 
   #define SDSORT_USES_RAM    false 
   #define SDSORT_USES_STACK  false 
   #define SDSORT_CACHE_NAMES false 
   #define SDSORT_DYNAMIC_RAM false 
   #define SDSORT_CACHE_VFATS 2     
                                    
 #endif
```

thinkyhead, 2018-03-14:
> the thing is i never changed any of these values from their default state. at least not on purpose...

If you started with a CR10 config, that could be the issue. That configuration pertains to a 1284p (Melzi) that has 16K of SRAM.

wbarber69, 2018-03-14:
Ahhh that does make more sense now... good to know as I do intend to drop the cr10 confit and use default from now on.

thinkyhead, 2018-03-14:
Duck confit? Too greasy for me.

wbarber69, 2018-03-14:
all good now at 68%