MarlinFirmware/Marlin #10097: [2.0.X] Lost Steps on evry Print

xelopics, 2018-03-13:
Hello
i try to get a working version of Marlin 2.0 on my MKS Sbase
i have try to modify digipot Currents but i am not sure is right
i print a XYZ Cube all works fine but on the last steps ( Starting with the Z upside ) i lost steps and the print head is not longer on curent position one or both steppers go to stall maybe any can help to find a solution

[Marlin 2.0 config files.zip](https://github.com/MarlinFirmware/Marlin/files/1808651/Marlin.2.0.config.files.zip)

zafff, 2018-03-13:
I believe that your motors are not providing enough torque to hold the bed. Can you also post the specifications of your stepper motors and the stepper motor drivers?
How are you checking the pots on the drivers? Multimeter, sound, others?

thinkyhead, 2018-03-14:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

closed 2018-03-14 by thinkyhead