MarlinFirmware/Marlin #10098: MK2_MULTIPLEXER- relay switches, driver without signal

honzahs, 2018-03-13:
When the MK2_MULTIPLEXER function is enabled, the relay switches. Unfortunately, only E1 works. E2 and E3 and E4 are switched on, but no signal from the stepper driver is in them. Where is the error?
FW: bugfix-1.1.x

revilor, 2018-03-14:
What is the state of the status LEDs on the multiplexer (there is one per relay) when you select T0, T1, T2, T3?

honzahs, 2018-03-14:
lights up. There is no such problem. I think so

AnHardt, 2018-03-14:
Likely the sequence of the wires from the stepper driver to the switch is wrong.

honzahs, 2018-03-14:
I measured the signal from the output of the board. E0 signal has, E1 E2 E3 signal is not

thinkyhead, 2018-03-14:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop them into your next reply. We're going to need them if we are going to do any testing.

honzahs, 2018-03-14:
thanks. Okey, here files:
[Marlin-bugfix-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/1811852/Marlin-bugfix-1.1.x.zip)

honzahs, 2018-03-14:
//#define LIN_ADVANCE

revilor, 2018-03-14:
My stepper multiplexer works just fine with Marlin so this is probably a hardware issue. Just to be sure: we are talking about this [image]

Then please check the status of the LEDs to rule out some wiring issue. When you put the stepper switcher as on the picture the topmost LED (D1) will always be on (power indicator). The two LEDs below (D2 and D3) will show the binary representation of the index of the currently selected stepper.
So it's

Tool | D1 | D2 | D3
------------ | ------------- | ------------- | -------------
T0 | ON | OFF | OFF
T1 | ON | ON | OFF
T2 | ON | OFF | ON
T3 | ON | ON | ON

honzahs, 2018-03-14:
Yes exactly this type. switching to another extruder does not flow into E-IN. the switch itself works

honzahs, 2018-03-14:
@revilor  can you please send me your configuration? I'd like to try it out

honzahs, 2018-03-14:
switching E1
[image]
switching E2
[image]
switching E3
[image]
switching E4
[image]

simply is not the master signal for the stepper motor for E2, E3 and E4. E1 works

honzahs, 2018-03-14:
my full package Marlin
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1813493/Marlin.zip)

revilor, 2018-03-15:
That's exactly how it is supposed to be. So it's not a Marlin issue but related to the switcher board itself. Maybe a cold solder joint at one of the relay pins?

AnHardt, 2018-03-15:
https://github.com/prusa3d/MK2mm_Stepper_switch/blob/master/stepper-switch.pdf

A+,A-,B+,B- != A+,B+,A-,B-

honzahs, 2018-03-15:
so why is not the signal going to E-IN?

simon-jouet, 2018-03-19:
@honzahs did you figure this out by any chance? I just built my own MK2 multiplexer and I'm having pretty much the same issue.

If I enable MK2_MULTIPLEXER and do tool change I can hear the relays clicking but any extrusion isn't working. (except for tool 0).

To debug things out, I disabled the multiplexer in the config and manually used M42 to set the pins state of the multiplexers and in that case it's working perfectly fine (I need to M18 between tool change)

I did something along those lines in prontrface for debugging
```
G91 ; Relative mode (easier to test extrusion)
G0 E10; Extrude on T0
M18 ;Disable steppers
M42 P40 S255 ; Switch to T1
G0 E10; Extrude on T1
M18
M42 P42 S255 ; Switch to T3
G0 E10
M18
M42 P40 S0 ;Switch to T2
G0 E10
```

revilor, 2018-03-19:
Ok guys, I guess the root of all evil is this option:

#define DISABLE_INACTIVE_EXTRUDER true

Change the value to false and I'm 99.9% sure all four extruders will work.

simon-jouet, 2018-03-19:
@revilor that was it for me, thanks :). That's with the latest Marlin 2.x and a custom MK2 like multiplexer

If it's of any use, my config is the following
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1826771/Marlin.zip)

honzahs, 2018-03-19:
@revilor Yes it is. two days ago I came alone

thinkyhead, 2018-03-21:
> Ok guys, I guess the root of all evil is this option:

Aha. Very nice catch. We'll patch that today.

commit 98759753 (2018-03-21): For MK2_MULTIPLEXER undef DISABLE_INACTIVE_EXTRUDER

Fix #10098
commit 7d9d6c75 (2018-03-21): For MK2_MULTIPLEXER undef DISABLE_INACTIVE_EXTRUDER

Fix #10098
commit ec46336d (2018-03-21): For MK2_MULTIPLEXER undef DISABLE_INACTIVE_EXTRUDER

Fix #10098
closed 2018-07-31 by thinkyhead