MarlinFirmware/Marlin #10104: Print size scaling

skyflyer, 2018-03-14:
### Description

This more of a question than a bug report.

I sliced a print, which is near bed bounds in one dimension. The slicer added the skirt around, and this results in the print size, exceeding the bed size.

For reference, the bed size is 200x200, the print is here:
[stand.zip](https://github.com/MarlinFirmware/Marlin/files/1811756/stand.zip)

When printing from SD card, I noticed that the print is somehow compressed (it is definitely not what it should be). I would've expected one of two things:

1) either a warning that it can't print
2) to not print outside of bed bounds (so it would just ignore the parts of the skirt which are outside)

Here is the visualization of the gcode from Repetier:
[image]

And here is the picture of the bed (still printing):
[image]

As you can see from the skirt, the print was actually modified in some way.

Is there a setting which controls the scaling of the print in such situations?

### Steps to Reproduce

1. Slice a file which is bigger than bed size
1. Have it printed
1. Observe the scaling effect.

Sebastianv650, 2018-03-14:
Marlin will just not allow movements beyond the specified print bed size. Movements along other axis (Y, Z, E in your case) are executed ignoring the X part.

Marlin doesn't know the size of your print, so it can't issue a warning at print start. It also can't do scaling.

skyflyer, 2018-03-14:
Good to know. This actually explains the deformed skirt. Thanks for the explanation and keep up the good work!

closed 2018-03-14 by skyflyer