MarlinFirmware/Marlin #10105: MKS_SBASE board compiling issues

pixel3design-hub, 2018-03-14:
Can't compile using BOARD_MKS_SBASE
Found an error in **Marlin\src\HAL\HAL_LPC1768/HAL_LCD_pin_routines.c** at line 37:
_#include "../core/macros.h"_
changed into

_#include "../../core/macros.h"_
and started to compile, but suddently it throws an error about LCDish things (and I don't use ANY).

```
[03/14/18 18:15:48] Processing LPC1768 (platform: nxplpc; board: lpc1768)
 
Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: NXP LPC > NXP mbed LPC1768

SYSTEM: LPC1768 100MHz 64KB RAM (512KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(light)
Collected 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CMSIS-LPC1768> v0.0.0
|-- <U8glib-HAL> v0.4
|-- <TMC2130Stepper> v2.1.5
<-----
a lot of .o files
----->
Linking .pioenvs\LPC1768\firmware.elf
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::available()':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:202: multiple definition of `TwoWire::available()'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:202: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::read()':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:207: multiple definition of `TwoWire::read()'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:207: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::peek()':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:212: multiple definition of `TwoWire::peek()'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:212: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::write(unsigned char)':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:175: multiple definition of `TwoWire::write(unsigned char)'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:175: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::write(unsigned char const*, unsigned int)':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:191: multiple definition of `TwoWire::write(unsigned char const*, unsigned int)'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:191: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::TwoWire()':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:60: multiple definition of `TwoWire::TwoWire()'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:60: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::TwoWire()':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:60: multiple definition of `TwoWire::TwoWire()'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:60: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::begin()':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:65: multiple definition of `TwoWire::begin()'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:65: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::requestFrom(unsigned char, unsigned char)':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:113: multiple definition of `TwoWire::requestFrom(unsigned char, unsigned char)'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:113: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::requestFrom(int, int)':

C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:136: multiple definition of `TwoWire::requestFrom(int, int)'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:136: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::beginTransmission(unsigned char)':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:141: multiple definition of `TwoWire::beginTransmission(unsigned char)'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:141: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::beginTransmission(int)':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:150: multiple definition of `TwoWire::beginTransmission(int)'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:150: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/include/Wire.o: In function `TwoWire::endTransmission()':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\include/Wire.cpp:153: multiple definition of `TwoWire::endTransmission()'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/Wire.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/Wire.cpp:153: first defined here
[... 51 lines cut ...]
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_I2C_routines.o: In function `u8g_i2c_send_byte':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/HAL_LCD_I2C_routines.c:156: multiple definition of `u8g_i2c_send_byte'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_LCD_I2C_routines.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/HAL_LCD_I2C_routines.c:156: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_I2C_routines.o: In function `u8g_i2c_stop':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/HAL_LCD_I2C_routines.c:165: multiple definition of `u8g_i2c_stop'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_LCD_I2C_routines.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/HAL_LCD_I2C_routines.c:165: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_pin_routines.o: In function `pinMode_LCD':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/HAL_LCD_pin_routines.c:58: multiple definition of `pinMode_LCD'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_LCD_pin_routines.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/HAL_LCD_pin_routines.c:58: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_pin_routines.o: In function `u8g_SetPinOutput':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/HAL_LCD_pin_routines.c:86: multiple definition of `u8g_SetPinOutput'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_LCD_pin_routines.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/HAL_LCD_pin_routines.c:86: first defined here
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_pin_routines.o: In function `u8g_SetPinInput':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/HAL_LCD_pin_routines.c:90: multiple definition of `u8g_SetPinInput'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_LCD_pin_routines.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/HAL_LCD_pin_routines.c:90: first defined here
[... cut ...]
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_pin_routines.o: In function `u8g_GetPinLevel':
C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768\u8g/HAL_LCD_pin_routines.c:105: multiple definition of `u8g_GetPinLevel'
.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_LCD_pin_routines.o:C:\Users\sgswi\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_LPC1768/HAL_LCD_pin_routines.c:105: first defined here
collect2.exe: error: ld returned 1 exit status
```

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1812255/Configurations.zip)

Studiodyne, 2018-03-14:
Totaly not in your request , but  i have visited your profile and i have some questions if you're ok 

1-) Why a Marlin for CNC ... May be for inexpensive machine and software suite ? i have my own cnc , and i have a smoothstepper board & mach3 , and I'm really afraid to use anything else for these machines that can explodes a tool if a little error...
2-) Why Marlin and not Smoothie on mks sbase ...? 
3-) Do you like to use CNC like this way ,and what are the software you use for milling ?
4-) What toolpath generator ?
5-) Do you know if makerbase , will add trinamic drivers on their next SBASE ?

p3p, 2018-03-14:
@pixel3design-hub I can't reproduce this, clone (or download) a fresh copy of bugfix-2.0.x and apply your configs (no other modifications), LPC1768 build clean, LPC1768 build, compilation was successful.

Bob-the-Kuhn, 2018-03-14:
I wasn't able to reproduce your problem.

Your configuration files with the latest 2.0.x compiled on PlatformIO without any errors.

Suggestions:
1) Try the latest bugfix-2.0.x if you haven't done so already.
2) Delete the **.piolibdeps** directory.  That'll force PlatformIO to get the latest versions of the libraries and deletes the libraries that are not needed.  Either of those can cause compile problems.

pixel3design-hub, 2018-03-15:
Hi.
* I tried the Bob-the-Kuhn approach with no luck.
* I uninstalled Atom, PlatformIO and all the relative directories. No luck...

Always the same error.
I'll try now to reinstall everything.
The fun fact is that the same error is thrown by another pc in a freshly installed environment

Bob-the-Kuhn, 2018-03-15:
When I do the uninstall/re-install cycle I delete the following directories before doing the re-install:
C:\Users\YOUR_USER_NAME\.atom
C:\Users\YOUR_USER_NAME\.platformio
C:\Users\YOUR_USER_NAME\AppData\Local\atom
C:\Users\YOUR_USER_NAME\AppData\Roaming\Atom

Until I found the last two, the uninstall/re-install cycle often didn't fix the problem.

There's still another place where Atom/PlatformIO info is stored as it remembers where the last project folder was at.  As far as I can tell that's not a problem.

Bob-the-Kuhn, 2018-03-15:
How are you re-installing PlatformIO?  I use Atom's File -> Settings -> +Install and then search for **platformio** and select **platformio-ide**.

pixel3design-hub, 2018-03-15:
```
Marlin\src\HAL\HAL_LPC1768\HAL_LCD_pin_routines.c:37:10: fatal error: ../core/macros.h: No such file or directory
#include "../core/macros.h"
^~~~~~~~~~~~~~~~~~
compilation terminated.

Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\LPC1768_Servo.o
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_LCD_pin_routines.o] Error 1
 [ERROR] Took 2.99 seconds
 
 [SUMMARY]
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment LPC1768_debug_and_upload    [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
Environment LPC1768                     [ERROR]
 [ERROR] Took 2.99 seconds
```
This is thrown if I don't modify the **Marlin\src\HAL\HAL_LPC1768/HAL_LCD_pin_routines.c** at line 37

Once modified the error is the same as before.

Could someone tell to the community (and me) how to compile correctly (including the needed tools) this branch?
IMHO I believe that a well documented (even in early stage) project, including the compiling tasks, will drive the project to another level. I'm not a coder, just a maker with some knowledge of the most used languages but, maybe for the same reason, I can take a look at a project and I see the awesomeness of it.
------- bool OFF_TOPIC_MODE = true ----------
Studiodyne:
1. I'm an electronic engineer. And I'm a Maker. I'm not afraid to play with mechanics/electronics, and those electronics are quite cheap, the mechanical part is not so expensive too... So... Why not?
2. Because I found a bug in SMOOTHIEWARE only on MKSSBASE boards. I suspect they deliberately scrambled the steppers code. I got misalignment on lates smoothie fw just on MKS boards. Changed mechanics, motors and boards. Smoothieware board works flawlessly, Three MKS boards fails. Rolling back to an earlier version, and magically MKS works flawlessly again. I'm investigating...
3. Well, Yes and no... I've made my own board but as I previously said... I like to play. XD
I use Fusion360 as CAD and CAM. As GCode sender a little project of mine written in TypeScript and NodeJS.
4. Well... That's embarassing.... I already answered to this one... XD
5. Short/long answer: I don't know... But adding it to the board is quite easy. Last version of Smoothieware fw can connect to Trinamic 22xx and 26xx and is well documented on Smoothieware.org.
But my smoothieboard is connected to another 3D printer at the moment and I have any Trinamic (excluding the 2100 ones) to play with... Surely those drivers are groundbreaking in machines like the Prusa i3 MK3 where Joseph managet to get rid of endstops, make the machine reliable even on power loss or external intervention on the carriage. If MKS really wants to make a step ahead and win the rush for new functionalities they NEED to switch to TMC drivers. BUT I suspect they will stay dormant until Smoothieboard 2 release, to develop their knockoff.
------- bool OFF_TOPIC_MODE = false ----------

p3p, 2018-03-15:
The file `Marlin\src\HAL\HAL_LPC1768\HAL_LCD_pin_routines.c` does not exist in bugfix-2.0.x, you need to download or clone the current version.

pixel3design-hub, 2018-03-15:
Ok... I downloaded 5 times the SAME branch (bugfix-2.0.x).
Re-downloaded 6th time. Now it works. I'm shocked...

> Lazarus: “Everybody knows you never do a full retard.” 
> 
> Speedman: “What do you mean?” 
> 
> Lazarus: “Check it out. Dustin Hoffman, ‘Rain Man,’ look retarded, act retarded, not retarded. Count toothpicks to your cards. Autistic, sure. Not retarded. You know Tom Hanks, ‘Forrest Gump.’ Slow, yes. Retarded, maybe. Braces on his legs. But he charmed the pants off Nixon and won a ping-pong competition. That ain’t retarded. You went full retard, man. Never go full retard.”

Sure... I gone full retard... Thanks for all the help.
If someone need help with MKS SBASE, ask me.

Cheers my friends

Bob-the-Kuhn, 2018-03-15:
@pixel3design-hub - I resemble that remark ...

How were you doing your downloads?  If you were on the Marlin site and clicked the **Clone or download button** then it was just fumble fingers or downloading to one directory & working from a different one.

If you've forked the Marlin repository and are using the desktop application then I know what happened.  The local repository's copy of bugfix-2.0.x was out of date.  There's git commands I use  everytime I create a branch of 2.0.x to be sure I'm getting the latest.

closed 2018-03-15 by Bob-the-Kuhn

pixel3design-hub, 2018-03-15:
As I said... I went full retard... Tanks again

Joost-Maker, 2020-06-15:
I had a similar error, compiler couldn't find files. I think it hsa to do with the folder path lenght.

So, i copied the src dir to my hard-drive root (c:\) and ran the compiler, no issues appeard and the compilation was a success!