MarlinFirmware/Marlin #10109: [2.0.x] Marlin reboot after some time is executin a simple gcode

GMagician, 2018-03-14:
As requested by @AnHardt in issue #10078 I post my configuration.
What happens?

Using Repetier and executing attached gcode after sometime of correct execution, system reboot.
Trying to use lcd knob system reboot in short time (if something happens clicking the knob)

In my opinion is due to a very very intensive use of resources (Mega 2560) and some watchdog timeout.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1813339/Configuration.zip)

GMagician, 2018-03-14:
Forgot gcode
[gcode.zip](https://github.com/MarlinFirmware/Marlin/files/1813341/gcode.zip)

AnHardt, 2018-03-14:
The absolute maximum step frequency Marlin can handle is 40.000/s.
With 400 steps/mm the maximum handleble speed is 100mm/s = 6000mm/minute.
Your g-code orders 7600mm/minute but this is limited to 100mm/s by DEFAULT_MAX_FEEDRATE.
So that should be ok if there were not UBL and SEGMENT_LEVELED_MOVES enabled.
That divides your single ~269mm move into ~53 5mm moves.
So with 100mm/s you have to additionally plan ~20 moves/s.
I guess that's what gives the final overload.

The problems observed in https://github.com/MarlinFirmware/Marlin/issues/10078#issue-304350960 do make much more sense now knowing we don't have one single but a subevided move.

GMagician, 2018-03-15:
Eprom is in sync with firmware defaults (M502 M500 M501), no ubl active.
Test was to calibrate max velocity/acceleration supported by motors (45° move at 7600mm/min is almost 90mm/s on single axes so speed shouldn't be limited at all).

I follow you till single movement is split to 53.5mm moves . How you computed additional 20 moves/s?

closed 2018-03-15 by GMagician