MarlinFirmware/Marlin #10111: [FR] Dynamic current control with TMC2130 / TMC2208

8Roman8, 2018-03-15:
Is it possible to adjust the stepper current dynamicaly to prevent lost steps additional to hybrid threshold?
So this would be possible tu run higher/ lower currents than normal for short times.

This feature would keep the drivers and the motors cooler 
and would step up the current only when needed.

something like this:
#define *_CURRENT_RUN                 400      //mA
#define *_CURRENT_MAX                 1200   //mA
#define *_CURRENT_THRESHOLD     80      //mm per second

The function could adjust the current lineary from 80mm/s to #define DEFAULT_MAX_FEEDRATE.
I think this could make spreadCycle mode unnecessary.

teemuatlut, 2018-03-15:
Trinamic has this coolStep technology that uses motor load to determine appropriate current for the driver.
I'll be adding support for it soon but don't want to add new features before 1.1.9 is out.

If you want to try how it works there's an example sketch with the TMC2130 library.

Unfortunately it only works with spreadCycle, but the benefit is that we could use higher max currents as most of the time the driver would use lower actual currents.

8Roman8, 2018-03-16:
Maybe this feature could be used to increase current during de/acceleration or use spreadcycle only in this time.

I don't "like" the spreadcycle mode, because my whole housing (thin wood) of my machine resonate and is extremely loud ^^

If that's impossible please tell me and I will close the issue/ fr request.

teemuatlut, 2018-03-16:
That would require us to start analyzing the queue quite a bit more than we do now.
For your case I would suggest you either lower your acceleration/jerk settings or increase current as much as your cooling allows. The Watterott stepstick protectors allow you to put a second heatsink under the driver PCB, increasing your cooling capacity.
There are also other ways to deal with vibrations.

8Roman8, 2018-03-16:
Question to others:
Is there a interest on this or a similar feature? 

If not I will close this thread.

nightdemonx, 2018-03-17:
As you correctly siad @teemuatlut, your task is implemented in the coolstep feature. Do not reinvent the wheel.

8Roman8, 2018-03-17:
Sure, but coolstep uses the spreadcycle mode, the feature I requested would not needs the mode... 

But it seems I'm only one who wants the funcion and I close the request.

thinkyhead, 2018-03-17:
> But it seems I'm only one who wants the funcion and I close the request.

Either that or people have lives and haven't yet had a chance to respond to the request.

alfredanil, 2018-03-17:
digital current control like coolStep is a must have feature. It can be configured through the spi bus as well. There's yet another feature called dcStep which slows down the motor when it reaches motor torque limit. If it can synchronize the speed of all other drives accordingly then this could be really big advantage with using Trinamic driver tmc2130.

closed 2018-03-29 by 8Roman8

thinkyhead, 2018-03-29:
> which slows down the motor when it reaches motor torque limit

Due to the way Marlin works, we can only slow down all motors together, and we have to do it programmatically. If steppers decide to slow down on their own the results are unpredictable.

AmyTheCute, 2019-07-14:
I know this is super old but I'd love this! specially with my pancake stepper that gets SUPER hot.