MarlinFirmware/Marlin #10112: Compile error on Due

ReneHezser, 2018-03-15:
### Description
When choosing Arduino Due (programming port) as board and setting the board to
#define MOTHERBOARD RAMPS4DUE_EFB
the sketch cannot be compiled with Arduino beta.

### Steps to Reproduce
1. checkout bugfix-2.0.x
2. change board to RAMPS4DUE_EFB
3. select "Arduino Due (Programming Port)" in Arduino 1.9.0-beta
4. click on Verify in the IDE

**Expected behavior:**
The sketch compiles

**Actual behavior:**
The compiler shows errors:
```
In file included from sketch\src\HAL\HAL_DUE\../../inc/MarlinConfig.h:29:0,

                 from sketch\src\HAL\HAL_DUE\EepromEmulation_Due.cpp:35:

sketch\src\HAL\HAL_DUE\../../inc/../pins/pins.h:205:53: fatal error: pins_CREALITY_ENDER.h: No such file or directory

   #include "pins_CREALITY_ENDER.h"    // ATmega1284P

                                                     ^

compilation terminated.

exit status 1
Error compiling for board Arduino Due (Programming Port).
```

Bob-the-Kuhn, 2018-03-15:
Change the motherboard define and it'll compile.

**FROM**
#define MOTHERBOARD RAMPS4DUE_EFB

**TO**
#define MOTHERBOARD BOARD_RAMPS4DUE_EFB

closed 2018-03-15 by Bob-the-Kuhn

ReneHezser, 2018-03-15:
Ups. What a stupid mistake :-(
Thank you!
I was on the wrong track, because of the mentioned ENDER file, that was recently merged.