MarlinFirmware/Marlin #10114: TMC2130 dont work on megatronic3.1

ekozan, 2018-03-15:
### Description

I try to setup my driver but they dont work i dont know if it's a bug or not 

My driver is well configured 

[image]

### Steps to Reproduce

- add CS_PIN on megatronics 3 pin_config
   #define X_CS_PIN 48
   #define Y_CS_PIN 47
   #define Z_CS_PIN 46

- add on adv 
  #define HAVE_TMC2130
  #define X_IS_TMC2130
  #define Y_IS_TMC2130
  #define Z_IS_TMC2130

- dont enable 
  //#define TMC_USE_SW_SPI

wirering on AUX3 ( SS1 (48) SS2 (47) SS3 (46))+ SDOUT (HW  SCK (50) + MOSI (51) + MISO (52) ) 

send M122

**Expected behavior:** [What you expect to happen]
Got Driver adress on the end 
**Actual behavior:** [What actually happens]
Got 3 times no adress ( 0X0000000 )

teemuatlut, 2018-03-15:
Check your SPI pins.
SCK = D52
MOSI = D51
MISO = D50

ekozan, 2018-03-16:
it's speak but i lost my logic analyser for make a full debug >.< 

i check continuity from each atmega pin to my driver pin it's seem ok :/

ekozan, 2018-03-16:
I found my analyser 

They speak O.o'

[image]

```cvs
Time [s], Analyzer Name, Decoded Protocol Result
-0.000001500000000,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0110  1111
0.000004291666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
0.000008666666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
0.000013875000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  1111
0.000018250000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0001  1111
0.000032666666667,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0000  0000
0.000037666666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0001
0.000043625000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
0.000049916666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  1111  1111
0.000056250000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
0.514753583333333,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0010  0111
0.514771250000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
0.514775625000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0011
0.514780791666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
0.514785166666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
0.514799625000000,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0110  1111
0.514804625000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
0.514810541666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
0.514816875000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
0.514823166666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  1111  1111
1.014029000000000,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0110  1111
1.014034791666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.014039166666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.014044375000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.014048750000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.014063166666667,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0011  0111
1.014068166666667,SPI,MOSI: 0b  1000  0000;  MISO: 0b  1111  1111
1.014074125000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.014080458333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0011
1.014086750000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.513969916666667,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0110  1111
1.513975708333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.513980083333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.513985291666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.513989666666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.514004083333333,SPI,MOSI: 0b  0111  1111;  MISO: 0b  1111  1111
1.514009083333333,SPI,MOSI: 0b  1000  0000;  MISO: 0b  1111  1111
1.514015041666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
1.514021375000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0011
1.514027666666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.014838458333333,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0110  1111
2.014844250000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.014848625000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.014853833333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  1111
2.014858208333334,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0111  1111
2.014872625000000,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0000  0000
2.014877625000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.014883583333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.014889875000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  1111  1111
2.014896208333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.515774041666667,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0011  0111
2.515779875000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.515784250000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.515789458333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0111
2.515793833333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  1111  1111
2.515808250000000,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0100  1000
2.515813250000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.515819208333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.515825500000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
2.515831833333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
3.015594250000000,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0100  1000
3.015600083333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
[... 19 lines cut ...]
4.016411000000000,SPI,MOSI: 0b  1000  0000;  MISO: 0b  1011  1111
4.016415375000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  1111  1111
4.016420583333334,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
4.016424958333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
4.016439416666667,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0110  1111
4.016444416666666,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
4.016450333333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
4.016456666666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
4.016462958333333,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
4.516295208333333,SPI,MOSI: 0b  0110  1111;  MISO: 0b  1110  1111
4.516301041666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
4.516305416666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
4.516310625000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  1111
4.516315000000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0111  1111
4.516329416666666,SPI,MOSI: 0b  0110  1111;  MISO: 0b  0000  0000
4.516334416666667,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
4.516340375000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000
4.516346666666666,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0011  1111
4.516353000000000,SPI,MOSI: 0b  0000  0000;  MISO: 0b  0000  0000

```

but they still dont have any adress 

I need to found back my note on spi x')

teemuatlut, 2018-03-16:
Did you fix your pins yet?
According to you, you've wired SCK to D50. The correct pin is D52.
You've wired MISO to D52. The correct pin is D50.

ekozan, 2018-03-16:
Did you fix your pins yet? 
**It's a typo error 🥇  dyslexia always come back :D**

ekozan, 2018-03-17:
I maybe find the problème,do I need to init the spi connection by enabling SDcard support?

ekozan, 2018-03-20:
I'm stuck any idees ? 

I try to enable HW or SW SPI  

my SPI seam to work but M122 continue to write 0x00:00... 

I try 4 driver

teemuatlut, 2018-03-20:
Did you also open the SPI bridge as shown [here](http://marlinfw.org/docs/features/tmc_drivers.html#fysetc-drivers)?

You can also try running some of the example sketches that come with the library. But remember to take out the belt(s) first as the sketch will not respect your endstops.

ekozan, 2018-03-21:
ok it's totaly my bad ... my driver power fuse was blow so i think th tmc2130 driver was in "error mode" that's why i havent any driver register but they speak 

sorry

closed 2018-03-21 by ekozan