MarlinFirmware/Marlin #10115: [2.0.x] Bad lcd estruder 2 direction

GMagician, 2018-03-15:
With configuration attached if I move extruder 1 or 2 with repetier (using g0 or manual command) everything is ok.
When moving extruder from lcd extruder 1 move ok extruder 2 move in reverse direction (positive increments physically retract filament but using extrude velocity)
Since stepper is one for both extruders advance direction for E2 is reversed from E1

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1817368/Configuration.zip)

thinkyhead, 2018-03-16:
Both config files would be useful. I wouldn't have to ask:
- Have you already set `MINIMUM_STEPPER_PULSE` to 1 or 2?

GMagician, 2018-03-16:
@thinkyhead since I use mega2560 it should not be necessary to set 2 even if I should because of drv8825, but is not that the issue. Motor move correctly using gcode and lcd, only extruder 2 direction is reversed using lcd (gcode direction is ok)

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1818352/Configuration.zip)

.

thinkyhead, 2018-03-16:
Indulge me.

thinkyhead, 2018-03-16:
Try this change to `Conditionals_LCD.h`:
```diff
#if ENABLED(SWITCHING_EXTRUDER)                               // One stepper for every two EXTRUDERS
  #if EXTRUDERS > 4
    #define E_STEPPERS    3
    #define E_MANUAL      3
  #elif EXTRUDERS > 2
    #define E_STEPPERS    2
    #define E_MANUAL      2
  #else
    #define E_STEPPERS    1
  #endif
- #define E_MANUAL        E_STEPPERS
+ #define E_MANUAL        EXTRUDERS
#elif ENABLED(MIXING_EXTRUDER)
  #define E_STEPPERS      MIXING_STEPPERS
  #define E_MANUAL        1
#else
  #define E_STEPPERS      EXTRUDERS
  #define E_MANUAL        EXTRUDERS
#endif
```

GMagician, 2018-03-16:
I'll try it as soon as I'm back at home, this evening, thanks

GMagician, 2018-03-16:
@thinkyhead it solves issue. Thanks again

closed 2018-03-16 by GMagician