MarlinFirmware/Marlin #10116: Latest Marlin broken for DUE RADDS

montycb, 2018-03-15:
Hi,

Im using a DUE with a RADDS Board and a RepRap Full Graphics Controller, In the latest bugfix-2.0 Version the serial connection is no longer working.

How to reproduce:

Get the default configuration and set the following settings:
#define MOTHERBOARD BOARD_RADDS

Compile and upload that:
Everything will work normally, except for the serial connection, you can see that Marlin is starting, and it'll send you temperatures, but you can't send any GCodes to it.

Enabling the display with:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

makes this even worse, marlin will get stuck in a bootloop because of a watchdog reset

I know the board is working, because everything is working fine with an older Marlin version.

ZAVHome, 2018-03-16:
Similar symptoms on a bundle of DUE + RURAMPS

ejtagle, 2018-03-16:
I`ve been experiencing the same thing lately... I don´t know if it is a boot loop, a hw problem or another thing...

Bob-the-Kuhn, 2018-03-16:
I just did RAMPS_FDV2 on the programming port and it works fine.

@ejtagle - what board are you running?

ejtagle, 2018-03-16:
I am running an Arduino DUE with a RAMPSv1.4 modded to work with it. It is strange: I don't know if it is a bootloop or a hw bug (self reset). When i try to debug it, it works...

ejtagle, 2018-03-16:
First, we (i) have to find out if it is a watchdog problem, or a self reset problem....

Bob-the-Kuhn, 2018-03-16:
I tried the BOARD_RADDS setting with the full graphics enabled and the host port connected.  Repetier Host connected normally and accepted a few G0 command.

Bob-the-Kuhn, 2018-03-16:
> When i try to debug it, it works...

FRUIT!!!  A hard to reproduce issue.  ☹️

ejtagle, 2018-03-16:
Yes... When i power the board, it usually gets stuck in the Marlin logo. 
But if i power it up using the programming port, then it usually works (but not always) ...
In fact, i have a Jlink with the proper adapter for the DUE (for other projects) . I think i will try to get it going and find out what is happening.

If it is a hard reset, there is a possible workaround.
If it is a bug in Marlin code, then we shoukd fix it.

I did not do the hw modification for the reset/clear pins. I do suspect something strange, as when the board is stuck, the Bossa bootloader port appears. Even if it was a watchdog reset, Marlin should reboot.  Even if it was a Hardware reset pulse, Marlin should reboot. The bossa bootloader should never appear, as we programmed the FLASH configuration bits to map the flash instead of the bossa bootloader.

Bob-the-Kuhn, 2018-03-16:
Try enabling/disabling the SD card.  Accessing the SD card usually happens between the boot screen and the status screen.

What do you use for a debugging environment with the J-Link?

I'm using Segger's Ozone.  The one that comes with PlatformIO is just a DOS window.

Bob-the-Kuhn, 2018-03-16:
Here's a link to [bugfix-2.0.x from 2018-03-10](https://www.dropbox.com/s/xhqwr32mzx5e57a/Marlin-bugfix-2.0.x_2018-03-10.zip?dl=0).  

If it works solidly then we can do some playing to see where the difficulties come in.

ejtagle, 2018-03-16:
I'll try it. I´ll try either segger or AVR Studio. Both can run Arduino DUE projects

Bob-the-Kuhn, 2018-03-16:
@montycb , @ZAVHome  - please try the 10 MAR version of bugfix & let us know if it is reliable.

With a problem like this we want as much confidence as possible that we're starting from a known good place.

montycb, 2018-03-16:
Just tested the version you linked in and it doesn't work for me.
It still gets a few watchdog resets while turning on but eventually starts up.
I can connect without any resets, but Marlin still doesn't answer any gcodes

The last version (which worked perfectly fine) i used was 1f8f8c9104f8028f9f94398a232dd1f1b55e5c9b
so something after this commit must've broken it.

Bob-the-Kuhn, 2018-03-16:
That PR was merged 28 days ago.

I'll see if I can get the DUE files that have changed since then.

Bob-the-Kuhn, 2018-03-16:
Here's some snapshots between 16 FEB and 10 MAR.  Please see if any work.

[22 FEB](https://www.dropbox.com/s/czn71h0p4s447y1/Marlin-2.0.x_2018-02-22_90fa423737448feb75ee59a2afc88a8c891b19ea.zip?dl=0)

[25 FEB](https://www.dropbox.com/s/lyze5rbu1gcdocy/Marlin-2.0.x_2018-02-25_61688233c7e03a0ada5c599f05aaa2fcf1d8f49d.zip?dl=0)

[3 MAR](https://www.dropbox.com/s/ggbi0ggubpvipa9/Marlin-2.0.x_2018-03-03_649f3754945390f55e8a14551560271badd26352.zip?dl=0)

montycb, 2018-03-16:
Ok, I'm done testing

22FEB and 25FEB:
no bootloop and fully functional serial connection
Everything fine as far as i can tell

3MAR 
no bootloop, but serial connection won't respond to gcodes

Bob-the-Kuhn, 2018-03-16:
Please report the RAM usage on the 25 FEB one and the 3 MAR one.

Another user is reporting strange things on an A8.  Reducing the planner's BLOCK_BUFFER_SIZE from 128 to 16 helped in his case which implies he ran out of RAM.

ZAVHome, 2018-03-16:
03 MAR
156456 byte (29%) all memory 524288 byte.

ZAVHome, 2018-03-16:
25 FEB 156456 (29%) 524288 byte

Bob-the-Kuhn, 2018-03-16:
That can't be right.  There's only 32K of RAM and 512K of FLASH.

ZAVHome, 2018-03-16:
> That can't be right. There's only 32K of RAM and 512K of FLASH.
How i'm can see this stat ?

_25FEB:
no bootloop and fully functional serial connection
Everything fine as far as i can tell_
+1

_3MAR
no bootloop, but serial connection won't respond to gcodes_
+1

montycb, 2018-03-16:
I'm getting similar values

25FEB:
Sketch uses 126036 bytes (24%) of program storage space. Maximum is 524288 bytes.

3MAR:
Sketch uses 126196 bytes (24%) of program storage space. Maximum is 524288 bytes.

Bob-the-Kuhn, 2018-03-16:
With just RAMPS_FD_V2 enabled, PlatformIO is reporting 103320 bytes of FLASH and 7792 bytes of RAM used.
```
Checking program size
text       data     bss     dec     hex filename
103320        0    7792  111112   1b208 .pioenvs\DUE_USB\firmware.elf
```

If I add #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER I get 131964 & 8656.
```
text       data     bss     dec     hex filename
131964        0    8656  140620   2254c .pioenvs\DUE_USB\firmware.elf
```

Adrduino 1.9 beta native port doesn't report RAM usage.

Bob-the-Kuhn, 2018-03-17:
I used the GitHub desktop app to do a diff between the 25 FEB and 3 MAR versions.

There were a lot of changes to gcodes and motion files to support scara and sensorless homing.

There were DUE timer & serial changes.  The timer changes were to add the tone option.

There were changes to the LCD software.

I didn't mention any difs associated with options that weren't enabled by default.

I've tried to move these changes into the 25 FEB version in the hopes that one of them will result in the problems.

Please start with this one.  Hopefully this is the same as the 25 FEB version.  We need to make sure it works on your systems before trying the others.
[25 FEB base.zip](https://www.dropbox.com/s/f4u9511s084xpca/25%20FEB%20base.zip?dl=0)

Then try this one.  If it doesn't fail then there's no reason to try the others.
[25 FEB with Gcode + motion + Due changes + LCD.zip](https://www.dropbox.com/s/boyqulh8kpn027q/25%20FEB%20with%20Gcode%20%2B%20motion%20%2B%20Due%20changes%20%2B%20LCD.zip?dl=0)

Then try these as you like.
[25 FEB with DUE changes only.zip](https://www.dropbox.com/s/iyzly975d01te6p/25%20FEB%20with%20DUE%20changes%20only.zip?dl=0)
[25 FEB with Gcode changes only.zip](https://www.dropbox.com/s/3mxcuqvuggoutz6/25%20FEB%20with%20Gcode%20changes%20only.zip?dl=0)
[25 FEB with Gcode + motion changes.zip](https://www.dropbox.com/s/y0wq6hw2dv4ubb2/25%20FEB%20with%20Gcode%20%2B%20motion%20changes.zip?dl=0)
[25 FEB with Gcode + motion + Due changes.zip](https://www.dropbox.com/s/63w7osodnqtm4qw/25%20FEB%20with%20Gcode%20%2B%20motion%20%2B%20Due%20changes.zip?dl=0)

I've moved the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER option to right after the motherboard selection to make life a little easier.

montycb, 2018-03-17:
Ok, here are my test results:

25 FEB base.zip:
```
text	   data	    bss	    dec	    hex	filename
132080	      0	   9048	 141128	  22748	.pioenvs/DUE/firmware.elf
```
no bootloop, fully functional serial connection

25 FEB with Gcode + motion + Due changes + LCD.zip:
```
text	   data	    bss	    dec	    hex	filename
132232	      0	   9056	 141288	  227e8	.pioenvs/DUE/firmware.elf
```
no bootloop, serial connection won't respond to gcodes

25 FEB with DUE changes only.zip:
```
text	   data	    bss	    dec	    hex	filename
132080	      0	   9048	 141128	  22748	.pioenvs/DUE/firmware.elf
```
no bootloop, fully functional serial connection

25 FEB with Gcode changes only.zip:
```
text	   data	    bss	    dec	    hex	filename
132112	      0	   9048	 141160	  22768	.pioenvs/DUE/firmware.elf
```
no bootloop, fully functional serial connection

25 FEB with Gcode + motion changes.zip:
```
text	   data	    bss	    dec	    hex	filename
132112	      0	   9048	 141160	  22768	.pioenvs/DUE/firmware.elf
```
no bootloop, fully functional serial connection

25 FEB with Gcode + motion + Due changes.zip:
```
text	   data	    bss	    dec	    hex	filename
132208	      0	   9056	 141264	  227d0	.pioenvs/DUE/firmware.elf
```
no bootloop, serial connection won't respond to gcodes

ejtagle, 2018-03-17:
humm... please, look at your configuration.h , and make sure SERIAL_PORT is set to -1  (that means to use the native port as COM port for commands. Or, if you want to use the programming port a default COM port

ejtagle, 2018-03-17:
set it to 0

montycb, 2018-03-17:
No, that is not the problem, the serial port is configured correctly.
Marlin will send me the usual startup messages, but it simply won't respond to gcodes.
If the wrong serial port was selected, i wouldn't get any messages i guess

ejtagle, 2018-03-17:
you are right. It is strange. More like Marlin is not parsing the received commands

Bob-the-Kuhn, 2018-03-17:
Dear sirs: Please provide a problem that is easier to troubleshoot.  This one is giving me a royal headache.

Bob-the-Kuhn, 2018-03-17:
Could you provide a file with a gcode sequence that fails along with the failure symptoms?

Bob-the-Kuhn, 2018-03-17:
I'll create a gcode + DUE and  a motion + DUE to see if we can narrow it down further.

montycb, 2018-03-17:
Well, none gcode i tried works i fear...
no matter if it's movement, heating, homeing or status, Marlin ignores everything.

montycb, 2018-03-17:
I just noticed something, i compiled 25 FEB with Gcode + motion + Due changes.zip with EEPROM Settings, which gives me the option to load, store and reset the eeprom from the controller, when i do that, i get an confirmation message on my serial connection.
I'm not sure if this helps, but it seems the serial connection is working one way.

Bob-the-Kuhn, 2018-03-17:
I've been using G1 on the X axis only.  It's always worked.  

Please give me a specific example and what does or doesn't happen.

Bob-the-Kuhn, 2018-03-17:
In case you're wondering - here's the file collections that I've been swapping in and out.  These are the 3 MAR files I've been dropping into 25 FEB.

[motion.zip](https://github.com/MarlinFirmware/Marlin/files/1821899/motion.zip)
[ultralcd.zip](https://github.com/MarlinFirmware/Marlin/files/1821900/ultralcd.zip)
[DUE changes.zip](https://github.com/MarlinFirmware/Marlin/files/1821901/DUE.changes.zip)
[gcode changes.zip](https://github.com/MarlinFirmware/Marlin/files/1821902/gcode.changes.zip)

ZAVHome, 2018-03-17:
> Please give me a specific example and what does or doesn't happen.

Any Gcode that is sent via USB does not reach the board.
But REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER normally works and controls the printer.

Bob-the-Kuhn, 2018-03-17:
To be 200% sure, please try **G1 X10 F4800**

When it fails, what does M114 report?

montycb, 2018-03-17:
G1: doesn't move
M114: returns nothing

montycb, 2018-03-17:
25-FEB-with-motion-+-DUE.zip:
doesn't work

25 FEB with motion only.zip:
works!

25 FEB with Gcode + DUE changes.zip:
compilation fails with 
```.pioenvs/DUE/FrameworkArduinoVariant/variant.cpp' not found, needed by target...```

ZAVHome, 2018-03-17:
> G1: doesn't move
M114: returns nothing

+1

Bob-the-Kuhn, 2018-03-17:
> .pioenvs/DUE/FrameworkArduinoVariant/variant.cpp' not found, needed by target...

Delete **.pioenvs** & try again.

Bob-the-Kuhn, 2018-03-17:
Maybe it's a library issue.  Try deleting **.piolibdeps** and see if **25-FEB-with-motion-+-DUE.zip** still fails.

montycb, 2018-03-17:
No, still doesn't work.

But i ran a diff between those other two versions and the only big difference is tone and timers, reverting that fixes it, so it is maybe related to that

Bob-the-Kuhn, 2018-03-17:
Sounds like it's ignoring the received characters.

Please post the host logs when it fails and when it works.

montycb, 2018-03-17:
I managed to compile 25 FEB with Gcode + DUE changes.zip, apparently platformio didn't like the folder name, but that version doesn't work for me

Here is a broken version:
```
echo:PowerUp
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Mar 17 2018
echo: Free Memory: 85924  PlannerBufferBytes: 1344
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
>>> M114
SENDING:M114
>>> M115
SENDING:M115
>>> G1 X10 F4800
SENDING:G1 X10 F4800
>>> M114
SENDING:M114
```

Here is what it should look like:
```
echo:PowerUp
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Mar 17 2018
echo: Free Memory: 85932  PlannerBufferBytes: 1344
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
>>> M115
SENDING:M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:EEPROM:0
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
>>> G1 X10 F4800
SENDING:G1 X10 F4800
>>> M114
SENDING:M114
X:10.00 Y:0.00 Z:0.00 E:0.00 Count X:800 Y:0 Z:0
```

Bob-the-Kuhn, 2018-03-17:
Time to create some code to look at the serial traffic.

Bob-the-Kuhn, 2018-03-17:
Still working on the code.

Here's another variation to try.  I doubt it makes a difference.
[25 FEB with motion + DUE - conditional.zip](https://www.dropbox.com/s/fi9k1zv9ur6ridu/25%20FEB%20with%20motion%20%2B%20DUE%20-%20conditional.zip?dl=0)

montycb, 2018-03-17:
tested, doesn't work

Bob-the-Kuhn, 2018-03-17:
Here's a parser.cpp that echoes the input line before it gets parsed.  That'll tell us what Marlin is receiving.

Please drop it into a failing one & post the results.

[parser.zip](https://github.com/MarlinFirmware/Marlin/files/1821997/parser.zip)

montycb, 2018-03-17:
Ok, i added that file to 25 FEB with Gcode + DUE changes.zip
It doesn't make a difference, marlin doesn't answer anything
```
>>> M114
SENDING:M114
>>> M115
SENDING:M115
>>> G1 X10 F4800
SENDING:G1 X10 F4800
>>> M114
SENDING:M114
```

Bob-the-Kuhn, 2018-03-17:
This guy enables the XMIT & RCV portions of SERIAL 0

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1822116/Marlin.zip)

montycb, 2018-03-17:
Unfortunately that didn't help, but i noticed that i got the following line:
```ECHO INPUT LINE:  M105```
However i can't reproduce that, it just randomly appeared, once.
Maybe the serial port is only working for very short period of time after boot.

Bob-the-Kuhn, 2018-03-17:
This one re-initializes SERIAL 0 at the very end of the startup.

[Marlin re-init.zip](https://github.com/MarlinFirmware/Marlin/files/1822146/Marlin.re-init.zip)

montycb, 2018-03-17:
that made no difference, i got that echo input line once more though, but i still can't reproduce that

montycb, 2018-03-17:
Ok, im fairly sure there is something wrong in toneInit()
commenting that out in Marin.cpp gives me a fully functional serial connection

thinkyhead, 2018-03-17:
Maybe I picked an inappropriate timer for `toneInit`? You can also set the `BEEPER_PIN` to -1 and then `toneInit` will be skipped.

Bob-the-Kuhn, 2018-03-17:
I haven't figured out where the tone ISR is but maybe lowering it's priority will help.

[HAL_timers_Due.zip](https://github.com/MarlinFirmware/Marlin/files/1822201/HAL_timers_Due.zip)

thinkyhead, 2018-03-17:
It runs with ver a low rate, not sure about its priority.
```cpp
void toneInit() {
  HAL_timer_start(TONE_TIMER_NUM, 1); // Lowest frequency possible
}
```

thinkyhead, 2018-03-18:
At the moment I'm unable to get Marlin running on a bare DUE (using `BOARD_RAMPS_FD_V2`) without watchdog resets. I'll keep messing with it.

[77 more replies]

closed 2018-03-23 by Bob-the-Kuhn