MarlinFirmware/Marlin #10117: UBL Compensation Not Applying While Printing

the1snm, 2018-03-15:
First off thanks to all on this project who have helped transform my first 3D printer a cheap A8 into a decently solid machine. I recently moved into using UBL leveling on my machine from linear to bilinear to the UBL. It's been quite an adventure trying to learn all that goes into setting up right and I *think* I'm there. I originally tried UBL with 1.1.8 but had some weird behaviors and decided to go with the current bug fix version instead. It took several days to get the mesh to finish to a point I was happy, with many many head crashes early on. So I've got (in my opinion lol) a pretty solid leveling and mesh while the G26 test print came out pretty well except for a lot of stringing. I'm printing a 'small' 4.5hr starter print for my first UBL print. Something I notice though is as the print is moving around 50 x 50mm item the Z motors don't seem to move at all i.e. compensating. When I used the previous living systems I could easily see the compensation happening on the Z motors. I don't know if this is a problem or not visible simply because the bed is rather level or not. So I thought I'd ask the experts after reading posts for hours ant not seeing a similar scenario.

Here is my latest mesh...
Recv: X:79.00 Y:117.00 Z:4.94 E:108.19 Count X:7900 Y:11700 Z:1974
Recv: ok
[...]
Send: M18
Recv: ok
[...]
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,9)                                                                   (9,9)
Recv: (10,210)                                                                (210,210)
Recv:   0.152    0.050    0.085    0.050   -0.005   -0.105   -0.160   -0.172   -0.207   -0.197
Recv: 
Recv:   0.118    0.052    0.088    0.032    0.005   -0.097   -0.142   -0.195   -0.265   -0.355
Recv: 
Recv:   0.137    0.063    0.085    0.043    0.037   -0.073   -0.085   -0.110   -0.165   -0.210
Recv: 
Recv:   0.132    0.052    0.085    0.022    0.012   -0.067   -0.065   -0.083   -0.100   -0.085
Recv: 
Recv:   0.145    0.067    0.090  [ 0.060]   0.017   -0.060   -0.063   -0.065   -0.075   -0.073
Recv: 
Recv:   0.142    0.067    0.085    0.048    0.007   -0.050   -0.060   -0.063   -0.088   -0.065
Recv: 
Recv:   0.130    0.043    0.057    0.032   -0.003   -0.067   -0.073   -0.065   -0.075   -0.048
Recv: 
Recv:   0.093    0.030    0.060    0.037    0.022   -0.045   -0.107   -0.102   -0.107   -0.075
Recv: 
Recv:   0.043   -0.025    0.015    0.005   -0.040   -0.115   -0.123   -0.125   -0.105   -0.095
Recv: 
Recv:  -0.080   -0.070   -0.017   -0.017   -0.070   -0.155   -0.102   -0.080   -0.110   -0.065
Recv: (10,10)                                                                    (210,10)
Recv: (0,0)                                                                     (9,0)
Recv: ok

Here is what my G26 print looked like as well.
[image]

Lastly here are my config files for reference or for those who have a similar setup and need some help.

[Cofigurations.zip](https://github.com/MarlinFirmware/Marlin/files/1817396/Cofigurations.zip)

Thanks for any feedback.
Steve

thinkyhead, 2018-03-15:
I see you have `RESTORE_LEVELING_AFTER_G28` enabled. Note that this will only re-enable leveling if it was already on _before_ `G28`. Therefore your starting G-code should always contain `M420 S1` at some point to enable leveling. If this command is missing there's no guarantee leveling will be enabled at the start of your print.

the1snm, 2018-03-16:
@thinkyhead Thanks for the reply, it's funny you mention that item because when I compiled 1.1.8 with UBL all went essentially fine. When I pulled down the bugfix and copied my config files into that direction and tried to recompile again it threw an error saying I needed that in order to compile. I assumed it was a new feature since it did not ask for it or error on it previously. So whether or not I have this enabled I should make sure my Gcode contains the M420 S1 in it then.

thinkyhead, 2018-03-18:
> So whether or not I have this enabled I should make sure my Gcode contains the M420 S1 in it then.

Yes. It only preserves the previous state of leveling. It doesn't force it to be on with all `G28`.

Roxy-3D, 2018-03-18:
>> So whether or not I have this enabled I should make sure my Gcode contains the M420 S1 in it then.

> Yes. It only preserves the previous state of leveling. It doesn't force it to be on with all G28

UBL defaults to the older behaviour of being disabled after a G28.   But if you do things to delibertly have it enabled all the time, it will respect that.   If you load your mesh, do a G29 A, and follow it with a M500, UBL will be active all the time.   If you do those things...  UBL will be alive and working when you power up the printer.    I don't even have a G29 in my Start Up GCode for my slicer.   The printer powers up and is doing the 'right' thing.

G28 still needs to turn it off to do its work.  But G28 will respect the bed leveling activation state and restore it at the end if you are running UBL.

thinkyhead, 2018-03-19:
> If you load your mesh, do a G29 A, and follow it with a M500, UBL will be active all the time

True! So watch out for this one little side-effect: Before `G28` Marlin will do Z correction as if the nozzle is at the home position. Marlin insists on a `G28` before proceeding with some procedures, which helps, but if you enable `NO_MOTION_BEFORE_HOMING` you can be assured of the best behavior.

the1snm, 2018-03-19:
Thanks for the help, I'll check these out and sounds like it should take care of my concerns.

closed 2018-03-19 by the1snm

Roxy-3D, 2018-03-19:
When you check it out...   Make sure the nozzle is only a few mm's above the build plate.   If the nozzle is too high, you won't get any correction because you are above the 'Fade Height'.