MarlinFirmware/Marlin #10122: [2.0.x] DUE - no host support on native USB if SD card not enabled

Bob-the-Kuhn, 2018-03-16:
The host can not connect on the native USB port if SDSUPPORT is disabled.

Repetier Host reports "port doesn't exist" when you click connect button.

Steps to reproduce:
1. start with default configuration
2. set SERIAL_PORT to -1

ejtagle, 2018-03-16:
It works for me, BUT... be careful!... Windows changes the assigned COM port to the printer if the configuration of marlin USB changes. Mine changed from COM10 to COM8. Probably that is also your problem

Bob-the-Kuhn, 2018-03-16:
Not for me.  Isn't Windows wonderful !!!

Port 9 is present per a couple of checks but when I go to connect it says "port doesn't exist".

I found out it isn't enabling SDSUPPORT that's the culprit.  Enabling an LCD also gets it to work.

ejtagle, 2018-03-17:
very, very strange...  the usb stack is fully interrupt driven. So sdsupport or lcd makes no difference. Initially, i thought maybe running a composite device with no MSD could be the problem, but i actually tested it and the COM port still works. I tried just a serial terminal program and i was able to send commands/receive responses with no issue at all. It could be interestimg to know under what conditions repetier host shows such message. ..

Bob-the-Kuhn, 2018-03-17:
Here's what Pronterface says when I try to connect.
```
[ERROR] Could not connect to COM9 at baudrate 250000:
Serial error: could not open port COM9: [Error 2] The system cannot find the file specified.
```

and
```
Connecting...
Traceback (most recent call last):
  File "printrun\pronterface.pyc", line 1053, in connect
  File "printrun\pronsole.pyc", line 720, in connect_to_printer
  File "printrun\printcore.pyc", line 46, in inner
  File "printrun\printcore.pyc", line 197, in connect
  File "serial\serialwin32.pyc", line 31, in __init__
  File "serial\serialutil.pyc", line 261, in __init__
  File "serial\serialwin32.pyc", line 71, in open
  File "serial\serialwin32.pyc", line 186, in _reconfigurePort
ValueError: Cannot configure port, some setting was wrong. Original message: [Error 2] The system cannot find the file specified.
```

How do I find the USB interrupt?  Maybe a little call stack tracing will show something.

Bob-the-Kuhn, 2018-03-17:
I've found out that if you wait 2-5 minutes after power up that the host will connect.

I have some USB protocol captures of works and fails if that's of interest to you.  Lots of traffic in both.

ejtagle, 2018-03-17:
hummm... that is very strange.... i'll take a look...

ejtagle, 2018-03-18:
Confirmed! 👍  The problem is that when the SDSUPPORT is disabled, we export an MSD interface with no units. Windows does not like that, and tries to access it nevertheless, locking the whole USB interface (including the USB CDC endpoint!).
I will do a smallish modification, and if no SDSUPPORT, then will only export the CDC as a simple (not composite) device...

Bob-the-Kuhn, 2018-03-19:
PR #10148 fixed this.

closed 2018-03-19 by Bob-the-Kuhn