MarlinFirmware/Marlin #10126: i cant get my auto lvel sesnsor to i get g28 z forbidden

Bishopeuk, 2018-03-16:
I get g28 z forbidden could someone please help me out when I send m119 I get the following

x_min: open
y_min: open
z_min: open
z_max: TRIGGERED

GMagician, 2018-03-16:
z_max triggered is the problem... but do you really have it?
Your cnf could help

Bishopeuk, 2018-03-17:
would you like me to post it what is it i am looking for

thinkyhead, 2018-03-17:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop them into your next reply. We'll check them over and see if anything is amiss.

Bishopeuk, 2018-03-17:
[configh-ad.zip](https://github.com/MarlinFirmware/Marlin/files/1821565/configh-ad.zip)
files as request thanks in advance for your asistance

Bishopeuk, 2018-03-17:
side not the carridge exceeds the bed when running auto level by build plate is 220 by 220

GMagician, 2018-03-17:
USE_ZMAX_PLUG is defined, sure you have limit switch for max pos?
Since I see all min switch have ?_MIN_ENDSTOP_INVERTING to true and Z_MAX_ENDSTOP_INVERTING is not I really think you have no Z max limit switch

GMagician, 2018-03-17:
Finally sure you have done G28 X and Y before Z? you have Z_SAFE_HOMING activated and Z homing requires X and Y before (Z home is made in bed center)
What is your marlin version? Config file seems not to up to date with 1.1.8

Bishopeuk, 2018-03-17:
I did use a an switch but ive been trying to use the original tronxy sensor that came with it because after upgrading to e3d v6 setup the switch is to low for the bed so I thought the sensor should solve that issue

Bishopeuk, 2018-03-17:
its an old ver v.1.1.0 I would like to update but I cant work how to change the newer code to suit my printer thanks for your assistance again I'm fariley certain g28 is trigger cos nozel moves to center of bed but when I try to level bed it does not stop and when I try print even though the g29 is in the start code it does not attempt level the bed before the print

GMagician, 2018-03-17:
Sorry but maybe I misunderstood you. You talk about G28 Z forbidden and then talk about bed level (that is G29)
What is the commands you send when you get error?

Bishopeuk, 2018-03-17:
let me try and explain my prob if I home all it put carridge in center of bed but when I try home z I get the forbidden msg and when I try print a job the does not trigger and rams the despite the g29 code in the starting code after the g28 and it does not attempt to level the bed

GMagician, 2018-03-17:
Since you have min and max swtich on Z do something like this:
1) M119 with z_max unpressed
2) M119 with z_max pressed
I would like to see z max switch is working

Bishopeuk, 2018-03-17:
Nope when i got the printer i had bother getting probe to work correctly and was told to switch to ramps from.melzi and replace sensor with switch but after upgradind to e3d it don't work even if i drive to screw it still truggers to high so I want get the sensor going on it, but the it triggered both min and max are triggered I'm impressed just the max what's needs ammending ?

GMagician, 2018-03-17:
I think I not correctly understood what you wrote but when switch are not engaged (pressed) you need to read 'open', not 'triggered'. If you try to home with micro engaged you may have some problems (depending on which one is pressed min or max).
Using M119 in both situation (pressed and released) you are sure that switches are working. If max is working and you get 'triggered' when released you have to set Z_MAX_ENDSTOP_INVERTING to true

Bishopeuk, 2018-03-17:
if u it make it any clearer I can send a pic of it in both pressed and depressed

GMagician, 2018-03-17:
I'll try...

This is standard situation
1) keep all switches unpressed and execute M119, you should have all micro 'open'
2) press and keep pressed all switches and execute M199, you should have all micro 'triggered')
    you can do this also micro by micro to see that the one you pressed will change status
Your situation
1) keep all switches unpressed and execute M119, you should have all micro 'open' but z max that it will say 'triggered'
2) press and keep pressed z max switch and execute M199, you should have micro 'open'

If you get this then set Z_MAX_ENDSTOP_INVERTING to true in your cnf, reflash firmware and try again, now you should be in 'standard situation' and your issue should be vanished....'should'

Bishopeuk, 2018-03-17:
ok but the problem is not with the switch I'm trying to get the proximity sensor to work instead of the switch as the switch is no good for my e3d upgrade now

  3.  keep all switches unpressed and execute M119, you should have all micro 'open' but z max that it will say 'triggered'
  4.  press and keep pressed z max switch and execute M199, you should have micro 'open'

GMagician, 2018-03-17:
ok but if you get z_max triggered this may be a problem homing axes

GMagician, 2018-03-17:
for homing usually you use z_min, but since you have configured also z_max you need to fix it as well

Bishopeuk, 2018-03-17:
but what the default settings for sensor though the guy I code the firmware off only uses switch as he gave up on the sensor are you say I only use on or the other for the sensor?

GMagician, 2018-03-17:
Well I'm getting lost...
You have a ramp, you have configured firmware to use z_max and z_min limit switches (USE_ZMIN_PLUG and USE_ZMAX_PLUG) and that can be ok (but it's not normal that you have z_max triggered when you are not over z_max, and this is something wrong)
On cartesian printer usually z_min input/limit switch is also used as a Z proximity for UBL (and you are not doing that because you have Z_MIN_PROBE_ENDSTOP defined)

In your configuration probe is connected to Z_MIN_PROBE_PIN and in ramp by default is pin 32.
then you have to connect your proximity there.

But again why z_max report triggered? If triggered Z can move up

Bishopeuk, 2018-03-17:
well I have just ran through the level bed setup it still rams into the build plate

Bishopeuk, 2018-03-17:
is there a percific order the wires for sensor have to go I just copied them out of the original plug the colours I have brown blue and black

________________________________
From: David Gulliver <bishopeuk@msn.com>
Sent: 17 March 2018 18:31
To: GMagician; MarlinFirmware/Marlin; MarlinFirmware/Marlin
Cc: Author
Subject: Re: [MarlinFirmware/Marlin] i cant get my auto lvel sesnsor to i get g28 z forbidden (#10126)

well I have just ran through the level bed setup it still rams into the build plate

Bishopeuk, 2018-03-20:
hi with the nozel away from bed and bed off switch and x carridge not pressed if I put place metal object under sensor all but the z m in or normal and when I try running bed level procedure the nozel does not top and I try to print something when the nozel travels left in centre of bed it raased away from can u please hlp me out

  *   Please put the nozzle in the center of the bed 50mm above the build plate.
  *   Do a M119. If anything is showing up as triggered.... Work to resolve it.
  *   Once that is done, and nothing shows up as triggered, Press and Hold each end stop as you do a M119 Each one should be reported as Triggered.

thinkyhead, 2018-03-21:
Since we are very busy, you should also post your question at:
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)

Bishopeuk, 2018-03-21:
Tried those groups are closed

thinkyhead, 2018-03-29:
> Tried those groups are closed

Well, technically so is this Issue Queue when you don't have a GitHub account.

Anyway, this queue is for reporting issues and requesting features, not for general user support. We specifically state this on the ["Contributing" page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question) linked in the "New Issue" template. Some GitHub users might respond to general questions, but we are trying to discourage this and direct users elsewhere so that our queue doesn't fill up and divert us from our work.

Simply sign up for any of those groups and you will have hundreds of helpful new friends. Note that although the RepRap Forum requires a login to ask questions, it doesn't require an account if you just want to search for related topics.

closed 2018-03-29 by thinkyhead