MarlinFirmware/Marlin #10127: Flashing XYZ and G92 not updating LCD/Firmware

midohioboarder1, 2018-03-16:
Marlin v 1.1.8
RAMP 14 EFB
I have not successfully run this version of Marlin on my printer yet.  Trying to get these last two issues cleared. 

Not sure if these two things are related so I thought I would mention them both.  I am mainly concerned that when I send a G92 command, I don't see the LCD display update.  So, if my Z is physically up a 130 mm and I jog it down close to the bed then send an X home command, the Z will raise back up 130 + 5 more for the lift... I cannot send a G92 to tell the software that the hot end is physically in a different place.

I would like to be able to run the printer doing 'homing' by hand without using any endstop switches.  (disable motors... move x and y  by hand.  slowly move z by jog controls paper gauge... and send G92. start printing)

Thanks for your help!

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1820763/Marlin.zip)

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* Read the "Contributing" page (under "Helpful resources") for additional guidelines.

GMagician, 2018-03-17:
With G92 don't you have to specify also axis name to achieve something?

midohioboarder1, 2018-03-17:
absolutely.  G92 X0 Y0 Z0 would tell the software to consider where ever the ...  I think I am answering my own question...
I read a response from one of the developers that you can't get around the end stops.   I think my problem is that the software is waiting for the Home position to be set. Almost sure of it now. 
I will start looking for a way to tell the software that where the hot end currently sits, is home.  Or a way to disable the whole homing thing.

closed 2018-03-17 by midohioboarder1

thinkyhead, 2018-03-17:
> I don't see the LCD display update.

Please use the `bugfix-1.1.x` branch. The on-screen position is fixed in that branch.

If you want to use `G92` like that, you must either turn off (or disable) software endstops or enable the `NO_WORKSPACE_OFFSETS` option. The correct command to adjust the Home Offsets is `M206`.

I have more to post in reply, but I have to run right now!