MarlinFirmware/Marlin #10129: Delta Auto Calibration With BLTouch Not Working

Cjslezak, 2018-03-17:
### Description

When I hit **Prepare>Delta Calibration>Auto Calibration** on my HE3D K280 3d printer it fails to level.  In the configuration.h file I set the correct offsets using **X_PROBE_OFFSET_FROM_EXTRUDER** and **Y
_PROBE_OFFSET_FROM_EXTRUDER** for my bltouch but it's completely ignoring them and running a z probe outside of the build area and failing every test.  Either I'm missing something or something's wrong.  Any help would be appreciated!

Video showing my issue: https://vimeo.com/260521105
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1821234/Configuration.zip)

### Steps to Reproduce

1. [First Step]
Click **Prepare**
2. [Second Step]
Click **Delta Calibration**
3. [and so on...]
Click **Auto Calibration**

**Expected behavior:** [What you expect to happen]
Automatically calibrate the bed.

**Actual behavior:** [What actually happens]
Probing outside build area and failing.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* Read the "Contributing" page (under "Helpful resources") for additional guidelines.

thinkyhead, 2018-03-18:
What version of Marlin are you using?

Cjslezak, 2018-03-18:
I'm using Marlin 1.1.8  
Upon a very long time of experimenting I found a workaround for my issue.  For some reason when I set the Delta Calibration Radius in Configuration.h the printer refused to acknowledge it.  I figured out how to set the radius with g-code so that the probe doesn't fall off the bed, but when I run G33 it still thinks that the probe is on the tip of the extruder.

Cjslezak, 2018-03-18:
I'm watching it do it's first print and due to the offsets not registering the first layer is looking pretty terrible.

LVD-AC, 2018-03-18:
For calibration carriages need to be at specific positions so the effector (nozzle) needs to be at the center and the other calibration points, not the probe. So set the delta_calibration_radius small enough the probe stays on the bed. (Either in FW followed with M502 or by M665 Bx.xx)

thinkyhead, 2018-03-19:
> set the delta_calibration_radius small enough the probe stays on the bed.

In firmware configuration the automated method is to use:
```cpp
#define DELTA_CALIBRATION_RADIUS \
  (DELTA_PRINTABLE_RADIUS - max(abs(X_PROBE_OFFSET_FROM_EXTRUDER), abs(Y_PROBE_OFFSET_FROM_EXTRUDER)))
```

AnHardt, 2018-03-19:
That's crap. The length of the diagonal would be right.

AnHardt, 2018-03-19:
Even better would be to test against a circle with `DELTA_PRINTABLE_RADIUS`, but shifted by `PROBE_OFFSET_FROM_EXTRUDER` center.

thinkyhead, 2018-03-19:
> Even better would be to test against a circle with DELTA_PRINTABLE_RADIUS, but a shifted by PROBE_OFFSET_FROM_EXTRUDER center.

Well, in fact the bounds checking has been altered so it does exactly that. However, for `G33` calibration I believe it's vital that the center of the probe area be in the center of the bed.

LVD-AC, 2018-03-19:
Indeed this was to restrictive; for calibration the probe must stay on the bed in order to be able to probe even if it goes outside the printable area between the towers.

In order to set delta_printable_radius you need to move the effector around to see at what point the arms or effector start hitting one of the belts; for delta_calibration_radius you need to move the effector around to see at what point the probe no longer sees the bed and triggers. The arms hitting the belts and the probe running of the bed are 2 different things that are unrelated and can't be calculated.

Cjslezak, 2018-03-19:
So I can't use an eccentric probe to do G33 calibration?  Would UBL work better?  I made the DELTA_CALIBRATION_RADIUS small enough that the probe doesn't fall off of the bed, but the bottom layer on all of my prints still looks pretty bad.
Is there a bed leveling function that takes PROBE_OFFSET_FROM_EXTRUDER into effect?  I've already got the offset set up for my probe in the firmware.

LVD-AC, 2018-03-19:
Sure you can use an eccentric probe. But every hardware has it's specific problems to be solved. Eccentric probes are much more sensible to effector tilt, resulting in a different z_offset being reported on different positions of the bed. This will make probing less accurate so calibration (G33) or bed levelling (G29) will suffer from that. Further G33 calibration will be more accurate the larger you set the delta_calibration_radius, eccentric probes will have to use a smaller radius as nozzle probes, so they will be less accurate. But G33 provides good results on any decent build delta printer.

What is your final std.dev at the end of G33?

Cjslezak, 2018-03-19:
My std.dev is 0.279

Cjslezak, 2018-03-19:
Just ran a longer scan and got std.dev .062  Will test it out later.

LVD-AC, 2018-03-20:
This indicates hardware problems. For reference (after a lot of tweaking the hardware) I obtain std.dev < 0.020 every time and flat layers all over the build plate as result.

Cjslezak, 2018-03-20:
What sort of hardware tweaking did you do?

LVD-AC, 2018-03-20:
Stabilizing the rods with springs, new effector, linear rails with new carriages,... but it all depends on YOUR printer build. Part of the fun with cheap Chinese kits; it gets you started but a lot of room for improvement.

boelle, 2019-02-20:
@Cjslezak still having issus when using latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead