MarlinFirmware/Marlin #10132: [Linear Advance 1.5] : Geared extruder with low jerk slows down print speed

T69, 2018-03-17:
Hi,

I'm using a Zesty Nimble and was testing linear advance 1.5 with great success. The problem I have is, Zesty recommends to use a extruder jerk value of 0.6 to ensure a smooth operation of the 1:30 worm gear. The effxt of having a very low jerk value for the extruder has the effect of slowing down the print speed with linear advance 1.5, which is limited now by the e jerk value as far as I understood the algorithm.
At the moment I'm using an ejerk of 2.5, which is a compromise, but I'm looking for another solution.

Any ideas?

Thanks & keep up the amazing work...

Sebastianv650, 2018-03-18:
I think I can understand why they recommend such a low jerk, most likely the flex shaft wouldn't translate a sudden rotation change on the stepper side to a rotation on the output end (quite the same problem we see on bowden printers).
Nevertheless I would do test prints (without LA), increasing ejerk to around 5 and see what it does in terms of print quality. If you don't get a noticeable quality change, the stated 0.6 is likely just a "we want to be safe" value, then I would use LA with a jerk which doesn't influence the quality like maybe the 5.

If higher jerks are realy impossible with this special extruder, I see no short term solution. In long term and with 32bit boards, we should realy think about S shaped acceleration curves. I'm quite sure they would improve print quality, and as there is no acceleration jump any more LA wouldn't lead to speed jumps on the extruder. This means nearly no extruder jerk would be needed.

T69, 2018-03-19:
Hi,
thanks for looking into this ... I'll try to increase ejerk up to 5 and see what happens.
Best regards ...

Tim-The-Woodsman, 2018-09-01:
Would you mind to share your results as well as a recommendation? Did LA do anything good in combination with the Nimble? I have to compile a new firmware anyways and I'm curious

Tim-The-Woodsman, 2018-09-02:
> I stopped experimenting with it as LA and the low jerk settings don't work
> together very well. I have very good results now with Marlin 1.1.9 and
> S-Curve acceleration ...

Would you mind sharing a link to that if you have one on hand? I have lost 
track after 1.7 and haven't heard of that but it sounds interesting.

T69, 2018-09-03:
Just download the new MArlin version here https://github.com/MarlinFirmware/Marlin ... then configure S-Curve Acceleration in configuration.h ... simple ...

Tim-The-Woodsman, 2018-09-03:
> Antworten an: MarlinFirmware/Marlin <reply@reply.github.com>

Thank you! I have to read into that. Sounds promising

boelle, 2019-02-20:
@Sebastianv650 should this one still be open?
@T69 problem solved?

T69, 2019-02-20:
Question answered, problem solved.

closed 2019-02-20 by T69