MarlinFirmware/Marlin #10136: Compiling failure: static assertion failed: Movement bounds ([XY]_MIN_POS, [XY]_MAX_POS) are too narrow to contain [XY]_BED_SIZE.

Wheelycool, 2018-03-17:
Why am I getting the following error?

static assertion failed: Movement bounds ([XY]_MIN_POS, [XY]_MAX_POS) are too narrow to contain [XY]_BED_SIZE.

--------------------------------------------------------------------------------------
//file Configuration.h Release:heads/1.1.x
//section info
#define STRING_CONFIG_H_AUTHOR "Hube3d_build1"
//section machine
#define MOTHERBOARD BOARD_MELZI
#define CUSTOM_MACHINE_NAME "Hube3d"
#define INVERT_Y_DIR false
#define X_BED_SIZE 280
#define Y_BED_SIZE 280
#define X_MIN_POS 20
#define Z_MAX_POS 300
//section extruder
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
#define EXTRUDE_MAXLENGTH 350
//section temperature
#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_BED 11
#define NOZZLE_PARK_POINT { (X_MIN_POS   130), (Y_MAX_POS - 165), 10 }
//section homing
//#define MANUAL_X_HOME_POS 20
//#define MANUAL_Y_HOME_POS 20
//#define MANUAL_Z_HOME_POS 5
//section motion
#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 4000, 105 }
//section lcd
#define SDSUPPORT 
#define REPRAP_DISCOUNT_SMART_CONTROLLER 

//file Configuration_adv.h Release:heads/1.1.x
//section lcd
#define LCD_INFO_MENU 
#define LED_CONTROL_MENU 
#define LCD_PROGRESS_BAR

GMagician, 2018-03-17:
Configuration file is missing. In your partial configuration there is only X_MIN_POS, what about X_MAX_POS and Y_MIN/MAX_POS?

Wheelycool, 2018-03-17:
Sorry, it didn't include all the info:

All the information on bed size and min/max limits were in the code when the error occurred. Not sure why it didn't copy all of it.

_______________________________________________-
//file Configuration.h Release:heads/1.1.x
//section info
#define STRING_CONFIG_H_AUTHOR "Hube3d_build1"
//section machine
#define MOTHERBOARD BOARD_MELZI
#define CUSTOM_MACHINE_NAME "Hube3d"
#define INVERT_Y_DIR false
#define X_BED_SIZE 280
#define Y_BED_SIZE 280
#define X_MIN_POS 20
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 300
//section extruder
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
#define EXTRUDE_MAXLENGTH 350
//section temperature
#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_BED 11
#define NOZZLE_PARK_POINT { (X_MIN_POS 130), (Y_MAX_POS - 165), 10 }

GMagician, 2018-03-17:
Ok that make sense....X_MIN_POS = 20, X_MAX_POS = 280, bed =280
X_MAX_POS-X_MIN_POS=260 hence your get your error message:
Movement bounds ([XY]_MIN_POS, [XY]_MAX_POS) are too narrow to contain [XY]_BED_SIZE.
self explaining error

Wheelycool, 2018-03-17:
Make sense,
 thanks.

GMagician, 2018-03-17:
if solved you can close it..

closed 2018-03-17 by Wheelycool