MarlinFirmware/Marlin #10137: Error MAXTEMP triggered if manual leveling bed is GT 0.05

fpaolo63, 2018-03-17:
### Description

Error MAXTEMP triggered if manual leveling bed is gt +0.05

Manual leveling the bed (3points) setting a value gt +0.075, trigger the "Error MAXTEMP"
soon Extruder Temperature is set.
(repro 100%)

Marlin SW: v. 1.1.8
controller: MKS Gen V1.4 - AVR mega 2560 - (Chinese clone) 
 
### Steps to Reproduce

Issue can be triggered using different sequence 

A.1) Start 3D printer, auto home.
A.2) Start manual bed leveling procedure, set same Z offset for the 3 points with value gt 0.075
A.3) try preheat the nozzle at 210c
A.4) display shows Error MAXTEMp

B.1) Start 3D printer, auto home.
B.2) preheat the nozzle at 210c
B.3)  Start manual bed leveling procedure, set same Z offset for the 3 points with value gt 0.075
B.4) Exiting manual bed leveling procedure display shows Error MAXTEMp

C.1) Start 3D printer, auto home.
C.2) Start manual bed leveling procedure, set same Z offset for the 3 points with value gt 0.075
C.3) Starting plot from SD, display shows Error MAXTEMp

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* Read the "Contributing" page (under "Helpful resources") for additional guidelines.
[MarlinBugConfig.tar.gz](https://github.com/MarlinFirmware/Marlin/files/1821680/MarlinBugConfig.tar.gz)
[Marlin-ErrTemp-Levelling.txt](https://github.com/MarlinFirmware/Marlin/files/1821687/Marlin-ErrTemp-Levelling.txt)

thinkyhead, 2018-03-18:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

fpaolo63, 2018-03-18:
hi, 
I test bugfix-1.1.x, same behaviour.
Going to have a look on code

thinkyhead, 2018-03-19:
Does this affect other leveling types besides 3POINT?

fpaolo63, 2018-03-21:
some more info!
I test LEVELING_LINEAR with 3 points grid and it is fine. 
... but if I disable  INDIVIDUAL_AXIS_HOMING_MENU also 3Point is fine.

let me recap:
INDIVIDUAL_AXIS_HOMING_MENU &&  AUTO_BED_LEVELING_3POINT => wrong behaviour 
 not INDIVIDUAL_AXIS_HOMING_MENU &&  AUTO_BED_LEVELING_3POINT =>  fine
INDIVIDUAL_AXIS_HOMING_MENU &&   AUTO_BED_LEVELING_LINEAR =>   fine

thinkyhead, 2018-03-24:
That's pretty strange. When we see bugs like this it makes us suspect the compiler is doing weird things, possibly due to some optimization flag. Sometimes these bugs go away if you use a different version of Arduino to compile the firmware. Issues like this can also imply a buffer overflow, so I'll have a look at the 3-point leveling and see if there's anything that could possibly be writing outside of bounds or blowing up the stack.

When you compile the firmware, how much SRAM usage does it report?

thinkyhead, 2018-03-24:
To see if this is a buffer overflow or not, please test this code:

https://github.com/thinkyhead/Marlin/archive/bf1_tool_change_debug.zip

If there's a buffer overflow this will throw a different error.

fpaolo63, 2018-03-28:
Hi 
I configured the bf1_tool_change_ debug for my machine and enable
INDIVIDUAL_AXIS_HOMING_MENU  AUTO_BED_LEVELING_3POINT

The behaviour is changed, now machine halt soon I save the 3rd point
on display I'm getting:
**Index Error!
PRINTER HALTED
Please reset**

Serial console log:

_> start
> echo:Marlin bugfix-1.1.x
> 
> echo: Last Updated: 2018-01-20 | Author: (none, default config)
> echo:Compiled: Mar 29 2018
> echo: Free Memory: 3584  PlannerBufferBytes: 1264
> echo:Hardcoded Default Settings Loaded
> echo:  G21    ; Units in mm
> echo:  M149 C ; Units in Celsius
> 
> echo:Filament settings: Disabled
> echo:  M200 D1.75
> echo:  M200 D0
> echo:Steps per unit:
> echo:  M92 X240.00 Y240.00 Z1200.00 E170.00
> echo:Maximum feedrates (units/s):
> echo:  M203 X66.00 Y66.00 Z5.00 E25.00
> echo:Maximum Acceleration (units/s2):
> echo:  M201 X500 Y500 Z100 E10000
> echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
> echo:  M204 P750.00 R750.00 T750.00
> echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
> echo:  M205 S0.00 T0.00 B20000 X4.00 Y4.00 Z0.30 E5.00
> echo:Home offset:
> echo:  M206 X0.00 Y0.00 Z0.00
> echo:Auto Bed Leveling:
> echo:  M420 S0
> echo:Material heatup parameters:
> echo:  M145 S0 H190 B50 F0
> echo:  M145 S1 H240 B110 F0
> echo:PID settings:
> echo:  M301 P22.20 I1.08 D114.00
> echo:  M304 P10.00 I0.02 D305.40
> echo:SD card ok
> echo:enqueueing "G28"
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
> echo:enqueueing "G28"
> echo:busy: processing
> echo:busy: processing
> X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
> echo:enqueueing "G29 V1"
> G29 Auto Bed Leveling
> Manual G29 point 1 of 3
> echo:busy: processing
> echo:enqueueing "G29 V1"
> Manual G29 point 2 of 3
> Saving Z with index 1
> echo:busy: processing
> echo:busy: processing
> echo:enqueueing "G29 V1"
> Manual G29 point 3 of 3
> Saving Z with index 2
> echo:busy: processing
> echo:busy: processing
> echo:busy: processing
> echo:enqueueing "G29 V1"
> Manual G29 point 3 of 3
> Saving Z with index 3
> Error:Printer halted. kill() called!
> 
> 
> 
> 
> 
>_ 

Compiler messages 
_Archiving built core (caching) in: /tmp/arduino_cache_944085/core/core_arduino_avr_mega_cpu_atmega2560_1bef5bc7a2a8a5b0275553b94e398f7b.a
Sketch uses 118606 bytes (46%) of program storage space. Maximum is 253952 bytes.
Global variables use 4601 bytes (56%) of dynamic memory, leaving 3591 bytes for local variables. Maximum is 8192 bytes._

regards
paolo

thinkyhead, 2018-03-28:
Ok, this is now patched in `bugfix-1.1.x`. Please confirm that it's fixed when you get a chance, and if all is well we can close this issue.

closed 2018-03-29 by thinkyhead

fpaolo63, 2018-04-01:
I confirm SOLVED.

commit 89592bb3 (2018-03-28): Fix linear/3-point manual leveling buffer overrun

Fixes #10137
commit 3c016fba (2018-03-28): Fix linear/3-point manual leveling buffer overrun

Fixes #10137
commit 6e782283 (2018-03-28): Fix linear/3-point manual leveling buffer overrun

Fixes #10137
commit 3edf9904 (2018-03-28): Fix linear/3-point manual leveling buffer overrun

Fixes #10137