MarlinFirmware/Marlin #10139: Servo pins not givng out 5V

RedDragon99, 2018-03-17:
Hello
I have a problem and after 2 hours of trying to solve it I decided to ask you here
I tried every solution on the internet but none of them works
I am using BLTOUCH sensor for auto leveling
I configured the required settings to enable it.
But it still doesnt work... the sensors light doesnt even blink.
I tried measuring the pins with multimeter and they dont give out 5V as they should.
My other endstops are working.
ANd yea I am using this for my DELTA project.

And my configuraion.h:

[https://pastebin.com/N3DtnFve](url)

TheSFReader, 2018-03-17:
If you're using a RAMPS setup, did you provide 5v to the servo lines? This can be done via a specific pin on the board either by an external 5v wire, or by bridging it with the neighboring vcc pin.

RedDragon99, 2018-03-17:
RAMPS are connected via the arduino MEGA ...
All the endstops have 5V when measured with multimeter

kAdonis, 2018-03-17:
Do you have this jumper set on RAMPS?
[image]

RedDragon99, 2018-03-17:
Ooh yea I didnt know I had to do that
Yea its working know thanks!!!

closed 2018-03-17 by Bob-the-Kuhn