MarlinFirmware/Marlin #10142: Feature request - Relay output to trigger on thermal runaway

bipsendk, 2018-03-17:
### Description

Optional Relay control of power to hotend/heated bed if thermal runaway

#### Additional Information

It would be desirable to be able to specify an pin, that could be used to switch off a relay if thermal runaway was detected.

When the printer is powered on, the designated output should go high to switch a relay on. The relay controls the power to the mosfets driving the heated bed and the hotend.

If a thermal runaway is detected, the output pin should change state to low to switch the relay off - and cut power completely to the mosfets.

 The reason for adding a relay before the mosfets is that if a mosfet fails, it will typically to so in a closed state (thereby still supplying power to a connected device with the risk of overheating/igniting a fire).
If power to the mosfets is controlled by a relay, it will be possible to cut power completely - even if a mosfet has failed.

ManuelMcLure, 2018-03-17:
One possible addition to this FR is if you have POWER_SUPPLY defined to something other than 0 add a configuration variable to tell Marlin to turn the power supply off if thermal runaway is detected. Another layer of protection.

bipsendk, 2018-03-17:
Using the PS_ON_PIN and POWER_SUPPLY might be possible together with an option to cut power on thermal runaway. For a relay-based solution, it will require the PS_ON_PIN to be initialized as active high (5V) - and then change state to low (0V), if thermal runaway is detected and Thermal Runaway power cut has been defined/enabled.

teemuatlut, 2018-03-18:
Thermal runaway already turns off the PSU.

thinkyhead, 2018-03-19:
> For a relay-based solution, it will require the PS_ON_PIN to be initialized as active high (5V) - and then change state to low (0V)

- For `POWER_SUPPLY == 1` the ON state is `LOW`.
- For `POWER_SUPPLY == 2` the ON state is `HIGH`.

closed 2018-03-19 by thinkyhead

ManuelMcLure, 2018-03-19:
> Thermal runaway already turns off the PSU.

Is that true? From what I see, `kill()` sets the `PS_ON` pin to INPUT which is high-impedance or pull-up, so if `POWER_SUPPLY == 1` you'll end up with the power supply still powered on. For the power supply to be turned off in the `POWER_SUPPLY == 1` case you need to leave the pin set to output and low.

I think the code in `kill()` should look like this:

```
#if HAS_POWER_SWITCH
  OUT_WRITE(PS_ON_PIN, PS_ON_ASLEEP);
#endif```

thinkyhead, 2018-03-19:
> `kill()` sets the `PS_ON` pin to `INPUT`

How odd! I'll have to see why that is. If there's no clear justification then we'll fix that.

thinkyhead, 2018-03-19:
There's nothing documented about it, so I've made the change. Note that there's also a `SUICIDE_PIN` that can be defined. It will be set to `LOW` on `kill` and also acts as a power control.

AnHardt, 2018-03-19:
What state should a power supply have when the Arduino is unpowerd? Off.
What state do the outputs of the Arduino have when unpowerd? Neither on, nor off nor pulled up.

Make the hardware so, that the power supply is off when the Arduino is off and all will be good.

thinkyhead, 2018-03-19:
@AnHardt — To which point are you responding?

AnHardt, 2018-03-19:
I'm talking about power supplies being internally pulled up/down to be off
and floating output of the `PS_ON` pin does not matter if working at all.

thinkyhead, 2018-03-19:
As I understand it we support two options: so-called "**ATX**" and "**XBox**" with opposite conditions for switching on, but of course they both rely on the board being powered to be meaningful in firmware terms.

I agree, the best setup is one that powers off when the board is unpowered.