MarlinFirmware/Marlin #10143: Z-axis problems with new FW

ProtoprintCZ, 2018-03-17:
Hello guys! :)

I have a problem with Marlin 1.1.18. With the old version it works ok but with the new one it always jumps in the Z axis (while the X and Y axes are OK) (see video here: https://www.facebook.com/Demonix691/videos/10204277656739474 ) 

- Was tried also bugfix verzion and results are the same
- Options are used from Kosseel XL (which I set with my own dimensions)
- When I control it manualy, works fine
- I have drivers  drv8825, Current 1A (based od earlier best practice)
- Not using correction, only autocalibration (always is set the same)

Thank you very much in advance!

Steven

thinkyhead, 2018-03-17:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

thinkyhead, 2018-03-17:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop them into your next reply. We'll check them over and see if anything is amiss.

sh4jonny, 2018-03-18:
Change
#define MINIMUM_STEPPER_PULSE 0
to 2 for your 8825

Bob-the-Kuhn, 2018-03-19:
If you are using a DUE then changing it to 2 is useful.  For the 8 bit board the actual minimum stepper pulse is in the 14uS range even when  **MINIMUM_STEPPER_PULSE** is set to 0.

closed 2019-07-17 by shitcreek