MarlinFirmware/Marlin #10146: BUGFIX-2.0 AZSMZ_12864 RAMPS SMART

maar1201gh, 2018-03-18:
Blank Screen. 

last compile from version 27/0172018 is working. any clues?

Thanks

thinkyhead, 2018-03-19:
> any clues?

It could be literally anything based on the details provided.

jeffyl, 2018-03-20:
It looks like it's been broken since e1fd9c08b33be7073dbcfe57b88cd6f724aa34df when it setup the LCD to use hw SPI over software SPI? 

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/ultralcd_impl_DOGM.h#L191

jeffyl, 2018-03-20:
Looking at it some more, it doesn't seem like the SMART RAMPS can use hardware SPI at all, as it doesn't break out the hardware SPI pins on the ICSP

Bob-the-Kuhn, 2018-03-20:
Try adding these to the `#if ENABLED(AZSMZ_12864)` section of **pins_RAMPS_SMART.h**
```
    #define DOGLCD_SCK          SCK_PIN
    #define DOGLCD_MOSI         MOSI_PIN
```

The AZSMZ_12864 uses the same SPI interface for the LCD and SD card sections.  The above definitions will result in the correct driver being selected (the hardware SPI driver).

----

While looking at the SMART RAMPS 1.42 pinout from the RepRap WIKI it appears the rotary encoder may be setup wrong in  **pins_RAMPS_SMART.h**.  At least they appear to disagree on two pins.  If the encoder doesn't work then make the following changes in the **pins_RAMPS_SMART.h** file:
```
  #define BEEPER_PIN       65 // RepRap pin out says 65, 66 was here originally 
  #define BTN_ENC          66 // RepRap pin out says 66, 67 was here originally 
```

----

Looks like we missed this LCD when we added the check for the shared SPIs.  If this fixes the blank screen then we'll need to see if there are other shields/boards that need this change.

Bob-the-Kuhn, 2018-03-20:
The **pins_RAMPS_SMART.h** file mentions an "easy adapter board" in the AZSMZ_12864 section.

Can someone point me to info on it?

Bob-the-Kuhn, 2018-03-20:
Looks like pins_AZSMZ_MINI.h is the only other place that uses the AZSMZ_12864.  It'll also need this change.

jeffyl, 2018-03-20:
>The above definitions will result in the correct driver being selected (the hardware SPI driver).

Tried it, and there is no change.

The SMART RAMPS should be using software SPI. The hardware SPI pins on the Due's ICSP are not connected to the SMART RAMPS at all, how can it use hardware SPI?

The beeper and encoder have always worked fine with those pins.

>"easy adapter board"

I'm not aware of aware such board.

jeffyl, 2018-03-20:
[HAL/HAL_DUE/spi_pins.h claims that only 4 pins](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_DUE/spi_pins.h#L29) can used for HW SPI SS. The SMART RAMPS doesn't use any of those, and defaults to the software SPI implementation.

Bob-the-Kuhn, 2018-03-20:
I was wondering what you meant by ICSP ...

Was the SD card portion working before?

That means we need the following defines to get the LCD portion working. 
```
    #define DOGLCD_SCK          52
    #define DOGLCD_MOSI         51
```

The SD card software also has a software SPI option.  I'll need to dig a bit to find those.

jeffyl, 2018-03-20:
Shouldn't the DOGLCD_SCK and MOSI already be set to that?

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/ultralcd_impl_DOGM.h#L190 currently resolves to true, which is why it attempts to use HW-SPI.

`#if DOGLCD_SCK == SCK_PIN && DOGLCD_MOSI == MOSI_PIN`

>Was the SD card portion working before?

Yes, SD card still works fine.

Bob-the-Kuhn, 2018-03-20:
Yes, they are defined in **Conditionals_post.h** when this type of display is enabled and they aren't defined anywhere else.

If the SD card works then there has to be a softspi enabled somewhere.  

The bottom line is your system can't use the DUE's hardware SPI which means `#if DOGLCD_SCK == SCK_PIN && DOGLCD_MOSI == MOSI_PIN` is getting in the way.

The light is slowly dawning.

A quick work around is to just replace `#if DOGLCD_SCK == SCK_PIN && DOGLCD_MOSI == MOSI_PIN` with `#if 0`

Bob-the-Kuhn, 2018-03-20:
Did the encoder work previously?

If yes I need to find a correct pinout for this board.

Bob-the-Kuhn, 2018-03-20:
The Smart RAMPS 1.41 pinout on the RepRap WIKI matches the file.

jeffyl, 2018-03-20:
Are they not correct with a12 and a13? 

Encoder has worked since I added the LCD support.

jeffyl, 2018-03-20:
According to https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_DUE/HAL_pinsDebug_Due.h that is dac0 and dac1 which match the 66 and 67

Bob-the-Kuhn, 2018-03-20:
I think the RepRap WIKI pinouts both have problems.

The DUE info I have says that 
```
DUE logical digital pin  D64  D65  D66  D67
DUE logical alternative  A10  A11  DAC0 DAC1
SAM3x8E port             B19  B20  B15  B16  
SAM3X8E A/D converter    AD12 AD13 DAC0 DAC1
RepRap 1.41                        A12  A13
RepRap 1.42                   A12  A13
```

jeffyl, 2018-03-20:
I see what you mean. Going off the analog number seems to be correct

Bob-the-Kuhn, 2018-03-20:
The RepRap WIKI schematics say "Due or MEGA", the connection between the shield and the CPU board says "Arduino MEGA Shield, and the pin names on the connection match the 2560 names which match the names on the 1.4.1 pinout.

Really strange.  Hope I don't need to explain this to anyone.

I'm happy if you're functional with the current pins_RAMPS_SMART.h file and the` #if 0` work around.  I'll need to do something about that `#if DOGLCD_SCK == SCK_PIN && DOGLCD_MOSI == MOSI_PIN` problem.

Bob-the-Kuhn, 2018-03-20:
Did the `#if 0` work around get you up & going?

Bob-the-Kuhn, 2018-03-20:
The fix is in PR #10168.

jeffyl, 2018-03-20:
Does that mean the board works on a MEGA 2560 too? Obviously wouldn't compile right now because of the board check, but if someone really wanted to?

Maybe update the pins documentation: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_RAMPS_SMART.h#L45 while you're at it?

Bob-the-Kuhn, 2018-03-20:
Yes, it looks like it well.

I'd be surprised if someone did.  The standard RAMPS card are cheap & I didn't see that this offered anything additional to 2560 users.

closed 2018-03-20 by Bob-the-Kuhn

thinkyhead, 2018-06-14:
The compiler has a bug. Remove the end-of-line comment from…
```cpp
   #define BEEPER_PIN       66   // Smart RAMPS 1.42 pinout diagram on RepRap WIKI erroneously says this should be pin 65
```
…and any other pins that the compiler complains about until the errors go away.