MarlinFirmware/Marlin #10147: [Bugfix_2.0] Unable to update configuration.h

bwgeez02, 2018-03-18:
When I try to change my z probe offset or temps for PLA/ABS preheat in configuration.h and upload it it doesn't change.

Step 1. Open bugfix.2.0 in Arduino 1.9 Beta
Step 2. Change z_probe_offset
Step 3. Upload to Due
Step 4. Run auto home and/or check z probe offset in printer menu (offset never changes)

I have changed other things like custom machine name and they update normally.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1823111/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1823112/Configuration_adv.zip)

AnHardt, 2018-03-18:
EEPROM!

p3p, 2018-03-18:
When settings do not change it is generally because you saved the settings to the EEPROM, this overrides the default configuration even after flashing a new firmware (as is intended).

When using an EEPROM to make sure you are using the settings from `Configuration.h` you need to use the following gcode after flashing a new firmware.
```
M502 ; Reset to factory default configuration
M500 ; Save to EEPROM
```

bwgeez02, 2018-03-18:
There's where my next problem comes. With the new bugfix2.0 my host software (pronterface) won't initiate commands anymore. I've also tried S3d. They connect to the printer and read settings but when I issue commands nothing happens.

bwgeez02, 2018-03-18:
I was able to initialize eeprom from the lcd menu and it saved the new settings, but It doesn't appear to be saving a new UBL.

p3p, 2018-03-18:
> With the new bugfix2.0 my host software (pronterface) won't initiate commands anymore

I've seen this issue brought up elsewhere recently but not solved, you should close this issue if it solves your original question, and open a new issue with your new question in the title to attract the attention of someone that may know better (@Bob-the-Kuhn).

> I was able to initialize eeprom from the lcd menu and it saved the new settings, but It doesn't appear to be saving a new UBL.

Do you mean it does not reset your UBL Mesh data?

bwgeez02, 2018-03-18:
It does reset. I just figured out how to save the new UBL. Had to watch pronterface while using the lcd menus to find out what commands were being sent. I'll start a new issue for the pronterface problem. Thanks everyone.

closed 2018-03-18 by bwgeez02