MarlinFirmware/Marlin #10149: [bugfix_2.0] Due can't be controlled with host

bwgeez02, 2018-03-18:
Connected through the programming port (haven't tried the native port) my host software will connect, however, won't issue commands. I've tried Pronterface and S3d. Both will show commands sent with the lcd, but it's difficult to know what the actual G or M code is for each menu item unless each menu item is selected/pressed. Due/Radds/RaddsLCD

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1823265/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1823266/Configuration_adv.zip)

Bob-the-Kuhn, 2018-03-19:
We're already working this problem in Issue #10116 so I'm going to close this issue.  Please do any further posts in Issue #10116.

I expect we'll have a work around merged tonight or early tomorrow.

Please post in Issue #10116 which revision of DUE board you have.  So far we have three older revision boards (FET left to right) that have no problems and one latest revision (FET top to bottom) that fails.

We don't know if the board revision makes a difference or not.

closed 2018-03-19 by Bob-the-Kuhn

bwgeez02, 2018-03-19:
Thanks! I guess I missed that issue.