MarlinFirmware/Marlin #10150: Bed PID autotune: heating failed

VanessaE, 2018-03-18:
### Description

This is the bugfix-1.1.x branch, at commit 0d13c5d0779293c59895f6d72217e509027a364a

Attempting to run PID autotune on my heated bed fails, as follows:

```
>>> m303 e-1 c8 s100
SENDING:M303 E-1 C8 S100
PID Autotune start
Error:Heating failed, system stopped! Heater_ID: bed
Error:Heating failed, system stopped! Heater_ID: bed

Error:Printer halted. kill() called!
Error:Printer halted. kill() called!
```
(Interesting that it throws those errors twice)

I can heat the bed up just fine during normal operations, with no incidents, and PID autotune works fine on the hotend.  When I tried it with a cold bed (ambient room temp of about 27°C), it made it to about 61° before it dumped out.  If I preheat the bed to, say, 80° first, then run PID autotune on it, it dies after 20-25 seconds and a rise of just over 1° (that speed is normal for this bed).

It's acting as though bed thermal runaway protection is getting in the way, even though it is *disabled* in my config (my bed can only reach barely 110° under normal usage, and it takes almost 20 minutes to get there; it can barely get to about 135° if I heavily insulate the top, so there's obviously no fire hazard there).

### Steps to Reproduce

Just run the above M303 command.

You may need to do this on a printer that uses a **12v PCB** heater (i.e. MK2A) or something equally slow.

#### Additional Information

Configs:  [marlin-configs-20180318.zip](https://github.com/MarlinFirmware/Marlin/files/1823267/marlin-configs-20180318.zip)

thinkyhead, 2018-03-19:
Does it work if you relax the `WATCH_BED_TEMP_PERIOD` / `THERMAL_PROTECTION_BED_PERIOD`?

VanessaE, 2018-03-19:
Well, it got a little further (71-73°) but it still dies.

thinkyhead, 2018-03-19:
Strange. It should work if the period is relaxed by enough.

VanessaE, 2018-03-19:
I should think so, but given that thermal runaway protection is disabled....... :smile:

thinkyhead, 2018-03-19:
It still says "Heating failed, system stopped!" with thermal runaway protection disabled?

VanessaE, 2018-03-19:
Correct.

thinkyhead, 2018-03-19:
That's simply not possible if both `THERMAL_PROTECTION_HOTENDS` and `THERMAL_PROTECTION_BED` are disabled. The code to print that string won't get compiled in.

VanessaE, 2018-03-19:
It's definitely disabled.  Check my configs. :smiley:

AnHardt, 2018-03-19:
Is it definitely compiled and uploaded?
The problem should have been fixed early this year -> bugfix.

thinkyhead, 2018-03-19:
Make sure that those those options are fully disabled in the configs and that you're definitely getting a successful flash, then test again. Thermal protection won't even be included in the binary (`M303`, `M104`, `M109`) when those options are disabled.

VanessaE, 2018-03-19:
Only bed runaway protection is disabled.

Hotend runaway protection is ENabled (it wouldn't be particularly smart not to).

I've kept those settings that way for months, and I am 100% certain what's in the printer matches my configs (as I always use the compile-and-upload function in the Arduino IDE, same place where I edit my configs when needed, so that there's no doubt).

AnHardt, 2018-03-19:
Got it.
If `THERMAL_PROTECTION_BED` is off and `THERMAL_PROTECTION_HOTENDS` is on,
`watch_temp_increase` and `watch_temp_period` are initialized with the values for `THERMAL_PROTECTION_HOTENDS`.
```
    #define HAS_TP_BED (ENABLED(THERMAL_PROTECTION_BED) && ENABLED(PIDTEMPBED))
    #if HAS_TP_BED && ENABLED(THERMAL_PROTECTION_HOTENDS) && ENABLED(PIDTEMP)
      #define TV(B,H) (hotend < 0 ? (B) : (H))
    #elif HAS_TP_BED
      #define TV(B,H) (B)
    #else
      #define TV(B,H) (H)
    #endif

    #if WATCH_THE_BED || WATCH_HOTENDS
      const uint16_t watch_temp_period = TV(WATCH_BED_TEMP_PERIOD, WATCH_TEMP_PERIOD);
      const uint8_t watch_temp_increase = TV(WATCH_BED_TEMP_INCREASE, WATCH_TEMP_INCREASE);
      const float watch_temp_target = target - float(watch_temp_increase + TV(TEMP_BED_HYSTERESIS, TEMP_HYSTERESIS) + 1);
      millis_t temp_change_ms = next_temp_ms + watch_temp_period * 1000UL;
      float next_watch_temp = 0.0;
      bool heated = false;
    #endif
```

AnHardt, 2018-03-19:
Try
temperature.cpp
```diff
@@ -422,23 +422,25 @@ uint8_t Temperature::soft_pwm_amount[HOTENDS],
         #endif
         next_temp_ms = ms + 2000UL;
 
         // Make sure heating is actually working
         #if WATCH_THE_BED || WATCH_HOTENDS
+          if ((WATCH_THE_BED && hotend < 0) || (WATCH_HOTENDS && hotend >= 0)) {
           if (!heated) {                                          // If not yet reached target...
             if (current > next_watch_temp) {                      // Over the watch temp?
               next_watch_temp = current + watch_temp_increase;    // - set the next temp to watch for
               temp_change_ms = ms + watch_temp_period * 1000UL;   // - move the expiration timer up
               if (current > watch_temp_target) heated = true;     // - Flag if target temperature reached
             }
             else if (ELAPSED(ms, temp_change_ms))                 // Watch timer expired
               _temp_error(hotend, PSTR(MSG_T_HEATING_FAILED), PSTR(MSG_HEATING_FAILED_LCD));
           }
           else if (current < target - (MAX_OVERSHOOT_PID_AUTOTUNE)) // Heated, then temperature fell too far?
             _temp_error(hotend, PSTR(MSG_T_THERMAL_RUNAWAY),
               hotend >= 0 ? PSTR(MSG_THERMAL_RUNAWAY) : PSTR(MSG_THERMAL_RUNAWAY_BED)
             );
+          }
         #endif
       } // every 2 seconds
 
       // Timeout after 20 minutes since the last undershoot/overshoot cycle
       if (((ms - t1) + (ms - t2)) > (20L * 60L * 1000L)) {
```

VanessaE, 2018-03-19:
That got me much closer (using the "EDITED !!!" version :stuck_out_tongue: ).  It still crapped out, but with a non-fatal "PID Autotune failed! timeout" error, because of how slow my bed is to heat up.

I can immediately restart the autotune, however, and it finishes just fine.

Note that I had to apply your changes by hand, as you cut off the start of the diff.

(Fwiw, it got to about 90° before it timed out, with the target temperature of 100°.  Ambient room temperature of about 22°C this time.)

AnHardt, 2018-03-20:
@VanessaE 
For your exceptional slow bed i made `MAX_CYCLE_TIME_PID_AUTOTUNE` configurable - but hidden.
Either increase the value directly in `temperature.cpp` od add a new define somewhere in the configs.

VanessaE, 2018-03-20:
Nah, there's no need for that, it's not like one has to PID tune their bed frequently.

thinkyhead, 2018-03-21:
Changes merged! Thanks for your sharp eyes, @AnHardt!

closed 2018-03-21 by thinkyhead

fefeaneta8, 2018-08-23:
Hi ich wollte meinen Anet A8 auch endlich mal auf Marlin upgraden.
(Hatte bis jetzt das Werksprogramm oben)

Habe schon den Hardware ordner in den Arduino ordner verschoben.
Bei Malin die Anet A8 dateien in den Hauptordner kopiert und gestartet.

Ich bekomme immer im arduino fenster beim Durchchecken eine Fehlermeldung: pasting "/* (bed)*/" and "_RPORT" does not give a valid preprocessing token und es makiert mir dabei folgende Zeile:
#define HEATER_BED_PIN 12 // (bed) 

Was kann ich dagegen machen?

thinkyhead, 2018-08-25:
@fefeaneta8 — It's a bug in the GCC compiler. In your selected `pins_MYBOARD.h` file, remove end-of-line comments from any pins that throw an error. Or, use a different version of Arduino IDE, or use PlatformIO to build instead.