MarlinFirmware/Marlin #10156: LIN_ADVANCE 1.5: Gaps on line starts

thinkyhead, 2018-03-19:
From @bruce356:

Hi, upgraded to the latest version of Marlin Bugfix 1.1.x with Lin-advance 1.5.
Previously used Marlin v1.1.6 with Lin-advance 1.0.
My setting #define LIN_ADVANCE_K 0.06.
Cartesian printer Mendelmax type, E3D V6 hot end 0.4mm nozzle, PLA 1.75mm direct drive extruder with 5 to 1 geared stepper. RUMBA Board with TMC2130 drivers.
With Lin-advance enabled there is a delay in starting to feed filament causing a gap.
With Lin-advance disabled there is no such problem.
This also occurs if Lin-advance is active and set to K 0.0
Pictures below - thanks
[image]
[image]

bruce356, 2018-03-19:
Please disregard the last line, "This also occurs if Lin-advance is active and set to K 0.0 "
That comment is incorrect, Printing is normal if K is 0
Sorry my bad.

Sebastianv650, 2018-03-19:
The symptom you described happens if K is too high. You might also try setting MINIMUM_STEPPER_PULSE to 2 if it's 0 at the moment or 4 if it's 2 now.
If it's not related to MINIMUM_STEPPER_PULSE, try printing a calibration pattern from Marlin documentation page using a range for K of 0 to 0.1 in 0.01 steps and see if it provides additional informations.

bruce356, 2018-03-20:
Hi Sebastianv650, MINIMUM_STEPPER_PULSE was set to 2 and I have now set it to 4.
Calibration pattern printed using a range for K of 0 to 0.1 in 0.01 steps, these are the settings I originally had to sellect a K factor of 0.06.
Photo below
MINIMUM_STEPPER_PULSE set to 4 does not seem to have made much difference.
Lin-Advance is still set at K 0.06
[image]
[image]

thinkyhead, 2018-03-20:
When I first got Simplify3D I noticed that it uses a flow multiplier of 90% as the default. I recommend changing that to 100%.
[image]

comps, 2018-03-20:
Looking at the pictures, it seems to me that there are more serious issues and a correct LA setting just makes them stand out more (by depositing the correct amount of material).

The most noticeable is extrusion path overlap - new paths are being extruded on top of previous paths (check extrusion multiplier, nozzle size, calibrate first layer height).

This however doesn't eliminate the holes and, if everything else is perfectly tuned, the holes are expected as, with a circular nozzle, you simply cannot get into those sharp corners, geometrically. Some slicers compensate for that with a special "hole fill" mode to squirt small bits of plastic into these holes, but a more reliable solution is to (either or both)
 * increase infil/perimeter overlap %
 * turn the infil another 45 degrees, so it makes lines parallel to perimeters
 * choose a different infil pattern

bruce356, 2018-03-20:
@thinkyhead My multiplier is set correctly in S3D, I have calibrated the flow rate to be correct.
@comps the  LA calibration pattern is not using UBL that is why the numbers on the right side failed.
My concern with LA 1.5 turned on is the slight gap left at the start of the outer perimeter layer line where no extrusion takes place until some 0.8mm of movement later. I have retraction turned on at 0.8mm NO wipe nozzle or coast at end. I can live without LA1.5 so its no biggy as the printer prints well without it.
Just pointing out  a floor. 
P.S. I am not worried about small gaps between outer layer and infill, that is not the reason I brought this up.
Regards - bruce

Sebastianv650, 2018-03-20:
It's a pleasure to see well focused images like this for debugging, I'm not used to that ;)

As stated before, this lack of extrusion during start is often a sign of too high K value. In your case (looking at the calibration pattern) it's definitly not a lot too much, but I would tend to use 0.04 or 0.05. The 0.06 one seems to thin out the deceleration part a little bit already. It's a small difference, but with 0.04 it's already a 33% change.

You are right, on such a printer it's quite possiblt to get nice parts without LA. But on the other hand, especialy this printers react very good to LA as you can see in the test pattern, so one it works it should give even better prints with nearly no overhead as K is so low.

bruce356, 2018-03-21:
Hi Sebastianv650, thanks for the reply, the focus is not perfect as its only my mobile phone camera.

I would like to state that this problem is resolved, I was totally to blame and apologise for any wasted time.

I keep getting caught with RAM settings not changing when an updated firmware is uploaded.
Li-Adv. was set on default 0.22 in RAM and I was therefore getting such large gaps at the start of a line of print and larger than normal radius on corners.

What finally made me realise something was not right is when I started applying the LA settings you recommended above (0.04 & 0.05) in the S3D script, and the print turned out as it should. Yet after configuring the Marlin firmware with the new LA settings and uploading, turning off the script settings,  the gaps reappeared in the print.  

Thanks for your help
Regards - bruce

closed 2018-03-21 by thinkyhead