MarlinFirmware/Marlin #10157: TMC 2130 changes required a reprint reset

meav33, 2018-03-19:
I'm not sure if this is a bit or some feature I don't understand, but I noticed that when ever I change the current in the config and then reflash it won't take effect until I manually reset the eeprom (m502 ->m500) and the power cycle the printer)

I'm using the latest Bugfix 1.1.x

expected behavior:
Changes in config take effect after flashing and reboot

actual behavior:
I need to send m502 -> m500 Everytime I changed the current in the config and flashed the firmware.

You can easily see this with the m122 command.

After changing current to 1000mah and flashing and reboot m122 still shows the old value. After sending m502 and then m502 it shows the new value

GMagician, 2018-03-19:
As I know every parameter that is written in eeprom override firmware defaults.
If there is some gcode to update current you have to use it and then M500 to save new value.
This way everytime you turn on you get updated values

AnHardt, 2018-03-19:
If you are always changing the config and reflash - simply turn off the EEPROM.

Purpose of the EEPROM is to let the values in it survive a reflash.

closed 2018-03-20 by meav33