MarlinFirmware/Marlin #10158: [Question] Mapping pins using fastio_1280.h

marcio-ao, 2018-03-19:
Hello,

I have a question. I've had occasions where I've needed to convert pin numbers in one of the "pin_*.h" files to Atmel ports. The way I have done this is by taking the pins from the "pins.h" file, opened up "fastio_1280.h" (assuming the board was a 1280 or 2560) and then looked up the pin in the [line next to 1280](https://github.com/MarlinFirmware/Marlin/blob/51d8ace4e55e51d1dda19fa4b18a7ba8824a15a2/Marlin/fastio_1280.h#L26), I can then read the Port from the middle line.

This trick has worked consistently for me, so I know it is correct. However, the thing that I've never understood is the 3rd line, labeled "Marlin", is not used. Can someone tell me where and under what circumstances those particular mappings are used?

Bob-the-Kuhn, 2018-03-19:
That table is FOOBAR!!!

I could have sworn it was fixed ages ago.  I'll get it fixed tonight.

In the mean time here's an Excel workbook with the correct info. [1280_2560_pins.zip](https://github.com/MarlinFirmware/Marlin/files/1826340/1280_2560_pins.zip)

marcio-ao, 2018-03-19:
@Bob-the-Kuhn: Comparing your spreadsheet to the table, it looks like my hunch is correct: the first line (labeled 1280) actually corresponds to the Marlin pins, but the line labeled "Marlin" does not. This is how I have been using the tables all along. The same pattern seems to happen with the other fastio files. So it looks like the labels need to be changed around a bit?

Bob-the-Kuhn, 2018-03-20:
My memory is even worse than I thought.  ☹️

Today the Marlin row has no meaning.

Back in the days of 1.0.x the AT90USB CPU wasn't officially supported by Arduino so there wasn't an "official" pin mapping.  Because of history it ended up with two different pin mappings.  One of them was called Marlin, the other Teensy.  Marlin was just straight "port A0 through port Z99 is assigned incrementing logical pin numbers starting at 0". 

Back in 1.1.2 all the different processor types were split out into separate fastio_xxx.h files.  I don't remember why the Marlin system was added to all the CPU types.  It was only ever used for the one type.

Then in 1.1.3 or 1.1.4 (maybe 1.1.5) the Marlin pin mapping was dropped for the AT90USB based controllers and all the pins_xxx.h files were converted to the Teensy system.

Bob-the-Kuhn, 2018-03-20:
I've just submitted PRs to delete the Marlin line and change "pins" to "logical pins"

closed 2018-03-20 by Bob-the-Kuhn

marcio-ao, 2018-03-20:
Cool. Glad my confusion lead to some house cleaning! :) Thank you!