MarlinFirmware/Marlin #10161: Dual endstod axis z

xero83, 2018-03-19:
I want to use a double endstop to use on xmax and xmin, with the endstop function for the z axis, both (xmax,xmin) will do the same function.

How can I implement it?

#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#ifndef ENDSTOPPULLUPS
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  // #define ENDSTOPPULLUP_XMAX
  // #define ENDSTOPPULLUP_YMAX
  // #define ENDSTOPPULLUP_ZMAX
  // #define ENDSTOPPULLUP_XMIN
  // #define ENDSTOPPULLUP_YMIN
  // #define ENDSTOPPULLUP_ZMIN
#endif

#ifdef ENDSTOPPULLUPS
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
#endif

// The pullups are needed if you directly connect a mechanical endswitch between the signal and ground pins.
const bool X_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Z_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool X_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Y_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Z_MAX_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Z_MIN_PROBE_ENDSTOP_INVERTING = false; // set to true to invert the logic of the endstop.
//#define DISABLE_MAX_ENDSTOPS
//#define DISABLE_MIN_ENDSTOPS

thinkyhead, 2018-03-21:
Sorry, it's not quite clear. Can you elaborate?

xero83, 2018-03-21:
in the extruder I have two endstops, which I want to connect each of them to zmin and zmax, and which do the same thing.
First of all I did the following on pin:
#define Z_MIN_PIN          33
#define Z2_MIN_PIN        32
#define Z_MAX_PIN          -1

In Stepper:
#if defined(Z2_MIN_PIN) && Z2_MIN_PIN > -1
					bool z_min_endstop=((READ(Z_MIN_PIN) && READ(Z2_MIN_PIN)) != Z_MIN_ENDSTOP_INVERTING);  //Using the Z_Max pin as the second Z_Min!!!!!!!!!!!!
				#else
					bool z_min_endstop=(READ(Z_MIN_PIN) != Z_MIN_ENDSTOP_INVERTING);
				#endif
			  if((z_min_endstop && old_z_min_endstop) && (current_block->steps_z > 0)) {
				endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
				endstop_z_hit=true;
				step_events_completed = current_block->step_event_count;
			  }
			  old_z_min_endstop = z_min_endstop;
			#endif

Now only the connection in z_max works for me but it doesn't work z_min.

Help¡¡

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle