MarlinFirmware/Marlin #10162: driver error TMC 2130

meav33, 2018-03-20:
Hello,
i`m getting seemingly random "driver errors" with my tmc 2130.
The LCD only shows "driver error" and i cant really get a clue from the seriel out also.
I sometimes happes while printing but sometimes also while ideling.

My drivers are actively cooled and i`m using the altest 1.1.x bugfix

`X driver error detected:
		X	Y
Enabled		true	true
Set current	1100	1100
RMS current	1049	1049
MAX current	1479	1479
Run current	18/31	18/31
Hold current	3/31	3/31
CS actual		18/31	18/31
PWM scale	112	132
vsense		0=.325	0=.325
stealthChop	true	true
msteps		16	16
tstep		389	202
pwm
threshold		0	0
[mm/s]		-	-
OT prewarn	false	false
OT prewarn has
been triggered	false	false
off time		5	5
blank time	24	24
hysterisis
-end		2	2
-start		3	3
Stallguard thrs	13	13
DRVSTATUS	X	Y
stallguard
sg_result		0	0
fsactive
stst
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:	X = 0x00:12:00:00
	Y = 0x00:12:00:00
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!`

teemuatlut, 2018-03-20:
I will need to change a few things to clear situations like these because I can't see anything wrong either. But for some reason the driver did report an error and Marlin reacted to it.

You can disable `STOP_ON_ERROR`. The main purpose for it is to save time and material on situations where there has been an error and the driver has shut itself down. Before this feature Marlin of course didn't know that a driver needs a reset and just kept printing thinking everything was fine.

sh4jonny, 2018-03-21:
Do you use SD card ?

Bob-the-Kuhn, 2018-04-05:
Try putting a 10K pulldown on the MISO line.  That solved TMC2130 problems on a couple of five TMC2130 systems that were using a daisy chained cable.

hevilp, 2018-06-08:
I figured out: There came with more weight on the X-axis. I changed from bowden to direct drive and those 200-300 gramms brings up the error.

TMC2130 can not handle a direct drive weight?

hevilp, 2018-06-09:
And again, 2 got 2 fans to cool them and got bowden setup...
```
< 11:37:00.405: N30137 G1 X245.500 Y138.679 E188.8947
> 11:37:00.450: X:52 |0b0|  Y:102 |0b0|  E:53 |0b0|  
> 11:37:00.845:  T:219.29 /218.00 B:70.33 /70.00 @:70 B@:0
> 11:37:00.950: X:52 |0b0|  Y:104 |0b0|  E:53 |0b0|  
> 11:37:01.450: X:54 |0b0|  Y:103 |0b0|  E:52 |0b0|  
> 11:37:01.844:  T:217.89 /218.00 B:70.39 /70.00 @:95 B@:0
> 11:37:01.949: X:54 |0b0|  Y:103 |0b0|  E:51 |0b0|  
> 11:37:02.406: echo:busy: processing
> 11:37:02.450: X:53 |0b0|  Y:103 |0b0|  E:52 |0b0|  
> 11:37:02.546: ok
< 11:37:02.547: M117 ETA 18:20:19 day 10
> 11:37:02.591: ok
< 11:37:02.592: N30138 G1 X225.541 Y158.637 E190.2677
< 11:37:02.592: N30139 G92 E0
> 11:37:02.615: ok
> 11:37:02.615: ok
< 11:37:02.616: N30140 G1 E-6.5000 F3600
< 11:37:02.616: N30141 G1 X206.998 Y176.539 F6000
> 11:37:02.667: ok
< 11:37:02.667: N30142 G1 E0.0000 F3600
< 11:37:02.667: N30143 G92 E0
> 11:37:02.845:  T:218.25 /218.00 B:70.44 /70.00 @:87 B@:0
> 11:37:02.950: X:50 |0b0|  Y:106 |0b0|  E:52 |0b0|  
> 11:37:03.450: X:51 |0b0|  Y:108 |0b0|  E:52 |0b0|  
> 11:37:03.844:  T:217.52 /218.00 B:70.30 /70.00 @:99 B@:0
> 11:37:03.949: X:52 |0b0|  Y:104 |0b0|  E:52 |0b0|  
> 11:37:04.053: X:225.54 Y:158.64 Z:2.97 E:0.00 Count X:18043 Y:12691 Z:2376
> 11:37:04.053: ok
> 11:37:04.058: ok
< 11:37:04.059: N30144 G1 X206.998 Y182.989 E0.3137 F2400
> 11:37:04.061: ok
> 11:37:04.065: ok
< 11:37:04.065: N30145 G1 X205.438 Y182.989 E0.3896
> 11:37:04.459: X driver error detected:
> 11:37:04.460: overtemperature
> 11:37:04.462: short to ground (coil A)
> 11:37:04.462:   X Y E0
> 11:37:04.463: Enabled  true true true
> 11:37:04.463: Set current 800 800 500
> 11:37:04.466: RMS current 795 795 489
> 11:37:04.468: MAX current 1121 1121 689
> 11:37:04.470: Run current 25/31 25/31 15/31
> 11:37:04.471: Hold current 12/31 12/31 7/31
> 11:37:04.472: CS actual  31/31 31/31 15/31
> 11:37:04.472: PWM scale 69 151 49
> 11:37:04.474: vsense  1=.18 0=.325 0=.325
> 11:37:04.475: stealthChop true true true
> 11:37:04.478: msteps  16 16 16
> 11:37:04.479: tstep  137 4294901888 1048575
> 11:37:04.479: pwm
> 11:37:04.479: threshold  98 98 263
> 11:37:04.482: [mm/s]  100.85 100.85 30.06
> 11:37:04.483: OT prewarn false false false
> 11:37:04.484: OT prewarn has
> 11:37:04.487: been triggered false false false
> 11:37:04.487: off time  5 5 5
> 11:37:04.488: blank time 24 24 24
> 11:37:04.489: hysterisis
> 11:37:04.489: -end  2 2 2
> 11:37:04.490: -start  3 3 3
> 11:37:04.491: Stallguard thrs 0 0 0
> 11:37:04.491: DRVSTATUS X Y E0
> 11:37:04.493: stallguard    
> 11:37:04.493: sg_result  0 0 0
> 11:37:04.493: fsactive    
> 11:37:04.495: stst    X
> 11:37:04.495: olb    
> 11:37:04.495: ola    
> 11:37:04.496: s2gb    
> 11:37:04.497: s2ga    
> 11:37:04.497: otpw    
> 11:37:04.497: ot    
> 11:37:04.498: Driver registers:
> 11:37:04.499:  X = 0x00:19:00:00
> 11:37:04.499:  Y = 0x00:19:00:00
> 11:37:04.501:  E0 = 0x80:0F:00:00
> 11:37:04.503: Firmware was halted, trying to reconnect. Eventually running print is stopped.
```

teemuatlut, 2018-06-09:
https://github.com/teemuatlut/Marlin/tree/TMC_debugging_v2
Please try my branch with updated monitoring code.
It makes only one call to the driver to determine the error conditions. This should/could improve detecting valid error states.

hevilp, 2018-06-10:
I'm running the Marlin 2.x.x. from friday and it is running flawless now about 15 hours. (previous was marlin 1.1.8), I will try your branch next days.

dventu, 2018-06-14:
Do you mind sharing your marlin folder as Im having the same issue.

JonnyLeRoy, 2018-06-14:
I don't know where to post my question. This looks like the correct place.
Ramps 1.4/ Marlin bugfix 2.0/ TMC2130s X,Y,Z,E0,E1

Every thing worked with 2130s only on X,Y,Z. Now with E0,E1. Every thing is okay until I add power.

My pins are E0 63 E1 48
X 59 Y 64 Z44

Power Off

Printer is now online.
echo: External Reset
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Jun 13 2018
echo: Free Memory: 3058 PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo: G21 ; Units in mm (mm)
echo: M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo: M200 D3.00
echo: M200 T1 D3.00
echo: M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E90.00
echo:Maximum feedrates (units/s):
echo: M203 X500.00 Y500.00 Z2.00 E2500.00
echo:Maximum Acceleration (units/s2):
echo: M201 X500 Y500 Z100 E2500
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo: M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo: M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo: M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo: M145 S0 H180 B70 F0
echo: M145 S1 H240 B110 F0
echo:PID settings:
echo: M301 P22.20 I1.08 D114.00
echo:Z-Probe Offset (mm): (mm):
echo: M851 Z0.00
echo:Stepper driver current:
echo: M906 X800
M906 Y800
M906 Z800
M906 T0 E800
M906 T1 E800
echo:Filament load/unload lengths:
echo: M603 T0 L930.00 U1000.00
echo: M603 T1 L930.00 U1000.00
>>> m122
SENDING:M122
X Y Z E0 E1
Enabled false false false false false
Set current 800 800 800 800 800
RMS current 1436 1436 1436 1436 1436
MAX current 2025 2025 2025 2025 2025
Run current 25/31 25/31 25/31 25/31 25/31
Hold current 12/31 12/31 12/31 12/31 12/31
CS actual 0/31 0/31 0/31 0/31 0/31
PWM scale 0 0 0 0 0
vsense 0=.325 0=.325 0=.325 0=.325 0=.325
stealthChop false false false false false
msteps 256 256 256 256 256
tstep 0 0 0 0 0
pwarn false false false false false
OT prewarn has
been triggered false false false false false
off time 0 0 0 0 0
blank time 16 16 16 16 16
hysteresis
-end -3 -3 -3 -3 -3
-start 1 1 1 1 1
Stallguard thrs 0 0 0 0 0
DRVSTATUS X Y Z E0 E1
stallguard
sg_result 0 0 0 0 0
fsactive
stst
olb
ola
s2gb
s2ga
otpw
ot
Driver registers: X = 0x00:00:00:00
Y = 0x00:00:00:00
Z = 0x00:00:00:00
E0 = 0x00:00:00:00
E1 = 0x00:00:00:00

;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;

Power On

00:00: X driver overtemperature warning! (800mA)
Z driver error detected:
overtemperature
short to ground (coil A)
short to ground (coil B)
X Y Z E0 E1
Enabled false false false false false
Set current 800 800 800 800 800
RMS current 795 1436 795 795 1436
MAX current 1121 2025 1121 1121 2025
Run current 25/31 25/31 25/31 25/31 25/31
Hold current 12/31 12/31 12/31 12/31 12/31
CS actual 31/31 0/31 31/31 31/31 0/31
PWM scale 255 0 255 255 0
vsense 1=.18 0=.325 1=.18 1=.18 0=.325
stealthChop true false true true false
msteps 0 256 0 0 256
tstep 4294967295 1048575 4294967295 4294967295 1048575
pwm
threshold 0 0 0 0 0
[mm/s] - - - - -
OT prewarn true false true true false
OT prewarn has
been triggered true false false false false
off time 15 0 15 15 0
blank time 54 16 54 54 16
hysteresis
-end 12 -3 12 12 -3
-start 8 1 8 8 1
Stallguard thrs 0 0 0 0 0
DRVSTATUS X Y Z E0 E1
stallguard X X X
sg_result 1023 0 1023 1023 0
fsactive X X X
stst X X X X X
olb X X X
ola X X X
s2gb X X X
s2ga X X X
otpw X X X
ot X X X
Driver registers: X = 0xFF:FF:FF:FF
Y = 0x80:00:00:00
Z = 0xFF:FF:FF:FF
E0 = 0xFF:FF:FF:FF
E1 = 0x80:00:00:00
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

JonnyLeRoy, 2018-06-14:
Now with 10k pull up to MISO. And power off.

connecting...
start
Printer is now online.
echo: External Reset
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Jun 13 2018
echo: Free Memory: 3058  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 T1 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E90.00
echo:Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z2.00 E2500.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X500 Y500 Z100 E2500
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Z-Probe Offset (mm): (mm):
echo:  M851 Z0.00
echo:Stepper driver current:
echo:  M906 X800
  M906 Y800
  M906 Z800
  M906 T0 E800
  M906 T1 E800
echo:Filament load/unload lengths:
echo:  M603 T0 L930.00 U1000.00
echo:  M603 T1 L930.00 U1000.00
X driver error detected:
overtemperature
short to ground (coil A)
short to ground (coil B)
		X	Y	Z	E0	E1
Enabled		false	false	false	false	false
Set current	800	800	800	800	800
RMS current	795	795	795	795	795
MAX current	1121	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31	12/31
CS actual		31/31	31/31	31/31	31/31	31/31
PWM scale	255	255	255	255	255
vsense		1=.18	1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true	true
msteps		0	0	0	0	0
tstep		4294967295	4294967295	4294967295	4294967295	4294967295
pwm
threshold		0	0	0	0	0
[mm/s]		-	-	-	-	-
OT prewarn	true	true	true	true	true
OT prewarn has
been triggered	false	false	false	false	false
off time		15	15	15	15	15
blank time	54	54	54	54	54
hysteresis
-end		12	12	12	12	12
-start		8	8	8	8	8
Stallguard thrs	0	0	0	0	0
DRVSTATUS	X	Y	Z	E0	E1
stallguard		X	X	X	X	X
sg_result		1023	1023	1023	1023	1023
fsactive		X	X	X	X	X
stst		X	X	X	X	X
olb		X	X	X	X	X
ola		X	X	X	X	X
s2gb		X	X	X	X	X
s2ga		X	X	X	X	X
otpw		X	X	X	X	X
ot		X	X	X	X	X
Driver registers:	X = 0xFF:FF:FF:FF
	Y = 0xFF:FF:FF:FF
	Z = 0xFF:FF:FF:FF
	E0 = 0xFF:FF:FF:FF
	E1 = 0xFF:FF:FF:FF
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

teemuatlut, 2018-06-14:
TMC drivers cannot communicate with Marlin if they don't have 12V power.

JonnyLeRoy, 2018-06-14:
How can isolate the fault?

hevilp, 2018-06-14:
Just disable Monitor_driver_status and try a long print

thinkyhead, 2018-06-14:
@teemuatlut — Is there some way we can check that a driver doesn't have any 12V power (in `tmc_status`, etc.) and output something like `---` for that driver? And/or can we check for 12V power being off to all the drivers and if `AUTO_POWER_CONTROL` is enabled, turn them on?

teemuatlut, 2018-06-15:
When the drivers don't have power the failed communication is indistinguishable from a faulty wiring or bad configuration. There were some specifics on what makes the SPI response be all zeros versus all ones.

The Marlin power control features are the only ways we can know if the power is turned on.
If the user has enabled `PS_DEFAULT_OFF` or `AUTO_POWER_CONTROL`, then I think the best course of action would be to turn on the PSU on startup long enough for Marlin to configure all the drivers and test working communication. The written configurations should persist even if VMOT power is lost.
This would also simplify some things as `power_on` would no longer need to worry about TMC drivers at all.

This does not solve a setup where the user would first power the board from USB and then manually switch power on before starting a print, or if it is controlled by OctoPrint, but we could implement a quick `M` code to refresh the drivers and the code could be inserted into the start gcode section. This would also require us to gracefully handle a failed comms check on startup.

JonnyLeRoy, 2018-06-15:
After some testing. It is clear to me. Configuring 5 pins at the same time is not the best method.  I will add one pin at a time. And report my finding. too bad I cut up my harness in a bit of rage.
Thanks to all. For all you help.

thinkyhead, 2018-06-15:
> Are there any pins that are not compatible with CS

Any of the free digital pins should be fine.

JonnyLeRoy, 2018-06-15:
!!!!!!!!!!!!!!!!!!!WTF!!!!!!!!!!!!!!!
I found a Titan Beetle. "Bug"
 Arduino IDE 1.9 build 44 and or Marlin is assigning random  pin numbers.
Short version. I finally understand what causes the system shutdown. 
So I striped my RAMPS down to just the 2130 (E1). Then I connect the CS pin 48. That where I set in Marlin. 
With the power on I reset RAMPS and try all the pins on AUX 2 one at a time. Until I find the correct pin.
The correct pin is 44.  I'm not drunk but I am tired. I hope I not crazy.
My CS pins are:  X 44 , Y 64 , Z 66 , E0 4 , E1 48

JonnyLeRoy, 2018-06-16:
I found this article on the 2130s when I was looking for a way to test one. 
Is there a way to test the 2130s to know if this is your issue? I have had my 2130s on stand buy with USB power for 4 weeks.

Boards with USB Power Supply
Only applicable for SilentStepSticks with variable 3-5V logic voltage (VIO): If you use a control board with USB power supply (like Arduino + RAMPS) then always ensure that the motor supply voltage (VM) is present, when you connect the board via USB. Otherwise the TMC2xxx is not powered via the internal voltage regulator and a high current can flow into VIO or the IOs and this can damage the internal logic. As safety workaround you can disconnect the 5V signal in the USB cable, so that the board cannot be powered over USB.
https://learn.watterott.com/silentstepstick/faq/

unixb0y, 2018-08-30:
I'm not sure if this is the right place to ask, but since somebody mentioned the SD card as a possible cause of error:
My 2130's run great and everything and I do have a smartcontroller with SD card slot attached to the RAMPS as well by just swapping the female AUX-3 connector with one that has very long legs so they stick all the way through the LCD connector.
But the issue is: As soon as I plug in an SD card, I see a "driver error" message on the screen and the printer is halted. 
I haven't checked / configured the SD connection prior to configuring my printer for TMC2130 so it's possible that some SD card settings are wrong and produce this issue somehow. 
Any help is appreciated, thank you in advance!

JonnyLeRoy, 2018-09-02:
unixb0y, I'm glad you asked.
 I replaced the SmartController. And that fixed the same issue on my system.

unixb0y, 2018-09-03:
@JonnyLeRoy thank you very much, I’ll try to borrow one to see if it’s the smart controller in my case as well!

closed 2018-09-05 by thinkyhead