MarlinFirmware/Marlin #10174: [2.0] SD Init Fail

michaelwu21, 2018-03-20:
### Description

When trying to get z-probe offset to work, I found that it would never change no matter what value I have it set to. I found that the issue lies in that the there is no eeprom.dat file that is being created on the sd card. 
Looking at terminal output, I see 2 repetitions of "SD Init Fail"

### Steps to Reproduce
Using default mks sbase config:
- Reprap Discount Controller(Removing it did not fix the issue)
- Added billinear bed leveling
- Enabled eeprom and sd card support
- Changed serial_port to -1 and commented out serial_port_2
- 32gb SD card in regular board slot with FAT32 format

**Expected behavior:**
SD card works and eeprom data is saved after a power cycle
**Actual behavior:** 
No Eeprom.dat file is created and SD card does not initialize. 
#### Additional Information
Using Marlin 2.0 from 2 days ago
I've tried disabling the reprapdiscount controller
Switched the order of the spiBegin() and pinMode() in sd2card.cpp
When doing M500, I do not get any errors and M503 shows the changes I have made, yet they disappear after a power cycle and does not show up, which is most likely due to the sd init fail.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1831364/Configuration.zip)

p3p, 2018-03-20:
This has come up before with with the mks sbase, but I don't have one to test with (only a Re-ARM), the "SD Init Fail" you are seeing are not caused by the onboard sd card that stores the eeprom.dat file, only the controller sdcard.

Can you try with both the controller and sdcard support disabled?
Does the usb mass storage device function correctly?

michaelwu21, 2018-03-20:
I'll try out disabling those once I'm finished with this print.

What would you mean by usb mass storage device function correctly? The SD Card? Windows says there's an issue with it but windows says that for every sd card I've used.

p3p, 2018-03-20:
When you plug the mks sbase into a computer it should show up as a usb drive showing the contents of the sdcard, that's how you flash new firmware? if that works the Marlin can access the sdcard.

michaelwu21, 2018-03-20:
Ah, I always flash by copying the files to the sd card.

I've never gotten it to work with windows, it always gives me a malfunction error. At first, I was able to access it via my computer to configure smoothie, but after the 2nd time doing this, it's never been able to connect to windows. Tried reinstalling mks drivers but it didn't fix the issue.

michaelwu21, 2018-03-21:
@p3p I've tried without controller and sdcard enabled, but eeprom is still not stored in the sd card.

AnHardt, 2018-03-21:
MKS BASE is  Atmega2560 based. So there will never be a Eeprom.dat. The normal EEPROM of the processor should be used.

p3p, 2018-03-21:
@AnHardt MKS called their LPC1768 (smoothieboard clone) based board the SBASE just to confuse people. 

@michaelwu21 the controller unplugged as well? I'm not aware of any difference with the onboard sd card between the re-arm and the sbase ..

michaelwu21, 2018-03-21:
Yes the controller is unplugged

p3p, 2018-03-21:
@Bob-the-Kuhn you have recently updated the config files for the mks sbase does that mean you own one?
@Spawn32 the onboard sdcard was working in your original pins and config port wasn't it?

After a quick look at the schematic there is a difference with the Re-ARM in that the sdcard on the controller shares the spi port that the onboard sdcard uses, this is why I was making sure there was no controller interfering and Marlin was not configuring those pins incorrectly. As you have the sbase configured now it should work the same as the Re-ARM, unless something else in the pins files is conflicting even with the features disabled...

The only error I can see in the mks sbase pins file is 
`#define SDSS               P0_06`
as it should be
`#define SDSS               P0_28`

and the readme in the example config folder mentions difficulties with the Reprap Discount Full Graphic Controller and needing to make a custom cable, but that shouldn't matter with it all disconnected and the features disabled.

Bob-the-Kuhn, 2018-03-21:
I have one on order.  Just seeing too many issues with it & debugging without one is slow.

The LCD & it's SD card need special cabling if it's to be functional when running Marlin.  They need to pick up MISO, MOSI & SCK from somewhere else.

There's a READ ME file in the example directory on this subject with a typical solution.

p3p, 2018-03-21:
The usb mass storage device assumes it has complete control of the SSP1 peripheral and the associated pins (as that is how the Re-ARM is) so I can see why they would need moved when a controller was in use,   but I was all this worked origional for Spawn32, although it must have been bugger under some conditions unless the schematic I'm reading is wrong .. Anyway it doesn't change the fact it should work with those features disabled.

Spawn32, 2018-03-21:
Yes, the on board sdcard has always worked, i think it's enabled by the bootloader, buuuttt my board always does a reset a few seconds after i do a M50X command, very annoying, read, writes the eeprom.dat file, but always does a reset after it's done :/
Can test anything on it for you guy's at the moment, blown main regulator, have order a new one :)

Spawn32, 2018-03-21:
gave up on getting a external sdcard to work, the shared card detect messes everything up :)

Bob-the-Kuhn, 2018-03-21:
Chris - could you post the schematic or point me to it?

Bob-the-Kuhn, 2018-03-21:
Do the M500 then M501 commands return the same CRC?

An external SD card controller can't be used to store the EEPROM data because the SDSS signal for the onboard SD card isn't available externally.

p3p, 2018-03-21:
It's not official, but the only source I could see.

https://github.com/pixel3design-hub/MKS-SBASE-FULL-DOCUMENTED/blob/master/MKS%20SBASE%20SCH_V12.pdf

> the shared card detect messes everything up

That shouldn't be an issue the onboard sd card has no card detect anyway?

> An external SD card controller can't be used to store the EEPROM data because the SDSS signal for the onboard SD card isn't available externally.

That will be true until marlin has more robust support for multiple disks (rather than my tacked on solution just to make it work), the main issue with both sdcards being on the same SSP channel is that the usb storage device requires access to the disk using interrupts, so could interlace access while the other sd card is being accessed breaking everything.

Spawn32, 2018-03-21:
wow, thanks for the schematic @p3p, been looking for that a long time :)
it's for the 1.2, a few mistakes was fixed on the current 1.3, but dont remember what it was:/

yes it has a card detect, when i was testing this out with @Bob-the-Kuhn lcd wiring fix, the lcd would detect a new card whenever i removed / inserted the on board and / or the sd card on the lcd, i gave up, LOL

p3p, 2018-03-21:
Ah ok yea that will happen for the controller sdcard on the sbase, but it shouldn't cause any issue with the onboard as it is always assumed to be installed (it's a requirement)

Bob-the-Kuhn, 2018-03-23:
Long shot

I haven't seen mention of re-formatting the card or trying a different card.

I just spent 12 hours ☹️ trying to figure out what was wrong with the DUE software only to find out my SD card's FAT was corrupted.  It worked fine in Windows and on the LCD when the host wasn't connected.  The only time it didn't work was when the host was connected to the DUE and then I got  "SD Init Fail".

The Marlin LCD software doesn't use the SD code modules's FAT routines so it didn't care.  I haven't figured out how it's done.  I just know it's done differently.

michaelwu21, 2018-03-23:
Unfortunately that is my only sd card under 32 gigabytes I can use. 

Yes, I have tried reformatting it using sd card formatter(on windows)

Bob-the-Kuhn, 2018-03-24:
You can repartition the bigger SD card so that it has only 32G or smaller partitions.  That also works.

michaelwu21, 2018-03-28:
Didn't really make a difference, tried with 64gb sd card partitioned into 32 gb parts.

No eeprom.dat

p3p, 2018-04-21:
@michaelwu21 can you confirm that you have formatted a 32Gb sd-card with https://www.sdcard.org/downloads/formatter_4/index.html (make sure it doesn't format as exfat), in the config disable EEPROM and SDSUPPORT. When you plug the mks sbase into a computer via usb it should now show up as a usb mass storage device, 

If that is not the case the last thing we can try is reducing the frequency the onboard sd-card tries to run at, in the file `frameworks/CMSIS/LPC1768/lib/chanfs/mmc_ssp.c` change `#define SCLK_FAST 25000000UL` to `#define SCLK_FAST 2000000UL`

thinkyhead, 2018-11-05:
Assuming this is now solved.

closed 2018-11-05 by thinkyhead