MarlinFirmware/Marlin #10175: Maxtemp and Mintemp not triggering when thermistors disconnected

flyingscotsman12, 2018-03-20:
### Description

In Marlin 1.1.8 and the most recent bugfix (with a completely virgin install), disconnecting the thermistor leads does not trigger a Mintemp or Maxtemp error. This is obviously a serious safety problem.

### Steps to Reproduce

The first case, with a custom Marlin Configuration:
1. A custom configuration of Marlin 1.1.8 was uploaded to the MKS Gen 1.4, with a PT100 amplifier board for the hotend sensor, 100k bed thermistor
2. The LCD displayed the temperature as expected, ~20C for bed and hotend.
3. When either or both of the sensors were unplugged, the LCD displayed the open-circuit temperatures (320C for the PT100 amplifier, -14C for the bed ) 
4. No error was triggered, the serial connection was not interrupted, the ATX supply remained activated

The second case, with a virgin install:
1. A completely unmodified version of Marlin 1.1.8 Bugfix (20-Mar-2018) was flashed to the MKS Gen 1.4 controller board (same pins as the default RAMPS 1.4 EFB)
2. The board was connected to Pronterface to read the temperature values and detect errors
3. The extruder thermistor (standard 100k) reading was observed to be ~20C
4. The extruder thermistor was unplugged
5. The connection continued, no mintemp error was triggered, extruder reading was ~-14C, well below the mintemp of 5C.

A third test:
1. Using the custom 1.1.8 configuration
2. The ATX supply was turned off, only the 5V standby fed the MKS Gen 1.4. No 12V power was applied to the board which should prevent hotend heating
3. The hotend temperature and bed temperature both read 20C, as expected
4. Through the LCD, the hotend temperature was commanded to be 235C
5. The hotend temperature did not change
6. After a few seconds, a "heating failed" error was triggered, disabling the printer and displaying the error on the LCD. The printer had to be reset as expected.

Please see the attached .ZIP with the configuration files, as well as an image of the LCD with the illegal temperatures.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1831389/Config.zip)

AnHardt, 2018-03-20:
Since the integration of the high temperature thermistors the related heater has to be on to cause an error.

flyingscotsman12, 2018-03-20:
Is there a way to disable that "feature"? It means that the firmware has no way of dealing with failed-closed MOSFETs or relays, which in the old firmware would cause a power supply or safety relay to shut off if the temperature exceeded the maxtemp, regardless of whether the heater was active. 
It does seem to handle unplugged amplifier boards well though, upon further testing.

dot-bob, 2018-03-22:
Perhaps in a failed mode where we have uncontrolled heating if we have no way of cutting the power we turn all fans 100% and start blaring a warning tone out the speaker.

At the very least we probably should add a continuous warning tone on the thermal overrun halt regardless.

thinkyhead, 2018-04-01:
Assuming that Marlin is alive, it can certainly be made to watch for temperature rises when the heaters are supposed to be off. The typical response would be to reset the board, but if that also fails to stop the heaters then the option described by @dot-bob would indeed be the only one left: fire alarm!