MarlinFirmware/Marlin #10180: Release strategy and use of branches.

joris57, 2018-03-21:
As a medium-term Marlin user, I very much appreciate the features that have been recently added to Marlin and the efforts that are put in by the developers.

But I also frequenly despair because of the lack of stability of the code, when it comes to plainly using my printers (and not regression testing Marlin).

Would it not make sense to split the branches for bugfixing of the current stable release off from the development branch? At the moment, it looks like both activities are running in bugfix-1.1.x, resulting in a massively unstable branch (bugfix-1.1.x) and a stable branch that doesn't get it's bugs fixed for months (1.1.x). I refer to the approach used by the Debian project.

I'd love to hear the views from other users and the development team.

Roxy-3D, 2018-03-22:
The Marlin code base is very large.   And the development team is very small.    

I understand your thinking.   And I'm actually in agreement.   But the problem is everybody contributing is a  'volunteer'.  They work on things that are of interest to them.   So just telling those developers how things have to be is going to be counter productive.

I am willing to work with you to accomplish what you want.   But it can't put extra barriers in front of the developers.   One idea would be to organize enough people that share your view and I'll empower that group to maintain a branch that is 'solid' and fully tested.   I'll even go so far as to say none of the developers will touch your branch.   Only your test and certification group can make changes to the branch.   My only constraint would be I want this (currently imaginary) branch to follow closely behind the bugfix branch.  Maybe within 3 or 4 weeks....

If you are willing to expend effort to make it happen...   I will help (and empower) you to make it happen.

TheSFReader, 2018-03-22:
@Roxy-3D I'm not sure the "imaginary" branch following closemly the "bugfox" branch would be what joris is thinking about, but as I understand the text,, a branch that consists of ONLY bug fixes to the last official release, excluding any new feature. 

So when say 1.10.0 is out, a new branch 1.10.0-bugfix is created, on which only bugfixes (maybe originating from the 1.1.x-bugfix "dev branch") are applied. That way, people who want/need the most stable version get that one, while more adventurous people get the "1.1.x-bugfix"/"2.x-bugfix" with a more up-to-date feature set. 

I'm not sure how many people would take the load for that, but while, I, as a developper understand the need to go forward.  as a user I would really welcome a way to get a clean/stable version when I need to install on my primary "working" 3D printer.

Roxy-3D, 2018-03-22:
Well...  We can debate what the best mechanism is to accomplish the goal.   Having a team of people that check out and verify the 'correctness' of different points of time in the bugfix branch is the key item.   The stable or release branch tag could be moved to that point in time.

I really don't care how we accomplish the goal.   But we do need to make progress in that direction.

TheSFReader, 2018-03-23:
Only problem with the moving slider is that at any point on the "bugfix" branch, new features or major updates can be introduced, potentially introducing regressions, whereas applying/backporting the known fixes to a specific commit should _converge_ to a "mostly correct/stable" version.

joris57, 2018-03-23:
So I envisage 2 workflows and 4 formal branches :

1) Bugfix workflow : bugfixes are developed and tested against the 'bugfix' branch and applied to the 'stable or production' branch, the 'development' branch and the 'testing' branch.

2) Development workflow : new/improved functionality is developed and unit tested against the 'development' branch and then applied to the 'testing' branch for integrated testing and deployment by users, who like living on the bleeding edge and need the new/improved funcionality.

At agreed points in time, testing branch gets frozen for final testing and stabilisation (only bugfixes get applied) until deemed stableand ready for the larger audience. Then commits are applied to stable and bugfix branches.

If a testing team could be marshaled, that would be great, but I believe that even the separation of the bugfix and development workflows by itself will contribute to stability.

Makes sense?

AnHardt, 2018-03-24:
The real art is to judge  if a PR solves more problems than it causes - how risky it is.
There are low risky new features like adding a new board where errors likely only affect these new feature.
And there are fixes for seemingly minor bugs breaking Marlin completely. 

The art to jude this isn't spread widely - especially not, when it's your own PR.

fiveangle, 2018-03-24:
[image]

joris57, 2018-03-24:
@AnHardt I agree with your last statement and would even go so far that in practical terms, it's impossible (based on some very bad experiences in my professional life).

However, what we **can** do is:
1) Make the bugfixing environment less 'fluid', by segregating it from the development environment. I don't think you can argue that that would not contribute to the **quality and/or risk** of the bugfixes.
2) Limit the changes to the 'stable' branch to bugfixes, which limits the **number** of changes.

Those who still want the new functionality, can opt for the 'testing' branch (but will still have a fall back if things go belly-up).

If at a later stage, here are types of functional improvement that are deemed harmless (like introduction of new boards), exceptions can still be agreed upon.

@fiveangle Meaning?

fiveangle, 2018-03-24:
@joris57 - you're going to drag me into a real answer huh ?

It's a long-contentious topic that ultimately never appears to make progress. The trouble is there isn't anyone interested in actually working toward that goal, which is really just seen as additional red tape to many.  Who can blame volunteers for wanting to focus on the more glamorous meat of functionality vs the mundane red-tape and consistent discipline to adequately manage release cycles with user experience in mind as primary goal rather than nice-to-have ? It is completely understandible (and I'm not being sarcastic).

Sensitive to that fact, some have tried to offer mechanisms that help prevent at least some of the most gross regressions in Marlin without any additional human oversight (just one small area of the problem being discussed) and for reasons only known to Scott were not accepted.  Like switching Travis to fail on warning instead of allowing warnings to proceed.  Warnings typically grow into issues that then require major unwinding later.

Frustrated by continually having to clean up regressions for my Printrboard configuration and other hamstrung boards (Anet, etc), I spent an entire weekend fixing all the many broken example configurations and then submitted with all example configs enabled for TravisCI  (#8784).  Scott could not accept the increased build time so he merged the example config fixes but made no change to TravisCI tests.  To try and overcome the build time issue (which I admit was frustratingly long, @approx 23min with the added example config compiles), I researched and found CircleCI would run in parallel to TravisCI to compile just the example configs and finish before the current standard TravisCI config, so I submitted #8806.  It was merged but was not enabled for the Marlin repo (so essentially is only useful for people like me who fork Marlin and enable CircleCI on their own repo, which only allows them to find the regressions later, essentially making the merge pretty much worthless to prevent regressions in mainline. 

The constant river of regressions that plague all but the primary mega targets wear some of us down (it did me).  I swear Printrboard build would not go more than 6 weeks before breaking from some optimization or change that didn't take into consideration all the lesser-capable boards and would have handily been caught by the example configs being in CI if it were allowed to.

Scott always has best intentions but in the case of my suggestion of compiling every example config he rightfully pointed out that it was not an elegant approach and that Marlin needs a methodical set of tests rather than the brute force method.  That's all well and good but by Scott's own argument, who will step up and provide the work toward that elegant solution ?  In lieu of an elegant solution, why not enable the solution provided for now, and replace with a better one later if/when someone finds the initiative ?   It's this double-standard that turns people away from helping with Marlin.  I've pretty much given up on tracking Marlin and have stuck with an old build with Linear Advance + MBL and won't upgrade unless I hit a show-stopper, since upgrading will result into me having to track down a handful of Printrboard regressions.  A good friend has been on Marlin 1.1 RC4 (from 2016 ?) for the same reason and I used to tout to him all the improvements in Marlin since then (UBL, serial flow control, improved Linear Advance, etc) but in the end I realized it was he who was having the last laugh.

Only if someone passionate about managing the release cycle in the way you describe steps up to the plate and shows both consuming interest and deep aptitude will this come to be.  It just isn't in the DNA of the very small team who I'm sure are doing the best they feel they can.

joris57, 2018-03-24:
@fiveangle I was hoping for a more substantial contribution, but ok.

thinkyhead, 2018-03-28:
The "Pull Request" feature of GitHub obviates the need to maintain multiple branches in the main repository. The contributor simply keeps the branch in their own fork. Hopefully they keep it up to date while it's being evaluated, but to do this effectively requires `git rebase` which is not supported by the Github Desktop application and not well-understood by 99% of contributors. (As a result, the project Owners end up doing the necessary rebasing to keep the more useful PRs from getting stale — a power granted to us by the creation of a PR.)

boelle, 2019-02-20:
@thinkyhead has this one not served its purpose?

joris57, 2019-03-03:
@boelle With all due respect, but I've not seen any progress on this and, more disappointing, not even the recognition of the issues with the current approach. I'll close the issue.

closed 2019-03-03 by joris57

thinkyhead, 2019-03-06:
At the moment the `bugfix-2.0.x` branch is the dev branch for `2.0.x`. Too late to rename it. The `bugfix-1.1.x` branch is the bug-fix branch for 1.1.9. There will be no further 1.1.x releases. After the 2.0.0 release we will revisit the question of whether to keep a separate `bugfix-2.0.x` and `dev-2.0.x` branch. But for now, the single branch is the only one we need.