MarlinFirmware/Marlin #10182: [bugfix-2.0.x] EEPROM datasize error

Rexima, 2018-03-21:
I got an  EEPROM datasize Error and i added this line to my configuration.h
**#define DEBUG_EEPROM_READWRITE**

Than i got this error on my octoprint **Field planner_extruder_advance_K mismatch.**

I have an MKS Gen 1.4 with 5 TMC2208

Roxy-3D, 2018-03-22:
The problem is also in bugfix-1.1.x.     The size is off by 22 bytes.   In my case:
```
echo: Index 629 Size: 651
```
I think this happened as a result of this commit:   https://github.com/MarlinFirmware/Marlin/commit/84bbd018ab7177c44e323a25a2ec014faa4c79ef

The size changed by 22 bytes because of
```
-  uint16_t tmc_stepper_current[11];                     // M906 X Y Z X2 Y2 Z2 E0 E1 E2 E3 E4
```

@thinkyhead

vMeph, 2018-03-22:
READ: Error:EEPROM datasize error.
READ: echo:Index: 629 Size: 651

getting error latest bug fix 1.1x

Index: 629 Size: 651
READ: Unified Bed Leveling System v1.01 inactive.
READ: echo:Hardcoded Default Settings Loaded
Hardcoded Default Settings Loaded

Roxy-3D, 2018-03-22:
Yeah...   I'm looking for the 'Revert' button...  But I can't find it.

tximy, 2018-03-22:
I have same error too I make M502, M500, and at M501 receive this error (Field planner_extruder_advance_K mismatch.) at

lienbacher, 2018-03-22:
Is there any hotfix for this?

Roxy-3D, 2018-03-23:
I think it is fixed now...    But I did not test it.  (I have to run out the door....)

If it is not fixed....   Please post a message!!!!!!

lienbacher, 2018-03-23:
seems to work fine! neither firmware update nor M502 and M500 threw an error.

tximy, 2018-03-23:
It did not work for me, I did not have TMC at this machine
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Mar 23 2018
echo: Free Memory: 2897  PlannerBufferBytes: 1232
Error:Field planner_extruder_advance_K mismatch.
Error:EEPROM datasize error.
echo:Index: 673 Size: 695
Unified Bed Leveling System v1.01 inactive.
echo:Hardcoded Default Settings Loaded
echo:SD card ok
echo:EEPROM version mismatch (EEPROM= Marlin=V54)
echo:Hardcoded Default Settings Loaded
ok

lienbacher, 2018-03-23:
i got this error too, do a M502 and M500 and see if it works. I do have a TMC though.

tximy, 2018-03-23:
I do M502, M500 but receive the error at M501 ;(

vMeph, 2018-03-23:
i only updated the configuration_store.h file sense is the file that was update i still get error on command M501
READ: echo:EEPROM version mismatch (EEPROM=? Marlin=V53)
EEPROM version mismatch (EEPROM=? Marlin=V53)
READ: echo:Hardcoded Default Settings Loaded
Hardcoded Default Settings Loaded

and dont pops up nothing more

Roxy-3D, 2018-03-23:
> i only updated the configuration_store.h file sense is the file that was update

huh???

[image]

Roxy-3D, 2018-03-23:
Yikes!     We need to change this one too.  
```
      #if HAS_TRINAMIC
        #define TMC_SET_PWMTHRS(P,Q) tmc_set_pwmthrs(stepper##Q, TMC_##Q, tmc_hybrid_threshold[TMC_##Q], planner.axis_steps_per_mm[P##_AXIS])
        uint16_t tmc_hybrid_threshold[TMC_AXES];
        EEPROM_READ(tmc_hybrid_threshold);
        if (!validating) {
          #if X_IS_
```

thinkyhead, 2018-03-23:
I'll patch this momentarily…

thinkyhead, 2018-03-23:
Got this one too:
[image]

tximy, 2018-03-23:
It work nice for me, thx.

gecko749, 2018-03-23:
Hi,

I get this Error with std Steppers. So it is not related to TMC steppers. Tried with bugfix_1.1.x.

thinkyhead, 2018-04-01:
@gecko749 — What error do you get, specifically?

gecko749, 2018-04-01:
Hi,

Size mismatch by 22 Bytes.

thinkyhead, 2018-04-02:
Does `M500` eliminate the error?

gecko749, 2018-04-02:
no

thinkyhead, 2018-04-03:
@gecko749 — As the original issue is fixed, please open a new issue and post your Configuration files there. Be sure to test the current `bugfix-2.0.x` branch, as there have been changes to the EEPROM code in recent days.

closed 2018-04-03 by thinkyhead