MarlinFirmware/Marlin #10184: Bugfix can't compile errors

valleysound, 2018-03-21:
Trying to "upgrade" from Marlin 1.1.8.  This is what I get.

C:\Users\valle\AppData\Local\Temp\arduino_build_773072\sketch\stepper_indirection.cpp:256:0:

C:\Users\valle\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino/HardwareSerial.h:93:7: error: redefinition of 'class HardwareSerial'

 class HardwareSerial : public Stream

       ^

In file included from C:\Users\valle\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino/Arduino.h:232:0,

                 from C:\Users\valle\Documents\Arduino\libraries\TMC2208Stepper-0.1.1\src/TMC2208Stepper.h:5,

                 from C:\Users\valle\AppData\Local\Temp\arduino_build_773072\sketch\Configuration_adv.h:1,

                 from C:\Users\valle\AppData\Local\Temp\arduino_build_773072\sketch\MarlinConfig.h:32,

                 from C:\Users\valle\AppData\Local\Temp\arduino_build_773072\sketch\stepper_indirection.h:47,

                 from C:\Users\valle\AppData\Local\Temp\arduino_build_773072\sketch\stepper_indirection.cpp:34:

C:\Users\valle\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino/HardwareSerial.h:93:7: error: previous definition of 'class HardwareSerial'

 class HardwareSerial : public Stream

Bob-the-Kuhn, 2018-03-21:
What are you upgrading to?

Please ZIP & post your configuration.h & configuration_adv.h files.  That'll help greatly.

valleysound, 2018-03-21:
I was trying to take my stable build Marlin 1.1.8 and upgrade to the Marlin-bugfix-1.1.x to see if it resolved a few issues I was having.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1835153/Configuration.zip)

Bob-the-Kuhn, 2018-03-21:
The following shouldn't have been added to the top of your configuration_adv.h:
```
#include <TMC2208Stepper.h>
#include <TMC2208Stepper_REGDEFS.h>
```		

When I delete them & compile with 1.8.5 I get linker errors.  I'll look at that later tonight.

thinkyhead, 2018-03-22:
@valleysound Be sure to avoid using the **Sketch > Tools > Include Library**  menu in Arduino IDE. Only the top item "**Manage Libraries…**" should be used. The other items will modify the Marlin source code, which is bad!

closed 2018-03-22 by thinkyhead

Bob-the-Kuhn, 2018-03-22:
@valleysound - did it compile successfully?  

I just tried Arduino 1.9 beta with TMC2208 library 0.1.1 and reproduced your errors.  0.0.4 and 0.0.3 also  gave me the same errors.  0.0.2 & 0.0.1 gave me different ones.

Bob-the-Kuhn, 2018-03-22:
I think Arduino hates me tonight.  Compiles just file in PlatformIO but Arduino is an ever changing set of compile errors.

Bob-the-Kuhn, 2018-03-22:
A reboot of my PC fixed my Arduino problems.  Now it compile in 1.8.5 with the 0.0.4 library.