MarlinFirmware/Marlin #10187: Delta UBL only test Right-Back bed zone

tximy, 2018-03-22:
Delta UBL only test Right-Back bed zone.
I try 21/03/2018 1.1.x Rc BugFix with the Delta generic example and enable the UBL over Anycubic Kossel Pulley did not test the complete bed.
Over put this 4 lines at config_advance.h it work nice for me

#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  //#define MESH_MIN_X MESH_INSET
  //#define MESH_MIN_Y MESH_INSET
  //#define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
  //#define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)

  #define MESH_MIN_X (X_MIN_POS + MESH_INSET) // this work nice ;)
  #define MESH_MIN_Y (Y_MIN_POS + MESH_INSET)
  #define MESH_MAX_X (X_MAX_POS - MESH_INSET)
  #define MESH_MAX_Y (Y_MAX_POS - MESH_INSET)

#endif

thinkyhead, 2018-03-29:
With the current `bugfix` you should be able to just comment out all those lines and let Marlin figure out the limits, assuming everything else is correct. How about posting your Configurations so we can give you meaningful feedback?

boelle, 2019-02-20:
@tximy tried what thinky suggested?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead