MarlinFirmware/Marlin #10190: [1.1.8] Please upgrade your toolchain

Titi9237, 2018-03-22:
Hi, I have some trouble to compile the Marlin Firmware on Ubuntu 17

### Description
I get a SanityCheck error asking me to update my packages.
SanityCheck.h:36:4: erreur : #error "Marlin requires C++11 support (gcc >= 4.7, Arduino IDE >= 1.6.8). Please upgrade your toolchain."
But, I am in the last version of everything : 
ii  arduino        2:1.0.5+dfsg all          AVR development board IDE and bui
ii  gcc             4:7.2.0-1ubu amd64        GNU C compiler

### Steps to Reproduce
Just try to compile the Marlin.ino file from Arduino won't work and will raise the previous error.

Expected behavior: to compile

#### Additional Information
All is done with a newly downloaded and freshly unzipped  MarlinFirmware !

Bob-the-Kuhn, 2018-03-22:
I just downloaded 1.1.8, made no changes, compiled with 1.8.5 and had no errors.

If you're on a Windows machine try rebooting.  For me that fixes a lot of strange errors like this.

If that's not the fix then download 1.8.5 installer or 1.9 beta and re-install Arduino.

teemuatlut, 2018-03-22:
> arduino 2:1.0.5

Upgrade your Arduino version. You're using v1.0.5.

Titi9237, 2018-03-23:
Sorry, noob mistake... Thanks !

closed 2018-03-23 by Titi9237

thinkyhead, 2018-10-01:
1.0.5 is too old. Upgrade to 1.8.7. Linux downloads are posted at https://www.arduino.cc/en/Main/Software

comarius, 2018-10-01:
got it. Thx