MarlinFirmware/Marlin #10191: Wont compile with bed leveling on

ferrocam, 2018-03-22:
I am using the same numbers from my 1.1.8 config file

My config files
[Desktop.zip](https://github.com/MarlinFirmware/Marlin/files/1839372/Desktop.zip)

1!. Trying st up 2.0 for mks sbase 

2.When I compile I get the following errors:

In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:51:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:922:8: error: #error "Z_SAFE_HOMING_X_POINT is outside the probe region."
#error "Z_SAFE_HOMING_X_POINT is outside the probe region."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\HAL_timers.cpp:31:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:922:8: error: #error "Z_SAFE_HOMING_X_POINT is outside the probe region."
#error "Z_SAFE_HOMING_X_POINT is outside the probe region."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:23:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:922:8: error: #error "Z_SAFE_HOMING_X_POINT is outside the probe region."
#error "Z_SAFE_HOMING_X_POINT is outside the probe region."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\HardwareSerial.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:922:8: error: #error "Z_SAFE_HOMING_X_POINT is outside the probe region."
#error "Z_SAFE_HOMING_X_POINT is outside the probe region."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\LPC1768_PWM.cpp:74:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:922:8: error: #error "Z_SAFE_HOMING_X_POINT is outside the probe region."
#error "Z_SAFE_HOMING_X_POINT is outside the probe region."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\LPC1768_Servo.cpp:63:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:922:8: error: #error "Z_SAFE_HOMING_X_POINT is outside the probe region."
#error "Z_SAFE_HOMING_X_POINT is outside the probe region."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\SoftwareSPI.cpp:32:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:922:8: error: #error "Z_SAFE_HOMING_X_POINT is outside the probe region."
#error "Z_SAFE_HOMING_X_POINT is outside the probe region."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:29:0,
from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:26,
from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:23:
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:832:19: error: 'DELTA_PRINTABLE_RADIUS' was not declared in this scope
#define X_MAX_POS DELTA_PRINTABLE_RADIUS
^
Marlin\src\HAL\HAL_LPC1768\../../inc/Conditionals_post.h:45:23: note: in expansion of macro 'X_MAX_POS'
#define X_MAX_LENGTH (X_MAX_POS - (X_MIN_POS))
^~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:321:15: note: in expansion of macro 'X_MAX_LENGTH'
static_assert(X_MAX_LENGTH >= X_BED_SIZE && Y_MAX_LENGTH >= Y_BED_SIZE,
^~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:833:19: error: 'DELTA_PRINTABLE_RADIUS' was not declared in this scope
#define Y_MAX_POS DELTA_PRINTABLE_RADIUS
^
Marlin\src\HAL\HAL_LPC1768\../../inc/Conditionals_post.h:46:23: note: in expansion of macro 'Y_MAX_POS'
#define Y_MAX_LENGTH (Y_MAX_POS - (Y_MIN_POS))
^~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:321:45: note: in expansion of macro 'Y_MAX_LENGTH'
static_assert(X_MAX_LENGTH >= X_BED_SIZE && Y_MAX_LENGTH >= Y_BED_SIZE,
^~~~~~~~~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\SoftwareSerial.cpp:39:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:922:8: error: #error "Z_SAFE_HOMING_X_POINT is outside the probe region."
#error "Z_SAFE_HOMING_X_POINT is outside the probe region."
^~~~~
[... 235 lines cut ...]
static_assert(X_MAX_LENGTH >= X_BED_SIZE && Y_MAX_LENGTH >= Y_BED_SIZE,
^~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:968:36: error: 'DELTA_PROBEABLE_RADIUS' was not declared in this scope
#define RIGHT_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS
^
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:856:45: note: in expansion of macro 'RIGHT_PROBE_BED_POSITION'
static_assert(LEFT_PROBE_BED_POSITION < RIGHT_PROBE_BED_POSITION, "LEFT_PROBE_BED_POSITION must be less than RIGHT_PROBE_BED_POSITION.");
^~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:967:37: error: 'DELTA_PROBEABLE_RADIUS' was not declared in this scope
#define LEFT_PROBE_BED_POSITION -(DELTA_PROBEABLE_RADIUS)
^
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:856:19: note: in expansion of macro 'LEFT_PROBE_BED_POSITION'
static_assert(LEFT_PROBE_BED_POSITION < RIGHT_PROBE_BED_POSITION, "LEFT_PROBE_BED_POSITION must be less than RIGHT_PROBE_BED_POSITION.");
^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\LPC1768_debug_and_upload\src\src\HAL\HAL_LPC1768\SoftwareSPI.o] Error 1
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:968:36: error: 'DELTA_PROBEABLE_RADIUS' was not declared in this scope
#define RIGHT_PROBE_BED_POSITION DELTA_PROBEABLE_RADIUS
^
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:856:45: note: in expansion of macro 'RIGHT_PROBE_BED_POSITION'
static_assert(LEFT_PROBE_BED_POSITION < RIGHT_PROBE_BED_POSITION, "LEFT_PROBE_BED_POSITION must be less than RIGHT_PROBE_BED_POSITION.");
^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\LPC1768_debug_and_upload\src\src\HAL\HAL_LPC1768\SoftwareSerial.o] Error 1
 [ERROR] Took 2.66 seconds

GMagician, 2018-03-23:
Don't know anything about delta printers configuration but I see X_BED_SIZE = 200 with a X_MAX_POS = DELTA_PRINTABLE_RADIUS of 85.
How can you expect to do a save home at Z_SAFE_HOMING_X_POINT = ((X_BED_SIZE) / 2)? That is execute Z homing at X=100?

I think first error message is almost self explaining: "Z_SAFE_HOMING_X_POINT is outside the probe region."

boelle, 2019-02-20:
@ferrocam still having issues when using latest bugfix 2.0?

ferrocam, 2019-03-05:
yes

thinkyhead, 2019-03-06:
With `DELTA` there is no need to use `Z_SAFE_HOMING`. If it were used, the safe homing point should be…
```cpp
  #define Z_SAFE_HOMING_X_POINT 0
  #define Z_SAFE_HOMING_Y_POINT 0
```

shitcreek, 2019-05-14:
any update on this issue?

closed 2019-05-16 by thinkyhead