MarlinFirmware/Marlin #10192: Feature Request: Output busy/idle status on PIN

tisfablab, 2018-03-23:
Is it possible to output the busy/idle status of the Printer to an IO Pin?
Or is there any way to "mis-use" an  already existing features? I'd like to detect the instants when the print job begins and when it finishes.
This will be used in a FabLab to monitor the use of the machinery by each user.

TheSFReader, 2018-03-23:
As far as _I_ understand, a _global_ "busy/idle" state doesn't exist for the printer : in serial mode for example, it receives a sequence of individual GCode commands (stacks them/interprets them etc.), and doesn't know if one is specifically the start or stop of a print. 

There are some exceptions of course : during lengthy processes, it answers with busy processing, but that's for the duration of a lengthy GCode instruction (G28/G29 for example)

An other exception is when printing from the SDCard, in which case you can ask for the "SDCard  status" (M27 ?)

If done through a server/host, the server will know if it's launched/finished a print itself, or received a serial SDCard status (which can have been started from a LCD), and maintain it's own state.

Otherwise, I suppose you could check the extruder's temp/target temp, and suppose that if it's set, the printer is busy ?

tisfablab, 2018-03-26:
Thank you!
The machine is used for printing from SD-Card only, therefore the SD Card status should be the right parameter to monitor.

Could someone give me a clue where to begin hacking the source code in order to write that status to a IO PIN? I'm a proficient Arduino coder, but I don't know Marlin (yet).

TheSFReader, 2018-03-26:
I don't know the 1.1 branch as well as the 2.0, but on the 2.0 I'd say go take a look around the sd/cardreader.h /cpp files for that. (but if you plan to share the feature, would impact configuration.h too). 

You'll of course need to choose an available pin to route the signal to, depending on your hardware steup, you'll find that in the pins folder.

If you need to add a command (Mxxx) , there'll be also some work around gcode.cpp / the gcode subfolders.

tisfablab, 2018-03-26:
Thank you! Looks like I'll dive into Marlin over Easter. :)

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

ellensp, 2026-11-11:
This doesn't need any new firmware code. You can set a pin from your slicer's start and end G-code with `M42`, so the pin is high while a job is printing.

1. **Enable the G-codes.** In `Configuration_adv.h`, enable `DIRECT_PIN_CONTROL` (for `M42`). Also enable `PINS_DEBUGGING` (for `M43`) while you choose a pin.

2. **Pick a free pin.** `M43` lists your board's pins and what Marlin uses each one for. Choose one that isn't in use. If `M42` answers with a "Protected Pin" error, Marlin is using that pin, so choose another one. Don't override the protection with `I`.

3. **Set the pin in the slicer's G-code.** Replace `<pin>` with your pin number.

   Start G-code:
   ```gcode
   M42 P<pin> S255   ; Job running
   ```
   End G-code:
   ```gcode
   M42 P<pin> S0     ; Job finished
   ```

4. **Cover cancelled prints.** A print cancelled from the LCD never reaches the end G-code, so add the same command to `EVENT_GCODE_SD_ABORT` in `Configuration_adv.h`:
   ```cpp
   #define EVENT_GCODE_SD_ABORT "G28XY\nM42 P<pin> S0"
   ```

**Wiring:** the pin is a bare MCU output, 5 V on AVR boards and 3.3 V on most 32-bit boards, and it can only supply a few mA. Drive your monitoring input through an optocoupler or a transistor rather than connecting it directly. After a reset or power cycle the pin is an input until a print starts, so put a pull-down resistor on the monitoring side so that it reads "idle" then.

closed 2026-11-11 by ellensp