MarlinFirmware/Marlin #10193: How to get Programatically the G-Code position actual progressing G-code Line&Content

xiaoandy80, 2018-03-23:
Hello,

currently searching for a solution to Upgrade the Advanced Pause function.
I want to be able to determine whenever the Printing process is while Infill or not.
Therefore I have to be able to see/monitor the actual progressed G-code line.
I guess the right Position for checking if the Print is in the Infill Area or not will be right before
this code in the Marlin_main.cpp:

--->Check if the Actual line contains Infill Speed or InfillStart Comment(Slic3r) 

    // Pause the print job and timer
    #if ENABLED(SDSUPPORT)
      if (card.sdprinting) {
        card.pauseSDPrint();
        sd_print_paused = true;
      }
    #endif
    print_job_timer.pause();

But how can I Check/Wait for a specific G-code line content to stop and proceed with the filament change?

Any Help and suggestions are welcome.
Thank you in advance.

thinkyhead, 2018-03-29:
Is that really the best approach? Does your slicer have an option to insert G-code before and after infill? Marlin ignores comments at a very low level in the parser.

xiaoandy80, 2018-03-29:
I guess I can set the infill speed to a specific like 33mm/s and can just wait for the infill speed then stop. But how to see what's progressed at the moment. I do not even care if there will be some more moves from the buffer. There must be a solution like u can read the X y z coordinates it is delayed also during the print but it can.be displayed there has to be a function where I can do the same like with the coordinates with the executed gcode line.

thinkyhead, 2018-04-01:
If the slicer has an option to insert G-code before an infill, my suggestion would then be to implement a G-code that performs the needed task and tell the slicer to insert it before starting infill. Another option is to tell the slicer to use a different extruder for infill and do some custom behavior in response to `T1` (where normally it would be ignored with a single extruder).

xiaoandy80, 2018-04-01:
I don’t want a workaround I need to know where the progressed gcode is handled the resto implement myself. I just need to know where the next code line is executed in the marlin firmware.

thinkyhead, 2018-04-02:
> I just need to know where the next code line is executed in the marlin firmware.

`GcodeParser::parse`, `process_next_command`, `process_parsed_command`

boelle, 2019-02-20:
@xiaoandy80 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

xiaoandy80, 2019-02-20:
Hello

You clearly not read my post it’s advice where to get information about a specific action procedure in the marlin firmware to maybe be able to Programm it myself. It has nothing todo with normal configuration and setup.

Regards,
Andreas Herrmann

Application Engineer
T: +65 83088048
E: Xiaoandy@me.com
Singapore

boelle, 2019-02-20:
the list is for bugs etc... not advice or questions

xiaoandy80, 2019-02-20:
Feel free to delete I’ll find out myself.

Regards,
Andreas Herrmann

Application Engineer
T: +65 83088048
E: Xiaoandy@me.com
Singapore

closed 2019-05-16 by thinkyhead