MarlinFirmware/Marlin #10194: Cant compile with 'HAVE_TMC2130' Defined. (Bugfix2.0)

maccadelic, 2018-03-23:
Using an MKS SBASE 1.3; With 'HAVE TMC_2130' defined i can not compile. With this commented out, I can compile and run marlin as expected. I have tried disabling SD support, different pins, setting TMC_SW pins, different serial port numbers.

The output from platformIO is as follows;

```
 ^
Marlin\src\module\configuration_store.cpp:1897:23: note: in expansion of macro 'SERIAL_ECHOPGM_P'
void say_M913() { SERIAL_ECHOPGM_P(port, "  M913 "); }
^~~~~~~~~~~~~~~~
Marlin\src\module\configuration_store.cpp:1897:40: note: suggested alternative: 'powl'
void say_M913() { SERIAL_ECHOPGM_P(port, "  M913 "); }
^

Marlin\src\module\../inc/../core/serial.h:102:69: note: in definition of macro 'SERIAL_PROTOCOLPGM_P'
#define SERIAL_PROTOCOLPGM_P(p,x)               (serialprintPGM_P(p,PSTR(x)))
^
Marlin\src\module\configuration_store.cpp:1897:23: note: in expansion of macro 'SERIAL_ECHOPGM_P'
void say_M913() { SERIAL_ECHOPGM_P(port, "  M913 "); }
^~~~~~~~~~~~~~~~
*** [.pioenvs\LPC1768\src\src\module\configuration_store.o] Error 1
 [ERROR] Took 48.39 seconds
 
 [SUMMARY]
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment LPC1768_debug_and_upload    [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
Environment LPC1768                     [ERROR]
 [ERROR] Took 48.40 seconds

```

My config files:
[Marlinconfigs.zip](https://github.com/MarlinFirmware/Marlin/files/1841073/Marlinconfigs.zip)

Bob-the-Kuhn, 2018-03-23:
Looks like a "replace all" grabbed too many things (or is it missed?) and those statements weren't setup correctly.

Try the following file and let me know how it works.
[configuration_store.zip](https://github.com/MarlinFirmware/Marlin/files/1842340/configuration_store.zip)

I'll start a PR to get this fixed.

Bob-the-Kuhn, 2018-03-23:
PR #10197 will fix this problem.

maccadelic, 2018-03-24:
I am now getting this error when compiling. I have tried removing and reinstalling the library in PlatformIO.

```
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:23:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1428:6: error: #error "Update TMC2130Stepper library to 2.2.1 or newer."
#error "Update TMC2130Stepper library to 2.2.1 or newer."
^~~~~
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:51:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1428:6: error: #error "Update TMC2130Stepper library to 2.2.1 or newer."
#error "Update TMC2130Stepper library to 2.2.1 or newer."
^~~~~
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HardwareSerial.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\LPC1768_PWM.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\LPC1768_Servo.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\SoftwareSPI.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\SoftwareSerial.o
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.o] Error 1
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\HAL_timers.cpp:31:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1428:6: error: #error "Update TMC2130Stepper library to 2.2.1 or newer."
#error "Update TMC2130Stepper library to 2.2.1 or newer."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\SoftwareSerial.cpp:39:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1428:6: error: #error "Update TMC2130Stepper library to 2.2.1 or newer."
#error "Update TMC2130Stepper library to 2.2.1 or newer."
^~~~~

In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\LPC1768_Servo.cpp:63:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1428:6: error: #error "Update TMC2130Stepper library to 2.2.1 or newer."
#error "Update TMC2130Stepper library to 2.2.1 or newer."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\LPC1768_PWM.cpp:74:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1428:6: error: #error "Update TMC2130Stepper library to 2.2.1 or newer."
#error "Update TMC2130Stepper library to 2.2.1 or newer."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\HardwareSerial.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1428:6: error: #error "Update TMC2130Stepper library to 2.2.1 or newer."
#error "Update TMC2130Stepper library to 2.2.1 or newer."
^~~~~
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:34:0,
from Marlin\src\HAL\HAL_LPC1768\SoftwareSPI.cpp:32:
Marlin\src\HAL\HAL_LPC1768\../../inc/SanityCheck.h:1428:6: error: #error "Update TMC2130Stepper library to 2.2.1 or newer."
#error "Update TMC2130Stepper library to 2.2.1 or newer."
^~~~~
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\LPC1768_Servo.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\SoftwareSerial.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\LPC1768_PWM.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HardwareSerial.o] Error 1
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\SoftwareSPI.o] Error 1
 [ERROR] Took 9.07 seconds
```

Bob-the-Kuhn, 2018-03-24:
We'll have to back the 2.2.1 check out.  

The check is done before the TMC2130 library is compiled so it can't pass.

Please delete lines 1427 & 1428 in \Marlin\src\inc\SanityCheck.h.  That'll get you going until we merge in the fix.

maccadelic, 2018-03-24:
Compile successful!

Thanks heaps on this issue and for your hard work overall on the project!

Bob-the-Kuhn, 2018-03-24:
PR #10203 fixes this.

closed 2018-03-24 by Bob-the-Kuhn