MarlinFirmware/Marlin #10196: BUG bugfix 1.1.x downloaded 3-23-2018 - 'class TMC2130Stepper' has no member named 'hysteresis_start'

dcwalmsley, 2018-03-23:
This morning I downloaded and updated Configuration.h, Configuration_Adv.h, and pins_EINSY_RAMBO.h to match my printer's configuration.  However, I have since received the below message when attempting yo VERIFY the modifications prior to Compiling and Uploading it to my HyperCube.

I've noticed some changes you all made to the Configuration_Adv.h file.  I've attached these files zipped for your review.

R,
Doug W

[DCW Files.zip](https://github.com/MarlinFirmware/Marlin/files/1842209/DCW.Files.zip)

 
Message Received when attempting to compile.
-------------------------------------------------------------------------------------------------------------

C:\Users\dcwal\AppData\Local\Temp\arduino_build_273005\sketch\stepper_indirection.cpp: In function 'void tmc2130_init(TMC2130Stepper&, uint16_t, uint16_t, uint32_t, float)':

stepper_indirection.cpp:185: error: 'class TMC2130Stepper' has no member named 'hysteresis_start'

     st.hysteresis_start(3);
        ^
stepper_indirection.cpp:186: error: 'class TMC2130Stepper' has no member named 'hysteresis_end'

     st.hysteresis_end(2);
        ^
C:\Users\dcwal\AppData\Local\Temp\arduino_build_273005\sketch\stepper_indirection.cpp: In function 'void restore_stepper_drivers()':
stepper_indirection.cpp:482: error: 'class TMC2130Stepper' has no member named 'push'

     stepperX.push();
              ^

stepper_indirection.cpp:488: error: 'class TMC2130Stepper' has no member named 'push'

     stepperY.push();
              ^

stepper_indirection.cpp:494: error: 'class TMC2130Stepper' has no member named 'push'

     stepperZ.push();
              ^

stepper_indirection.cpp:500: error: 'class TMC2130Stepper' has no member named 'push'

     stepperE0.push();
               ^

Using library TMC2130Stepper at version 2.1.4 in folder: C:\Users\dcwal\Documents\Arduino\libraries\TMC2130Stepper 
Using library SPI at version 1.0 in folder: C:\Users\dcwal\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI 
Using library U8glib in folder: C:\Users\dcwal\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.1\libraries\U8glib (legacy)
exit status 1
'class TMC2130Stepper' has no member named 'hysteresis_start'

Bob-the-Kuhn, 2018-03-23:
Try updating your TMC2130 library to 2.2.1

There's some warnings about redefining the E_MUX0_PIN &  E_MUX1_PIN .  Looks like the definitions in **pins_EINSY_RAMBO.h** are going to be used rather than the ones in **configuration.h.**

Bob-the-Kuhn, 2018-03-23:
I was finally able to get Arduino to run.

With your library I get the same errors.  With 2.2.1 it compiles without errors.

thinkyhead, 2018-03-23:
This will now throw an alert in `SanityCheck.h`.

closed 2018-03-23 by thinkyhead

dcwalmsley, 2018-03-23:
Upgraded TM 2130 library and compiled latest bugfix with no errors. Thanks 
for providing me the recommendation.

Closed #10196.