MarlinFirmware/Marlin #10201: [bugfix-1.1.x] ABL and Z_OFFSET and Dual Extruders

gecko749, 2018-03-23:
### Description

With 1.1.x ABL - bilinear didn't work because I couldn't set negativ Z-Offsets in the Firmware. I switched to the bugfix-1.1 and managed to get the printer working with Extruder 1. Testing with extruder 2 failed again.

As with 1.1.x, the printhead is moved to max_x, max_y and the extruder 2 starts to extrude as fast as possible.

This seems to be linked to negativ values for M851. But unfortunatly these value must be negativ or the head crashes on the bed.

thinkyhead, 2018-03-23:
> Testing with extruder 2 failed again.

Pretty vague! In what sense?

We have bullet points in the Issue Template that would really help us if you could fill them out.

- Expected behavior
- Actual behavior
- Steps to reproduce
- Attach your configurations

When we have more information we can proceed to investigate.

gecko749, 2018-03-24:
Hi,

The printer is a Flsun Cube with dual Extruders.

I tried the stansard 1.1.x with the Z_offset problems and ABL.

I switched to bugfix-1.1.x yesterday and it worked for extruder 1. That means I could print a test.

then I wanted to print the same test with extruder 2 and it failed. I did nothing in between exept loading the new File for extruder 2.

Expected behavior: printing the same part with extruder 2.
Actual behavior: After heating and homing, head went to max_x and max_y, z remained he same. The extruder startet to extrude as much material as possible.

So the extruder showed the same failure as with std_1.1.x and extruder 1.

[Arduino.zip](https://github.com/MarlinFirmware/Marlin/files/1844226/Arduino.zip)

The tests prints were ok without ABL enabled. (std 1.1.x )

gecko749, 2018-03-24:
Hi,
now I got the same failure with std-1.1.8 Bed leveling disabled, eeprom disabled.
Using extruder 2.
First test print was ok, then I changed manually the bed levelling and startet the same print again. No reboot, no other change. This time the extruder went to x_max, y_max z unchanged and startet extruding as fast as possible.

Next test with eeprom enabled and reset ( M502 -M500 ).
Same failure.

Using extruder 1. Same problem.

Looks like a c-pointer problem. A value is overwritting something.

gecko749, 2018-03-25:
More Tests ...

I created a gcode file and tested it on an other single extruder printer.

Then I flashed the 1.1.4 from Flsun set to single extruder.

Testeted ok.

Reflashed with dual extruder 1.1.4.

Tested ok. Inserted T0 before any other command. Test ok. Changed T0 to T1. Print run, but with z=0.0 instead 0.2.

gecko749, 2018-03-25:
I installed 1-0.2-2 and its working w/o ABL Didn't test ABL yet.

thinkyhead, 2018-03-25:
Thanks for all the extra info. I will investigate over the next few days. I may ask for more testing with added debugging.

gecko749, 2018-03-25:
Thanks for the feedback.

A few more Info - the problem ocures meanwhile even with 1.0.2-2. 

Th printer is new for me and came with a TFT attached to AUX1. I extended the RX Buffer to 256 B without luck.
I run out of ideas ...

thinkyhead, 2018-03-29:
> After heating and homing, head went to max_x and max_y,

Sounds like the coordinate space is getting shifted by your tool change. If you simply select `T1` while the machine is idle what does the `M114` report say about the current position?

Please post the G-code you're trying to use. It would be good to make sure that there's no `G92` or some other command messing with the current position.

gecko749, 2018-03-29:
Hi,

it is getting more wired with that printer.

I managed to print some testpatterns to adjust the distance beetween the nozzles. 

Until I started a longer print, the problem occcured after homing, before printing. If the head moves to the first print position at z_homing it starts to go to max_x_y with a different speed.. 
If ok, the head moves to start position and z_start and the print starts and finishes ok.

in that longer print the printer lost his z_position after half an hour and the head went to z_0.

At this time I decided to put the printer on hold and do something different.

Looks like there is an other problem interfering. I'll test differrent hardware to ort this out.

Thanks for your support,I 'll keep this updated if I find a solution.

boelle, 2019-02-20:
@gecko749 is this still an issue with latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead