MarlinFirmware/Marlin #10202: UBL + Dual extruder  = compile error

alexxy, 2018-03-24:
If i enable UBL + BLtouch + Dual extruder, i cannot compile marlin for avr (i use bugfix-2.0.x)

```
Compiling .pioenvs/megaatmega2560/src/src/sd/Sd2Card.o
Marlin/src/module/tool_change.cpp: In function 'void tool_change(uint8_t, float, bool)':
Marlin/src/module/tool_change.cpp:458:25: error: 'class Planner' has no member named 'apply_leveling'
planner.apply_leveling(current_position[X_AXIS], current_position[Y_AXIS], z1);
^
Marlin/src/module/tool_change.cpp:459:25: error: 'class Planner' has no member named 'apply_leveling'
planner.apply_leveling(x2, y2, z2);

```
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1844360/configs.zip)

alexxy, 2018-03-24:
Seems like its because PLANER_LEVELING is not defined when HAS_MESH defined

autonumous, 2018-03-24:
I get the same error today and was going to determine what was the cause was and when it was introduced....

Bob-the-Kuhn, 2018-03-24:
@Roxy-3D - please take a look at these lines in **tool_change.cpp**.              
```
458  planner.apply_leveling(current_position[X_AXIS], current_position[Y_AXIS], z1);
459  planner.apply_leveling(x2, y2, z2);
```

Planner leveling shouldn't be used because UBL has been selected.

Bob-the-Kuhn, 2018-03-24:
PR #10199 extended the Z compensation change to UBL yesterday. 

Work around is to change line 449 back to **#if ENABLED(MESH_BED_LEVELING)**

autonumous, 2018-03-24:
Just re-read original post. my bad, but I get this issue with bugfix-1.1.x not in 2.0.x

autonumous, 2018-03-24:
so probably   https://github.com/MarlinFirmware/Marlin/pull/10198  :-)

Bob-the-Kuhn, 2018-03-25:
My best guess is the offending code change was part of the Issue #10130 investigation.  

I think it's best to stick with the work around until that issue is resolved.

thinkyhead, 2018-03-28:
I've posted the simplified tool change code in a PR at #10243. The direct download for that PR is at https://github.com/thinkyhead/Marlin/archive/bf1_tool_change_debug.zip if anyone wants to give it a test run.

alexxy, 2018-04-04:
@thinkyhead #10302 for issue with z-offset

closed 2018-04-05 by thinkyhead