MarlinFirmware/Marlin #10205: [bugfix-1.1.x] G26 Mesh validation not using M206 Z offset?

brookstalley, 2018-03-24:
I'm having some mesh trouble, and I can't tell if it is my config, my hardware, or an actual issue with bugfix-1.1.

Steps:
- G29 I25 to invaldate all points in mesh
- G29 P2 B0.2 with metal feeler to get mesh
- G26 to get validation print
- G29 P5 to get mean mesh height (ex: 0.8)
- G29 P6 C-0.8 to subtract mean 
- M206 Z0.8 to set Z offset
- G26 for validation print

Exect: same grid as before moving the mes mean into M206
Results: nozzle crashes into bed

Testing with various values of M206 seems to indicate that it is ignored for G26.

thinkyhead, 2018-03-29:
You are correct! This is a bug. This change is needed:
```diff
-   destination[Z_AXIS] = g26_layer_height;
+   destination[Z_AXIS] = LOGICAL_Z_POSITION(g26_layer_height);
```

thinkyhead, 2018-03-29:
I think the issue is that you are using `M206` improperly. It should only be adjusted if your Z home is off, and should not be used to compensate for the mean mesh height. Instead you should be using `M851 Z` and adjusting the existing value by 0.8 so that future `G29` readings have a mean mesh height of 0.0.

closed 2018-03-29 by thinkyhead

brookstalley, 2018-03-29:
Thanks... that definitely explains it. But I don't think that works well for printers that home to z-max.

I can set M851 Z appropriately, but means M851 Z6.2 despite the probe actually being 0.5mm from the nozzle. Or I can set M851 Z0.5 (actual distance from probe to nozzle), and get mesh values of >5mm, which blows up in a bunch of ways. 

I think this means that I need to compensate for different bed heights by adjusting the Z_MAX_POS in the firmware and recompiling... which is a bit ugly.  I had thought that's what M206 was for, but I hear you that it's not.

Or... maybe you're saying that M851 is really [distance from probe to nozzle + difference between Z_MAX_POS and bed height]?

thinkyhead, 2018-03-29:
`M206 Z` is for adjusting the Z home position. You should set it to whatever value results in `G1 Z0` putting the nozzle right on the bed (or ever-so-slightly above it). That's a good first calibration step.

Once you've established a good home position, then you should figure out how high the nozzle is when the probe triggers. (This can be done with `M401`+`G1 Z0` or just `G30`, for example.)

Negate that Z height and use it for your `M851 Z`. Then your next `G29` should return a mean mesh value close to 0, assuming your home position is not far off.

jaapp, 2018-04-30:
> I can set M851 Z appropriately, but that means M851 Z6.2 despite the probe actually being 0.5mm from the nozzle. Or I can set M851 Z0.5 (actual distance from probe to nozzle), and get mesh values of >5mm, which blows up in a bunch of ways.

@brookstalley It's not the distance from the probe-tip to nozzle-tip that's defined as the offset. If you're using a capacative/inductive sensor the triggerpoint of the sensor can very well be 6.2mm above the nozzle tip because it's sensing over a distance.